FastLED/FastLED #700: Error when increasing the number of LEDs

agootday, 2018-12-17:
Good day. For a long time I can not understand what the problem is.
I get the LED number and color from a com-port, after which I draw a tape. When using 30 LEDs - everything works fine. But if I install 60 LEDs, then everything stops working as it should. Colors start to be displayed incorrectly and with a delay, provided that the same data for the first 19 LEDs will be obtained through the com port. That is, I only change the size of the array and that's it.
Help me please.

`#define NUM_LEDS 60  //30-ok, 60-error
#define DI_PIN 12
#define BRIGHT 200   
#define serialRate 19200 
#define PACSIZE 5

uint8_t prefix = 255;
byte setLedNum = 0,
     setColorR = 0,
     setColorG = 0,
     setColorB = 0;

#include <FastLED.h>
CRGB leds[NUM_LEDS];  

void setup()
{
  FastLED.addLeds<WS2812, DI_PIN, GRB>(leds, NUM_LEDS);
  FastLED.setBrightness(BRIGHT);
  delay(1000);
  Serial.begin(serialRate);
}

void loop() {
  for (uint8_t i = 0; i < PACSIZE; i++) {
waitLoop: while (Serial.available() == 0) {}
    if (i == 0) {
      if (prefix == Serial.read()) {
        continue;
      }
    } else {
      if (i == 1) {
        setLedNum = Serial.read();
        if (setLedNum != prefix) {continue;}
      } else if (i == 2) {
        setColorR = Serial.read();
        if (setColorR != prefix) {continue;}
      } else if (i == 3) {
        setColorG = Serial.read();
        if (setColorG != prefix) {continue;}
      } else if (i == 4) {
        setColorB = Serial.read();
        if (setColorB != prefix) {
          leds[setLedNum].r = setColorR;
          leds[setLedNum].g = setColorG;
          leds[setLedNum].b = setColorB;
          FastLED.show();
          continue;
        }
      }
    }
    i = 0;
    goto waitLoop;
  }
}`

agootday, 2018-12-17:
[This](https://github.com/FastLED/FastLED/wiki/Interrupt-problems) solved my problem, sorry for the trouble.

marcmerlin, 2018-12-30:
@agootday  can you please resolve/close the issue? Thanks.

closed 2019-01-02 by agootday