FastLED/FastLED #703: unused variable "clocks" warnings

putzlicht, 2018-12-20:
Compiling a sketch for Teensy3.6 with clockless chips in block-mode gives me a lot of
unused variable "clocks" warnings.
In platforms/arm/clockless_block_arm_k66.h i just changed:

	virtual void showPixels(PixelController<RGB_ORDER, LANES, LANE_MASK> & pixels) { 
		mWait.wait();
		uint32_t clocks = showRGBInternal(pixels);
		#if FASTLED_ALLOW_INTERRUPTS == 0
		// Adjust the timer
		long microsTaken = CLKS_TO_MICROS(clocks);
		MS_COUNTER += (1 + (microsTaken / 1000));
		#endif

		mWait.mark();
	}

To:

	virtual void showPixels(PixelController<RGB_ORDER, LANES, LANE_MASK> & pixels) { 
		mWait.wait();

		#if FASTLED_ALLOW_INTERRUPTS == 1
		showRGBInternal(pixels);
		#endif

		#if FASTLED_ALLOW_INTERRUPTS == 0
		uint32_t clocks = showRGBInternal(pixels);
		// Adjust the timer
		long microsTaken = CLKS_TO_MICROS(clocks);
		MS_COUNTER += (1 + (microsTaken / 1000));
		#endif

		mWait.mark();
	}

to avoid this.
And i changed this again starting from line 242 in the file.

FastLED version 3.002.001
Arduino IDE 1.8.5
Teensyduino 1.42

marcmerlin, 2018-12-30:
I don't have commit access to the code, but usually the best way to get this fixed while saving time for the overworked maintainers, is to create a pull request on github. You can look in github help how to do this if you've never done it.

marcmerlin, 2019-01-04:
Actually since you are using teensyduino , check if the FastLED code has the same issue there than in FastLED master from the source. Depending on that you should report the fix to teensy or send a PR to this project.