FastLED/FastLED #709: Order problem chinese WS2812B rings

esp13, 2018-12-30:
Hi,
I bought some WS2812B rings, individually they are working great.

If I wire like this:
[DataPin] > in [Type 2] out > in [Type 1] > in [Type 1]
all are working

If I wire like this:
[DataPin] > in [Type 1] out > in [Type 2] > in [Type 1]
only the fisrt "Type 1" is working

If I wire like this:
[DataPin] > in [Type 1] out > in [Type 1] > in [Type 2]
only the "Type 2" is not working

Two questions:

- I suppose this comes from differences between chips speed? Could I slow down all to make all working?

- Is there a way to declare multiple strip on same data pin like one big part but to change the order from the parts?

Subsidiary question:

- I have a third ring type where two colors are reversed. I now how to correct that in FastLed when there is only one type per data pin. But is there a way to get multiple parts types on same data pin and correct only the one which is wrong?

Thank you for having read :)  (and sorry for my poor english)

focalintent, 2018-12-30:
The ws28xx style chips each regenerate the signal - and it sounds like the signal generated by the chips on your type 1 rings is not usable by the chips on your type 2 rings.  There isn’t a lot that can be done at the library level for this, I believe. 

And FastLED doesn’t allow for multiple chips or rgb ordering a on a single line.  You will have to reorder the leds by hand before calling FastLED.show and then un-do that afterwards.

esp13, 2018-12-30:
> The ws28xx style chips each regenerate the signal - and it sounds like the signal generated by the chips on your type 1 rings is not usable by the chips on your type 2 rings. There isn’t a lot that [...]

Oh damn :'(
Thank you for your answer.
Do you know a function/library to make the manual reordering easier?

- Invert color
- Invert way
- Shift

focalintent, 2018-12-30:
I haven't worked something like this into the library yet - and haven't tried it out specifically - but you could use something like this:

```
template<EOrder RGB_ORDER> void swapRGBOrder(CRGB * leds, unsigned int numLeds) {
  for(unsigned int i = 0; i < numLeds) {
    CRGB orig = leds[i];
    leds[i][0] = orig[RO(0)];
    leds[i][1] = orig[RO(1)];
    leds[i][2] = orig[RO(2)];
  }
}

template<EOrder RGB_ORDER> void unswapRGBOrder(CRGB * leds, unsigned int numLeds) {
  for(unsigned int i = 0; i < numLeds) {
    CRGB swapped = leds[i];
    leds[i][RO(0)] = swapped[0];
    leds[i][RO(1)] = swapped[1];
    leds[i][RO(2)] = swapped[2];
  }
}
```

and then in your code:

```
   swapRGBOrder<GRB>(leds + swappedOffset, numberOfSwappedLeds);
   FastLED.show();
   unswapRGBOrder<GRB>(leds + swappedOffset, numbetOfSwappedLeds);
```

closed 2019-01-20 by focalintent