FastLED/FastLED #711: Wrong number of LEDs on ws2812b

joaovalentee, 2019-01-01:
Hey guys I have a quick question. I recently bought a ws2812b strip. I firstly tested it with an Arduino and everything worked fine but I really wanted to control it with an ESP32. The problems appeared when I tried the ESP32. 
My strip has 300 leds but I want to control only 90. But when I define the NUM_LEDS 90 somehow more than 90 leds turn on. Can’t understand why.
Thanks.

RococoN8R, 2019-01-01:
I'd suggest trying the DemoReelESP32 with your strip and see if the same 
issue arises.
I had a similar experience, but it was the result of poorly sequenced 
code early in the definitions.
If the demo works, use it to confirm your coding flow is correctly 
sequenced.

Ray

marcmerlin, 2019-01-04:
@focalintent  @samguyer  seems that https://github.com/FastLED/FastLED/tree/master/examples/DemoReelESP3 should be removed as per https://plus.google.com/113400909968993296978/posts/Bb4F6jPsYJD
Would you agree?

closed 2019-08-12 by focalintent