FastLED/FastLED #716: FastLed library not compiling on the new Arduino Uno Wifi Rev. 2

PMPoulsen, 2019-01-09:
Hi,
I have used one of the examples from the library and tried to compile it to my new Arduino wifi rev. 2 board,but I get the following error message:

------------------
Arduino: 1.8.8 (Windows 10), Board: "Arduino Uno WiFi Rev2, ATMEGA328"

In file included from C:\Arduino Projects\libraries\FastLED\examples\ColorPalette\ColorPalette.ino:1:0:

C:\Arduino Projects\libraries\FastLED/FastLED.h:14:21: note: #pragma message: FastLED version 3.002.001

 #    pragma message "FastLED version 3.002.001"

                     ^

In file included from C:\Arduino Projects\libraries\FastLED/FastLED.h:65:0,

                 from C:\Arduino Projects\libraries\FastLED\examples\ColorPalette\ColorPalette.ino:1:

C:\Arduino Projects\libraries\FastLED/fastspi.h:110:23: note: #pragma message: No hardware SPI pins defined.  All SPI access will default to bitbanged output

 #      pragma message "No hardware SPI pins defined.  All SPI access will default to bitbanged output"

                       ^

In file included from C:\Arduino Projects\libraries\FastLED/FastLED.h:48:0,

                 from C:\Arduino Projects\libraries\FastLED\examples\ColorPalette\ColorPalette.ino:1:

C:\Arduino Projects\libraries\FastLED/fastpin.h: In instantiation of 'class FastPin<5u>':

C:\Arduino Projects\libraries\FastLED/platforms/avr/clockless_trinket.h:96:49:   required from 'class ClocklessController<5u, 6, 8, 6, (EOrder)66u, 0, false, 10>'

C:\Arduino Projects\libraries\FastLED/chipsets.h:385:7:   required from 'class WS2811Controller800Khz<5u, (EOrder)66u>'

C:\Arduino Projects\libraries\FastLED/FastLED.h:108:52:   required from 'class WS2811<5u, (EOrder)66u>'

C:\Arduino Projects\libraries\FastLED/FastLED.h:295:39:   required from 'static CLEDController& CFastLED::addLeds(CRGB*, int, int) [with CHIPSET = WS2811; unsigned char DATA_PIN = 5u; EOrder RGB_ORDER = (EOrder)66u]'

C:\Arduino Projects\libraries\FastLED\examples\ColorPalette\ColorPalette.ino:40:67:   required from here

C:\Arduino Projects\libraries\FastLED/fastpin.h:207:2: error: static assertion failed: Invalid pin specified

  static_assert(validpin(), "Invalid pin specified");

  ^

exit status 1
Error compiling for board Arduino Uno WiFi Rev2.
---------------

I understand that the new Wifi Rev. 2 board uses a different chip hence the pinout may not have been defined in the library yet.

Are there any simple workarrounds for this issue?

focalintent, 2019-01-09:
Nothing simple, no - at the very least there need to be pin definitions setup in fastpin_avr.h - and that’s assuming there’s no major differences between the atmega4809 and the 32u4/328.

PMPoulsen, 2019-01-09:
OK, that is beyond my skills, so I am hoping for an update to the library soon.
Thanks

scr34m88, 2019-01-16:
I've got the same Problem. I hope you'll update the Library soon.
Thanks

scr34m88, 2019-02-04:
Does someone fix this problem for himself?
At the moment I cannot use my Wifi Rev2. anymore because I need this library :)

ngyl88, 2019-05-22:
Hi all, I've hit the same issue as well. Well, pin definitions are beyond my skills. Hopefully someone can help to update the library soon.

barnosch, 2019-07-12:
Same happens on the new Nano Every Board which is aswell Atmega4809.

Any news here? Sadly i can't fix that myself.

jvcop, 2019-08-04:
Same for me with the Arduino Nano Every. Are there plans on supporting Atmega4809?

I'm willing to contribute, but this is new ground for me.

focalintent, 2019-08-12:
I have a nano every board on order and will be adding support for that soon, depending on when I have time.   Once I have that running I might be able to get support for the WiFi uno rev 2 without having to buy a board.

FDelporte, 2019-10-09:
Hello Daniel, any updates on this issue? Just discovered this issue after I bought a uno wifi v2 for a project for which the code was tested OK on an older uno... Would like to help but not enough knowledge of this level of coding for Arduino.

kriegsman, 2019-10-09:
Hi @FDelporte.  I'm sorry to bring the sad news, but Dan passed away just about a month ago.  As you can imagine, that's slowed everything down.  
FastLED will continue, but I'm taking a bit of a break first.  In the meantime, the FastLED community on Reddit is full of helpful people to assist with general questions -- although that probably doesn't help with the Uno Wifi v2 right now.  Sorry to be the bearer of bad news.
-Mark

FDelporte, 2019-10-09:
Wow @kriegsman what a very bad news! I'm working on a book and am using this library as an integration library between arduino, java and raspberry pi. Was going to try to get in contact with Dan to thank him for this fantastic piece of work. Unfortunately I'm too late for that... My condolences to his family, friends and colleagues!!!
Never mind the v2 support right now!

kriegsman, 2019-10-09:
Thank you. It’s great to hear that you’re using FastLED in your book. I’ll let the family know that it was yet another way that Dan was helping people — and still is.

jacopomazzoni, 2019-10-29:
Hi @kriegsman, 
I'm sure this is a difficult time and I'd like to add my voice to the chorus of the miriad of makers that  loved Dan's work and felt the vacuum left by his absence.

With that in mind I think if you could tell us briefly what to do and what files to edit the community would be more than happy to help adding support to future boards.

I think we would all like to chip in to help keep his legacy alive, we just need to know what steps to take to add support to new boards, we can split the workload and figure it out on our own leaving you and your team the time to grieve without leaving many students ( like mine ) that were allured by the arduino nano every price-point stranded. 

Cordially,
Jacopo

kriegsman, 2019-10-29:
Hi and thank you for reaching out. We have family matters that are taking up a lot of our time and hearts, but FastLED is prominent in my thoughts as well. I hope to have a plan for going forward by the end of the year, at latest. 
Thanks again-
-Mark

jacopomazzoni, 2019-10-29:
Thank you guys for all you've done over the years.
Could you also let us know if there is a fund we can donate to, again we'd love to give back to Dan and his family since he gave us so much.

stephensrj, 2020-02-05:
I noticed that version 3.3.3 has been released. Does it work with the UNO Wifi Rev 2 now?

kriegsman, 2020-02-05:
To the best of my knowledge, support for the Atmega4809 is still on the to-do list.
I'm going to rename this ticket to "Support for Atmega4809", and let's use this thread as a place to start gathering the information we need about this new chip so that we can support it.  I suspect that once we have all the relevant Pin/Port info collected, it'll be relatively quick.  So: please feel free to add links to comments on this Issue with specific info about the ATmega4809's Pin/Port mappings and such.

kriegsman, 2020-02-05:
Collecting some info here:
https://hackaday.io/project/134831-atmega4809-developing-board-project/log/143959-evolution-of-my-struggling-with-atmega4809

> As a proof of AVR, it has Fuse bit!, but indeed ATmega4809 is more similar to ATXmega, not general AVR like Mega328 or Tiny series in the programming manner. (You will see if you touch this chip., like DDRB is not accepted by ATMega4809 but PORTB.DIRSET)..
> 
> 
>   | Mega,Tiny | ATMega4809
> -- | -- | --
> Port direction | DDRB | PORTB.DIRSET
> Value definition | PORTB | PORTB.OUTSET
> 
> 
>

stephensrj, 2020-02-06:
Thank you for supporting this.

jacopomazzoni, 2020-02-20:
Hi @kriegsman,
Not sure if it helps but the datasheet is pretty extensive:
https://www.microchip.com/wwwproducts/en/ATMEGA4809

could you point us to the right section of the library documentation to learn how to help?

I'd be willing to test it with my arduino nano every but I'm not too sure where to start altering the library to add support for this specific hardware.

If such a documentation does not already exist it might be a good time investment as it would allow the community at large to jump in and contribute more.

as always thanks for all you do,
Best,
Jacopo

Multensor, 2020-04-03:
Is there any news on the support for ATmego4809? I got some old nano's laying around as well, but i rather use the every for my project :).

SEKCobra, 2020-04-10:
Any news on this? I am also hoping to use this on an Arduino Nano Every

Jueff, 2020-04-28:
maybe information on this page help with port/pin mapping
https://tomalmy.com/gpio-on-arduino-nano-every/

Jueff, 2020-04-28:
or have a look in file ..\packages\arduino\hardware\megaavr\1.8.6\cores\arduino\NANO_compat.h

```
/*
  ARDUINO PIN  ATMEGA 328  ATMEGA 4809
  D0           PD0         PC4 (TX1)
  D1           PD1         PC5 (RX1)
  D2           PD2         PA0
  D3           PD3         PF5
  D4           PD4         PC6
  D5           PD5         PB2
  D6           PD6         PF4
  D7           PD7         PA1
  D8           PB0         PE3
  D9           PB1         PB0
  D10          PB2         PB1
  D11          PB3         PE0 (MOSI)
  D12          PB4         PE1 (MISO)
  D13          PB5         PE2 (SCK)
  A0           PC0         PD3
  A1           PC1         PD2
  A2           PC2         PD1
  A3           PC3         PD0
  A4           PC4         PA2 (SDA)
  A5           PC5         PA3 (SCL)
  A6           ADC6        PD4
  A7           ADC7        PD5
*/

```

Jueff, 2020-04-28:
e.g. enable ATmega328 register emulation and add following code to fastpin_avr.h

```
#elif defined(AVR_NANO_4809_328MODE)

#define MAX_PIN 21
//_FL_DEFPIN(6, 4, F);

_FL_DEFPIN(0, 4, C); _FL_DEFPIN(1, 5, C); _FL_DEFPIN(2, 0, A); _FL_DEFPIN(3, 5, F);
_FL_DEFPIN(4, 6, C); _FL_DEFPIN(5, 2, B); _FL_DEFPIN(6, 4, F); _FL_DEFPIN(7, 1, A);
_FL_DEFPIN(8, 3, E); _FL_DEFPIN(9, 0, B); _FL_DEFPIN(10, 1, B); _FL_DEFPIN(11, 0, E);
_FL_DEFPIN(12, 1, E); _FL_DEFPIN(13, 2, E); _FL_DEFPIN(14, 3, D); _FL_DEFPIN(15, 2, D);
_FL_DEFPIN(16, 1, D); _FL_DEFPIN(17, 0, D); _FL_DEFPIN(18, 2, A); _FL_DEFPIN(19, 3, A);
_FL_DEFPIN(20, 4, D); _FL_DEFPIN(21, 5, D);

//#define SPI_DATA 11
//#define SPI_CLOCK 13
//#define SPI_SELECT 10
//#define AVR_HARDWARE_SPI 1
#define HAS_HARDWARE_PIN_SUPPORT 1

//#define SPI_UART0_DATA 1
//#define SPI_UART0_CLOCK 4

```

Godzila543, 2020-05-07:
It looks like you've figured it out Jueff, but I'm still having trouble. I haven't ever dealt with editing libraries before, and I don't understand where exactly to add the code you've given. Can you please tell me where to add the lines in fastpin_avr.h, clockless_trinket.h, and paltform.cpp?

Jueff, 2020-05-11:
The attached zip contains the four changed files mentioned above.

[FastLED.zip](https://github.com/FastLED/FastLED/files/4608545/FastLED.zip)

Install Release 3.3.3 and replace files with that ones in the zip.

jacopomazzoni, 2020-05-11:
> 
> 
> The attached zip contains the four changed files mentioned above.
> 
> [FastLED.zip](https://github.com/FastLED/FastLED/files/4608545/FastLED.zip)
> 
> Install Release 3.3.3 and replace files with that ones in the zip.
> 
> read article https://www.arduino.cc/en/Guide/Libraries for details on libraries/locations. Chapter "Manual installation" describes where libraries are stored.

Can this be merged in the official library? 
My students are novices and the easier the installation, the better.
Thanks <3

Jueff, 2020-05-11:
> Can this be merged in the official library?
> My students are novices and the easier the installation, the better.

I can't merge it, but please feel free to do so...

david-nc, 2020-06-21:
>and in platform.cpp add
> ```
> #if defined(__AVR_ATmega4809__)
>     volatile unsigned long timer0_millis = 0;
> #endif
> ```

The ATmega4809 uses timer_millis instead of timer0_millis, if you add that in the leds_sysdefs.h file then it is not necessary to make this modification.

Your modifications appear to work at 16MHz and 20MHz clock speeds, but not at 8MHz or below. 

Might be more logical to create a separate megaavr directory under platforms for the modified files, that way you can remove the code that is not applicable to the megaavr architecture, and not have to modify the files every time FastLED updates.  Only the #includes in platforms.h and led_sysdefs.h would need to be modified with each new release.

ngyl88, 2020-07-11:
> Can this be merged in the official library?
> My students are novices and the easier the installation, the better.
> Thanks <3

CalebMcDowell, 2020-07-15:
I've been experiencing an issue when working with an Arduino Nano Every and a WS2813 LED strip. I initially posted about the issue on FastLED's subreddit. The issue was traced down to the relationship between the Nano Every board and the modified FastLED library files I got from here. I'm posting to make you aware of the issue and also see if anyone has any ideas on how to solve the problem.

Thanks in advance.

FastLED Reddit post: https://www.reddit.com/r/FastLED/comments/hq09js/ws2813_and_arduino_nano_every_two_leds_light_for/

ngyl88, 2020-07-15:
> I've been experiencing an issue when working with an Arduino Nano Every and a WS2813 LED strip. I initially posted about the issue on FastLED's subreddit. The issue was traced down to the [...]

Hi @CalebMcDowell, I think the pin def for Arduino every might be different and not yet included in the fastpin_avr.h file. Have you try with adding a definition block for the board? The pin def from below might be helpful. Let me know if it helps. Thanks! 
https://github.com/FastLED/FastLED/blob/avrmega/platforms/avrmega/fastpin_avrmega.h

atait, 2020-10-25:
To add a data point, #1050 worked for me using a Nano Every and WS2811 chips. I developed my procedure on an Uno. Here is the error when compiling against a Nano Every.
```
.../Arduino/libraries/FastLED/fastpin.h:210:2: error: static assertion failed: Invalid pin specified
  static_assert(validpin(), "Invalid pin specified");
```
Then I checked out that branch (https://github.com/ngyl88-arduino/FastLED:master) and it worked. It warns about bit banging, but that's expected.

One caveat: It did not like WS2811 driver, so I had to change this line
```
FastLED.addLeds<WS2811, DATA_PIN, RGB>(leds, NUM_LEDS);
```
to this
```
FastLED.addLeds<WS2812, DATA_PIN, RGB>(leds, NUM_LEDS);
```
The WS2812 is the same serial interface, so its driver works just fine with WS2811 chips.

Thank you for that @ngyl88 and @Jueff!

dephillipsmichael, 2020-12-15:
First off, thank you @Jueff for the port - I verified it is working on my Arduino Wifi Rev 2 to control 350 WS2812B LEDs.

I also have an Arduino Uno Rev 3, chipset ATmega328P,  which I was originally using, until I wanted something I could control with a mobile app through BT.  

When comparing the two, **I noticed that FastLED.show() is less performant than the Wifi Rev 2 by a factor of about two**. 

Are there performance unit/integration tests that I can run to give more details/clarity?

I'm not really familiar with the performance expectations of these two when put against each other, but I figured it was something to bring up in this git discussion.

It may be that optimization is missing in the function FastLED.show() with regards to the Wifi Rev 2, or, it may just be that the Wifi/BLE module is consuming resources behind the scenes and this is the best performance we are going to get on that board.

Anyways, I wouldn't consider it a blocker, as the library works well on the Wifi Rev 2, and I'm happy to be able to move forward with my project!

ngyl88, 2021-09-05:
Hi @kriegsman , are you aware about any availability of unit/performance test that @dephillipsmichael requesting?

I think the performance and optimization could be treated as a separate issue?

RexCyrus530, 2023-04-23:
Hello, just checking in almost two years later. I downloaded the four edited files and swapped them for the ones in version 3.3.3, but I am still having issues. I can't get it to compile for the Uno WiFi Rev2. Has something changed since then, or am I doing something wrong?

ngyl88, 2023-04-23:
> Hello, just checking in almost two years later. I downloaded the four edited files and swapped them for the ones in version 3.3.3, but I am still having issues. I can't get it to compile for the Uno WiFi Rev2. Has something changed since then, or am I doing something wrong?

Hi @RexCyrus530 , please try with release 3.4.0 or the latest. The #1050 should be included in release 3.4.0, end of the year 2020.

RexCyrus530, 2023-04-23:
> Hi @RexCyrus530 , please try with release 3.4.0 or the latest. The #1050 should be included in release 3.4.0, end of the year 2020.

Ok, great. That worked for me. Thanks for the help.

ngyl88, 2023-04-27:
Hi @RexCyrus530 Thanks for reporting the issues with serial monitor. For your case, is the connection to the WS281x drivers good? Are you using these drivers, by chance? I'm referring something like the previous updates from  @dephillipsmichael on Dec 20.

I think we will need others to support the serial monitor interfaces. I'm suspecting but not sure if it related to the SPI support.

Just for a record for other contributors, if it's Arduino Uno WiFi Rev 2, the pin definition is here.
https://github.com/FastLED/FastLED/pull/1050/files#diff-fbacd0d9102add5bb88db1941999c09bbab3d6b94d502bff19f10ef37cb5bcbfR227

RexCyrus530, 2023-04-30:
> 

Yes, I am using the WS2812 drivers, and they're working just fine. I am also using the <SPI.h> library in the same sketch (for the nRF24L01 module), and it's working just fine too.

RexCyrus530, 2023-04-30:
> Hi @RexCyrus530 Thanks for reporting the issues with serial monitor. For your case, is the connection to the WS281x drivers good? Are you using these drivers, by chance? I'm referring something like the previous updates from @dephillipsmichael on Dec 20.
> 
> I think we will need others to support the serial monitor interfaces. I think it's related to the SPI support, which hasn't enabled. Even if it's enabled, we need someone to test if it's works magically just by uncommenting the relevant lines. Not sure if you are up for that exploration?
> 
> Just for a record for other contributors, if it's Arduino Uno WiFi Rev 2, the pin definition is here. https://github.com/FastLED/FastLED/pull/1050/files#diff-fbacd0d9102add5bb88db1941999c09bbab3d6b94d502bff19f10ef37cb5bcbfR227

Yes, I am using the WS2812 drivers, and they're working just fine. I am also using the <SPI.h> library in the same sketch (for the nRF24L01 module), and it's working just fine too.

ngyl88, 2024-09-18:
Hi @RexCyrus530  , sorry for the reply after a long while. I would need your help to confirm if [the issue "lost all serial communication capabilities with the new pin declarations" you reported on Apr 26, 2023](https://github.com/FastLED/FastLED/issues/716#issuecomment-1523666742) still exist with the latest release. If the issue still present, please also help us to confirm your setup. Thanks.
- Board :  Arduino Uno WiFi Rev 2
- FastLED release version
- other relevant info related to your issue, such as drivers / chipset you are using

I found that the previous version 3.4.0 doesn't define the SPI interface
- The SPI interface are added into ATmega4809 support, only starting release version 3.5.0. Ref: [changes in fastpin_avr.h](https://github.com/FastLED/FastLED/compare/3.4.0...3.7.7#diff-e2822923d46387a57fd370c8f3d4147ae42181f7fd048c93345aca315ff55107R256-R270)
- In latest release version 3.7.7, the pins `SPI_UART0_DATA` and `SPI_UART0_CLOCK` aren't confirmed and enabled yet, but it's likely not required for serial monitor output.
- FYI the support for ATmega4809 is added incrementally when there's a request / issue, as it powers multiple board.

> > > > Hello, just checking in almost two years later. I downloaded the four edited files and swapped them for the ones in version 3.3.3, but I am still having issues. I can't get it to compile for the Uno WiFi Rev2. Has something changed since then, or am I doing something wrong?
> > > 
> > > 
> > > Hi @RexCyrus530 , please try with release 3.4.0 or the latest. The #1050 should be included in release 3.4.0, end of the year 2020.
> > 
> > 
> > Ok, great. That worked for me. Thanks for the help.
> 
> Just for your reference: it seems that I have lost all serial communication capabilities with the new pin declarations. The code compiles just fine, but the serial monitor always fails to open.