FastLED/FastLED #717: FastLed

Horvat2003, 2019-01-14:
Hello,
I am using WS2811 leds 
I have a problem with fastled library.
I am using Arduino Uno and when i program my leds to be red they glow green, and when i type green they glow red, the blue is okay.
Why is it not showing right colors?

chrisparton1991, 2019-01-14:
Hi @Horvat2003, your problem is that you haven't specified the correct color ordering for your LED strip.

You can follow the steps at https://github.com/FastLED/FastLED/wiki/Rgb-calibration to determine the correct ordering for your strip.

Questions like these are best asked at the FastLED community page, which you can find at https://fastled.io/+

Good luck!

Horvat2003, 2019-01-15:
@chrisparton1991 These are RGB programable led lights, you can specify which color you want when programing, when i type red in a program they glow green and when i type green they glow red. Blue is okay

JonasVorwerk, 2019-01-15:
You might find your answer here: https://github.com/FastLED/FastLED/wiki/Rgb-calibration