FastLED/FastLED #718: Release 3.2.2?

ToBeReplaced, 2019-01-18:
Would the maintainers be willing to cut a 3.2.2 release at this point? There have been a number of improvements, and I would like to peg a version that has support for global brightness on the APA102.

Thank you for all of your hard work.

focalintent, 2019-01-20:
It'll go up soon - asking for a release doesn't need an issue itself - and it's always possible to manually download the library : )

closed 2019-01-20 by focalintent

ToBeReplaced, 2019-01-20:
No problem. I checked for information on a release schedule and couldn't find any, so I figured I'd just ask. So that I (and onlookers) know, what was the appropriate venue to ask? Google+? 

Yes, we currently use a downloaded version that's patched to include another version number in the meantime.