FastLED/FastLED #719: ESP32 corrupted data within rmt device

Crejan, 2019-01-18:
Hi
i have a problem with my ESP32 controlled WS2812 LEDS. As long as I do not use Bluetooth or WiFi everything is fine. But after enabling Bluetooth or connecting WiFi to an existing network (AP seems to work fine) Strange things begin to happen in the RMT subsystem.

The problem is a color flickering starting from one (random) LED in the stripe and affecting all following ones appearing about once or twice a second.

First I thought that the problem is due to interruptions by the WiFi or Bluetooth communication but nothing so I tried the following:
- Disable Interrupts: FASTLED_ALLOW_INTERRUPTS 0
- Only using one RMT channel: FASTLED_RMT_MAX_CHANNELS 1
- Remove every other thread from Core 1 (by modifying freeRTOS) xTaskCreatePinnedToCore and xTaskCreate 

Nothing helped. So I  tried to diagnose the problem by only sending static data to the LEDS and connecting an Oscilloscope to the data channel so every frame should look the same:
The first result was that the "bad" frames are a little bit longer than expected. and the content was different too. So it is no issue of the LEDs itself. I expected to see the frame tear appart at some point because of an interrupt disturbing the software during RMT buffer fillup but that was not the case. In fact the data itself seems to be corrupted. I achieved to get an image of a frame that was damaged it the very beginning so it was easy to compare it to a valid frame and found out that after 32 bits a zero byte got inserted into the frame. This inserted byte shifts all following bytes by 8 and therefore makes the LEDs show strange things afterwards.

You can find the Image and csv data from the oscilloscope here:
https://drive.google.com/open?id=1ESd-eMWhegF0WsEPzl21v0lOIQfqUmc_

I also tried to replace the rmt code of esp by a very simplified code which causes the same issues.

```
int cur = 0;
        while (pixels.has(1)) {
            uint8_t byte = 0x04;
            //uint8_t byte = pixels.loadAndScale0();
            led_data[cur++] = byte & 0x80 ? mOne : mZero;
            led_data[cur++] = byte & 0x40 ? mOne : mZero;
            led_data[cur++] = byte & 0x20 ? mOne : mZero;
            led_data[cur++] = byte & 0x10 ? mOne : mZero;
            led_data[cur++] = byte & 0x08 ? mOne : mZero;
            led_data[cur++] = byte & 0x04 ? mOne : mZero;
            led_data[cur++] = byte & 0x02 ? mOne : mZero;
            led_data[cur++] = byte & 0x01 ? mOne : mZero;
            //byte = pixels.loadAndScale1();
            led_data[cur++] = byte & 0x80 ? mOne : mZero;
            led_data[cur++] = byte & 0x40 ? mOne : mZero;
            led_data[cur++] = byte & 0x20 ? mOne : mZero;
            led_data[cur++] = byte & 0x10 ? mOne : mZero;
            led_data[cur++] = byte & 0x08 ? mOne : mZero;
            led_data[cur++] = byte & 0x04 ? mOne : mZero;
            led_data[cur++] = byte & 0x02 ? mOne : mZero;
            led_data[cur++] = byte & 0x01 ? mOne : mZero;
            //byte = pixels.loadAndScale2();
            led_data[cur++] = byte & 0x80 ? mOne : mZero;
            led_data[cur++] = byte & 0x40 ? mOne : mZero;
            led_data[cur++] = byte & 0x20 ? mOne : mZero;
            led_data[cur++] = byte & 0x10 ? mOne : mZero;
            led_data[cur++] = byte & 0x08 ? mOne : mZero;
            led_data[cur++] = byte & 0x04 ? mOne : mZero;
            led_data[cur++] = byte & 0x02 ? mOne : mZero;
            led_data[cur++] = byte & 0x01 ? mOne : mZero;
            pixels.advanceData();
            pixels.stepDithering();
        }
        led_data[cur++] = mReset;

        rmtWrite(rmt_send, led_data, cur);
```

I am out of ideas how to solve this.I hope anyone here can help me?

Best regards
Jan

focalintent, 2019-01-18:
The drive url you gave doesn’t work.  

Also while it appears to be either an extra 16 bits was inserted - alternatively, it could be a byte was dropped - you’re sending the same 3 bytes repeatedly, so both things happening would look the same.  

@samguyer - thoughts?

Crejan, 2019-01-19:
Then the system crashes right away. I tried this too before all of my modifications but if I remember right the system crashed immediately with this option.

Crejan, 2019-01-19:
I updated the link in the post above. I am sure that bytes get inserted, because the total signal for one frame is longer too in the error case.

focalintent, 2019-01-19:
Ok - I think what might be happening is the interrupt to refill the rmt buffer may be getting handled too late - causing a chunk of data to get written out again.  @samguyer - do you know why the rmt builtin driver was crashing?  Also the rest of the code appears to assume that the interrupt will be handled and doesn’t appear to have any checks to verify that it was.

Crejan, 2019-01-21:
I changed the code of the clockless_esp32.h to use SPI for the signal generation: 
```

template<int DATA_PIN, int T1, int T2, int T3, EOrder RGB_ORDER = RGB, int XTRA0 = 0, bool FLIP = false, int WAIT_TIME = 5>
class ClocklessController : public CPixelLEDController<RGB_ORDER> {

    SPIClass * hspi = nullptr;
    uint8_t *led_data = nullptr;
    // H:875 L:375
    static const uint8_t mOne = 0xf8;
    // H:250 L:1000
    static const uint8_t mZero = 0x80;
    int mSize = 0;

public:

    void init() {
        hspi = new SPIClass(HSPI);
        hspi->begin();
    }

    virtual uint16_t getMaxRefreshRate() const {
        return 400;
    }

protected:

    virtual void showPixels(PixelController<RGB_ORDER> &pixels) { ;
        const int RESET_LENGTH = 10;
        int size_needed = pixels.size() * 3 * 8  + RESET_LENGTH;
        if (size_needed != mSize) {
            if (led_data != nullptr)
                free(led_data);
            mSize = size_needed;
            led_data = (uint8_t*) malloc(mSize * sizeof(rmt_data_t));
            for(int i=size_needed-RESET_LENGTH; i<size_needed; ++i){
                led_data[i] = 0;
            }
        }

        int cur = 0;
        while (pixels.has(1)) {
            uint8_t byte = pixels.loadAndScale0();
            led_data[cur++] = byte & 0x80 ? mOne : mZero;
            led_data[cur++] = byte & 0x40 ? mOne : mZero;
            led_data[cur++] = byte & 0x20 ? mOne : mZero;
            led_data[cur++] = byte & 0x10 ? mOne : mZero;
            led_data[cur++] = byte & 0x08 ? mOne : mZero;
            led_data[cur++] = byte & 0x04 ? mOne : mZero;
            led_data[cur++] = byte & 0x02 ? mOne : mZero;
            led_data[cur++] = byte & 0x01 ? mOne : mZero;
            byte = pixels.loadAndScale1();
            led_data[cur++] = byte & 0x80 ? mOne : mZero;
            led_data[cur++] = byte & 0x40 ? mOne : mZero;
            led_data[cur++] = byte & 0x20 ? mOne : mZero;
            led_data[cur++] = byte & 0x10 ? mOne : mZero;
            led_data[cur++] = byte & 0x08 ? mOne : mZero;
            led_data[cur++] = byte & 0x04 ? mOne : mZero;
            led_data[cur++] = byte & 0x02 ? mOne : mZero;
            led_data[cur++] = byte & 0x01 ? mOne : mZero;
            byte = pixels.loadAndScale2();
            led_data[cur++] = byte & 0x80 ? mOne : mZero;
            led_data[cur++] = byte & 0x40 ? mOne : mZero;
            led_data[cur++] = byte & 0x20 ? mOne : mZero;
            led_data[cur++] = byte & 0x10 ? mOne : mZero;
            led_data[cur++] = byte & 0x08 ? mOne : mZero;
            led_data[cur++] = byte & 0x04 ? mOne : mZero;
            led_data[cur++] = byte & 0x02 ? mOne : mZero;
            led_data[cur++] = byte & 0x01 ? mOne : mZero;
            pixels.advanceData();
            pixels.stepDithering();
        }
        const SPISettings settings(6400000, MSBFIRST, SPI_MODE0);
        hspi->beginTransaction(settings);
        hspi->writeBytes(led_data, cur);
        hspi->endTransaction();
    }

};
```

This solves the problem for me. But I still think, that there is a bug inside the RMT code somewhere

moose4lord, 2019-03-02:
I think I'm seeing this problem as well. I "dumbed down" @samguyer's DemoReelESP32 example sketch to get a simple animation running. That works fine. But when I introduce WiFi into the sketch, the LEDs will intermittently flash as if extra data is being delivered to the LED strip. Just initializing WiFI seems to cause the problem.

Here is my minimal test sketch:
[DemoReelESP32_WiFi.zip](https://github.com/FastLED/FastLED/files/2922365/DemoReelESP32_WiFi.zip)
Note, my hardware setup has more than the 30 LEDs defined in the sketch, and with WiFi initialized,  more than 30 LEDs illuminate. I think that's telling me extra pulses are being delivered to the LED strip. Simply commenting out the call to setupWifi() causes the problem to go away.

I hate to punt and drop back to SPI when the RMT solution offers so much.

samguyer, 2019-03-02:
OK, clearly I need to get cracking on this problem. I will have time this weekend. Is there anything special about the WiFi setup? Do I need to actually connect and transmit, or is it enough to just start the service?

moose4lord, 2019-03-02:
I just started the WiFi service and connected to my local wifi network.
```c++
void setupWifi(void) {
  WiFi.mode(WIFI_STA);
  WiFi.begin(WIFI_SSID, WIFI_PASSWORD);

  // Wait for connection
  Serial.print("Connecting to WiFi");
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());
}
```
Pretty standard stuff. Once started, the wifi is just idle. Didn't need to actually make a connection/request to the ESP32 to see the problem. If you try my test sketch, you'll need to change the WIFI_SSID and WIFI_PASSWORD parameters for your local network.

samguyer, 2019-03-02:
Wait, you're not going to give me your network ID and password?!?!? ;-)
OK, I will check this out this evening or tomorrow.

samguyer, 2019-03-03:
@moose4lord  I think the issue in your example is actually the call to FastLED.delay. The problem is that FastLED.delay calls FastLED.show, which interacts in some bad way with the show running on the other core. I can't figure out (yet) what the nature of that interaction is, because it shouldn't matter. But I suspect it has something to do with the delivery of interrupts in a multicore chip -- basically, which core gets the RMT interrupt. 

I recommend using built-in delay instead of FastLED delay. That fixed the problem for me. Another solutions that seems to work is to just run FastLED on the main core (and dispense with all of the tasks nonsense).

focalintent, 2019-03-03:
Sam - FastLED.delay() calls show to drive dithering - so “just don’t use FastLED.delay()” is only a short term fix :)

moose4lord, 2019-03-03:
ok, I replaced the FastLED.delay with a vanilla delay and that seemed to fix it. Until I increased the number of LEDs in my animation. With 30 LEDs it works fine, but if I increase it to 60 or 100 I start to see the same kind of flickering and "extra illuminated LEDs at the end" symptoms. My test strip is a meter of 144 LEDs, so extra pulses on the ws2812b Din pin show up as extra LEDs being lit.  It seems to happen much less often, but it still occasionally misbehaves.

I'll try "dispensing with the tasks nonsense" (your phrasing made me laugh :), although I kinda like separating the LED process from the WiFi process. That's always been my problem with the ESP8266, driving lots of LEDs always seems to muck up the WIFi.

samguyer, 2019-03-03:
@focalintent @moose4lord I'm wondering now if what's happening is that I'm not waiting properly for the RMT send process to finish, so that when you call show() again, or too quickly, the two shows end up overlapping. That might explain the extra bits at the end of the strip. I'll take a look at the synchronization -- it's somewhat gnarly, so maybe I messed it up. Or maybe I just need an extra interlock to prevent overlapping shows.

moose4lord, 2019-03-03:
I have to take it back. Even with 30 LEDs the problem occurs, just less often when using the vanilla delay.

I changed the code that sets the led colors from this
```c++
  for(int i=0; i<NUM_LEDS; i++) { // moving block of four green leds
    if(i >= pos && i < (pos + 4)) {
      leds[i].green = 255;
    } else {
      leds[i].green = 0;
    }
  }
```
to this
```c++
  for(int i=0; i<NUM_LEDS; i++) { // moving block of four green leds
    if(i >= pos && i < (pos + 4)) {
      leds[i].green = 255;
      leds[i].blue = 0;
    } else {
      leds[i].green = 0;
      leds[i].blue = 16;
    }
  }
```
so that all leds get some color data. And changed NUM_LEDS to 60 and FRAMES_PER_SECOND to 60, so the strip is updated more often. Those changes bring out the problem pretty easily even with the vanilla delay.

blackketter, 2019-06-23:
Just curious if anybody has found a fix for this problem?  I'm seeing intermittent flashing too, though it seems to come and go (and can be temporarily fixed by rebooting).

samguyer, 2019-06-23:
@blackketter Can you tell me more about your setup?

blackketter, 2019-06-26:
Thanks, sorry for the delay getting back.

It’s an ESP32 driving with WS2811 on 6 outputs driving 120 LEDs each.  Flickering and flashing on a couple of pins, seems worse on one of them.  It seems to come and go and I haven’t been able to capture it on a scope.  

Pixels flash in the middle of the string and the patterns of errant pixels seem to be repetitive.

I have a second hardware configuration with the same number of pins but fewer LEDs per strip.  I have seen rare flashes there, but less frequently.  Same code base.

seymar, 2019-09-01:
After rewriting/optimizing the driver for my own application I think the problem is that the `fillHalfRMTBuffer` function called by the interrupt takes too much time when you are using multiple channels, this does not leave enough room for other things (Wi-Fi stuff).

samguyer, 2019-09-02:
@seymar Have you tried the current head? We did some work on fillHalfRMTBuffer that should speed it up quite a lot. My pull request was merged into the main branch about 2 weeks ago.

BatsIhor, 2020-09-22:
I still have the same issue. I see issue is open just want to make sure it's get visible as for a long time no one was working on it.

jacob-jarick, 2020-11-24:
I came across this issue while researching. 
My project uses 8-16 parallel strips of 144 pixels each.

Using the latest arduino release 3.3.3

If I enable FASTLED_RMT_BUILTIN_DRIVER like so

`#define FASTLED_RMT_BUILTIN_DRIVER 1
#include "FastLED.h"`

My code immediately crashes, to assist debugging here is the stack trace and decoded messages.

**Stack trace:**

```
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5816
entry 0x400806ac
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400d2611  PS      : 0x00060031  A0      : 0x80085bcd  A1      : 0x3ffbe790  
A2      : 0x00000000  A3      : 0x3ffc3448  A4      : 0x00000000  A5      : 0x00000000  
A6      : 0x3ffbf400  A7      : 0x00060923  A8      : 0x8008f084  A9      : 0x3ffbe790  
A10     : 0x00000003  A11     : 0x00060023  A12     : 0x00060021  A13     : 0x00000000  
A14     : 0x00060023  A15     : 0x00000000  SAR     : 0x00000017  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x0000001c  LBEG    : 0x4014c425  LEND    : 0x4014c43e  LCOUNT  : 0x00000000  
Core 1 was running in ISR context:
EPC1    : 0x400d2611  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x400879d9

Backtrace: 0x400d2611:0x3ffbe790 0x40085bca:0x3ffbe7c0 0x40085355:0x3ffbe7f0 0x4000bfed:0x3ffcb8f0 0x40090539:0x3ffcb900 0x4008f1c8:0x3ffcb920 0x400d2f92:0x3ffcb960 0x400d3d35:0x3ffcb980 0x400d463d:0x3ffcb9c0 0x400d2296:0x3ffcba00 0x400d6111:0x3ffcba40 0x4008f3bd:0x3ffcba60
```

**Decoded Stack Trace:**
```

`PC: 0x400d2611: ClocklessController15, 60, 150, 90, (EOrder)10, 0, false, 5>::doneOnChannel(rmt_channel_t, void*) at C:\Users\mem\Documents\Arduino\libraries\FastLED/platforms/esp/32/clockless_rmt_esp32.h line 449
EXCVADDR: 0x0000001c

Decoding stack results

0x400d2611: ClocklessController15, 60, 150, 90, (EOrder)10, 0, false, 5>::doneOnChannel(rmt_channel_t, void*) at C:\Users\mem\Documents\Arduino\libraries\FastLED/platforms/esp/32/clockless_rmt_esp32.h line 449
0x40085bca: rmt_driver_isr_default at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/driver/rmt.c line 581
0x40090539: vTaskExitCritical at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/tasks.c line 4274
0x4008f1c8: xQueueGenericReceive at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/queue.c line 1591
0x400d2f92: ClocklessController15, 60, 150, 90, (EOrder)10, 0, false, 5>::showPixels(PixelController(EOrder)10, 1, 4294967295u>&) at C:\Users\mem\Documents\Arduino\libraries\FastLED/platforms/esp/32/clockless_rmt_esp32.h line 333
0x400d3d35: CPixelLEDController(EOrder)10, 1, 4294967295u>::show(CRGB const*, int, CRGB) at C:\Users\mem\Documents\Arduino\libraries\FastLED/controller.h line 408
0x400d463d: CFastLED::show(unsigned char) at C:\Users\mem\Documents\Arduino\libraries\FastLED/controller.h line 90
0x400d2296: loop() at C:\Users\mem\Documents\Arduino\libraries\FastLED/FastLED.h line 500
0x400d6111: loopTask(void*) at C:\Users\mem\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\cores\esp32\main.cpp line 19
0x4008f3bd: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143

```

Testing with the latest git version I also recieve the same error.

If i can assist with further debugging please let me know.

Thanks

IVData, 2021-01-08:
Same issue here on 3.4, FASTLED_RMT_BUILTIN_DRIVER causes crash as others have said.

MarkJeronimus, 2022-04-25:
I'm having the same problem with 73 LEDs (haven't tried with less) on IDF 4.4.

I had to look up what FastLED is, and it looks like this library does stuff similar to how I do it. My implementation is derived from https://github.com/espressif/esp-idf/blob/master/examples/peripherals/rmt/led_strip/main/led_strip_main.c and it looks like FastLED with `FASTLED_RMT_BUILTIN_DRIVER 1`.

I have this running in a task pinned to core 0, Wi-Fi is **OFF**, but I do lots of other stuff in the background including SPI and UART. The glitches appear about every 1 second (slightly faster) which doesn't align with anything i do.

As others said, when the glitch happens, it sends more RGB data than specified, but it's always aligned a multiple of 8. I counted 2592 bits sent out in a correct frame and 3424 bits in a corrupt frame. looks like it restarts sending half-way but I haven't confirmed. At the very least, the last 2592 bits of a corrupt frame are identical to the beginning of a good frame. I did toggle a GPIO every time I start sending data, and this nicely lines up with the start of each RMT frame (there are no GPIO events half-way an RMT frame). The frames take about 5.5ms, and I send every 10ms (100Hz).

samguyer, 2022-04-25:
@MarkJeronimus I have not kept up the code for the built-in driver, but I should look again. The big issue, from my standpoint, is memory use. For larger installations, converting all the bits to RMT items requires 32X the memory. I'm also unsure on the degree to which is supports parallel output. I looked briefly at the Neopixel library for ESP32 from AdaFruit, and they use a callback that translates blocks of bits to RMT items on the fly.

BUT I might be able to fix the retry code to work better. It might just need a short delay (50 microseconds) to allow the LEDs to latch.

MarkJeronimus, 2022-05-02:
I've implemented the SPI method in "native" IDF and it works, at the expense of the VSPI pripheral that I liked to have reserved for future additions.

Used constants (note that this is not for WS2812B, but for Hongbright RGB LEDs): 
```C
static const uint8_t mZero = 0b11000000; // 300:900 ns
static const uint8_t mOne  = 0b11111100; // 900:300 ns
static const uint8_t mIdle = 0b00000000; // >=80000ns (send 67x)
```

Initialize with:
```C
    int numBits = numStatusLEDs * 24 + 67;
    bitVector.reserve(numBits);
    for (size_t i = 0; i < numBits; i++) {
        bitVector.push_back(mIdle);
    }

    spi_bus_config_t buscfg = {
            .mosi_io_num     = dataGPIO,
            .miso_io_num     = GPIO_NUM_NC,
            .sclk_io_num     = GPIO_NUM_NC,
            .quadwp_io_num   = GPIO_NUM_NC,
            .quadhd_io_num   = GPIO_NUM_NC,
            .data4_io_num    = GPIO_NUM_NC,
            .data5_io_num    = GPIO_NUM_NC,
            .data6_io_num    = GPIO_NUM_NC,
            .data7_io_num    = GPIO_NUM_NC,
            .max_transfer_sz = numBits, // 0 = Default
            .flags           = 0, // SPICOMMON_BUSFLAG_*
            .intr_flags      = 0 };

    ESP_ERROR_CHECK(spi_bus_initialize(VSPI_HOST, &buscfg, SPI_DMA_CH_AUTO));

    const spi_device_interface_config_t devcfg = {
            .command_bits     = 0,
            .address_bits     = 0,
            .dummy_bits       = 0,
            .mode             = 1,
            .duty_cycle_pos   = 0,
            .cs_ena_pretrans  = 0,
            .cs_ena_posttrans = 16,
            .clock_speed_hz   = 4000000, 
            .input_delay_ns   = 0,
            .spics_io_num     = GPIO_NUM_NC,
            .flags            = SPI_DEVICE_NO_DUMMY,
            .queue_size       = 1,
            .pre_cb           = nullptr,
            .post_cb          = nullptr };

    ESP_ERROR_CHECK(spi_bus_add_device(VSPI_HOST, &devcfg, &spiHandle));
```

send with this:
```C
void setPixel(size_t index, rgb888_t color) {
    index *= 24;
    uint8_t value;

    value = color.g;
    bitVector[index++] = value & 0x80 ? mOne : mZero;
    bitVector[index++] = value & 0x40 ? mOne : mZero;
    bitVector[index++] = value & 0x20 ? mOne : mZero;
    bitVector[index++] = value & 0x10 ? mOne : mZero;
    bitVector[index++] = value & 0x08 ? mOne : mZero;
    bitVector[index++] = value & 0x04 ? mOne : mZero;
    bitVector[index++] = value & 0x02 ? mOne : mZero;
    bitVector[index++] = value & 0x01 ? mOne : mZero;
    value = color.r;
    bitVector[index++] = value & 0x80 ? mOne : mZero;
    bitVector[index++] = value & 0x40 ? mOne : mZero;
    bitVector[index++] = value & 0x20 ? mOne : mZero;
    bitVector[index++] = value & 0x10 ? mOne : mZero;
    bitVector[index++] = value & 0x08 ? mOne : mZero;
    bitVector[index++] = value & 0x04 ? mOne : mZero;
    bitVector[index++] = value & 0x02 ? mOne : mZero;
    bitVector[index++] = value & 0x01 ? mOne : mZero;
    value = color.b;
    bitVector[index++] = value & 0x80 ? mOne : mZero;
    bitVector[index++] = value & 0x40 ? mOne : mZero;
    bitVector[index++] = value & 0x20 ? mOne : mZero;
    bitVector[index++] = value & 0x10 ? mOne : mZero;
    bitVector[index++] = value & 0x08 ? mOne : mZero;
    bitVector[index++] = value & 0x04 ? mOne : mZero;
    bitVector[index++] = value & 0x02 ? mOne : mZero;
    bitVector[index++] = value & 0x01 ? mOne : mZero;

    dirty = true;
}

void send() {
    // TODO Bus Acquiring?

    spi_transaction_t trans;
    memset(&trans, 0, sizeof(spi_transaction_t));
    trans.flags     = 0;
    trans.tx_buffer = bitVector.data();
    trans.length    = bitVector.size() * 8; // Number of bits, ot bytes!

    esp_err_t err = spi_device_queue_trans(spiHandle, &trans, 0);
    if (err != ESP_OK) {
        WARN("Error sending SPI: %d", err);
    }

    dirty = false;
}
```

Thanks to [Crejan](https://github.com/Crejan) for part of the code.