FastLED/FastLED #720: Errors with FastLED Delay.h

petzeaj, 2019-01-18:
I am a rookie when it comes to this stuff, but when I am trying ot use arduino to upload code to my ESP8266, it gives me several issues with the file mentioned in the title. Can anyone help me fix it? 
[image]

focalintent, 2019-01-18:
What code are you trying to compile?

petzeaj, 2019-01-19:
Its for sound reactive lighting Heres the link. https://github.com/hansjny/Natural-Nerd/blob/master/SoundReactive2/led_master.ino

focalintent, 2019-01-19:
Can you copy the entire text from the build window (it might be a lot of text - paste it into gist.github.com and share the link here) - the screenshot you shared only shows the end of the set of errors, they’re most likely getting caused by something reported further up in the window.

focalintent, 2019-01-20:
I was able to have this code compile here by changing the line:

```
#define BUTTON_PIN D1
```

to 

```
#define BUTTON_PIN 1
```

but even before fixing that, I wasn't seeing the compile errors that you are seeing.  Do you have the most recent versions of the library, Arduino, and esp8266 platform?

petzeaj, 2019-01-21:
Here is the link to the GIST that you asked for. I think I provided what you asked. But with the change in the Button-Pin D1, I think that needs to remain the same because of the controller units wiring with the actual button. (See the second link) https://gist.github.com/petzeaj/c0e6d13c31ca7c89ad646d901519c137
https://drive.google.com/file/d/1PRDaSGIENCrb3MHXDk6Qwi3e8gnxunyV/edit

petzeaj, 2019-01-21:
Yes, everything is because I downloaded them a couple of days ago for the first time updated in terms of library, Arduino,etc

focalintent, 2019-01-21:
So this first error:

```
In file included from /Users/Petze/Documents/Arduino/libraries/libraries/FastLED/FastLED.h:44:0,
                 from /Users/Petze/Documents/Arduino/libraries/Controller/Controller.ino:2:
/Users/Petze/Documents/Arduino/libraries/libraries/FastLED/fastled_delay.h:13:1: error: a template declaration cannot appear at block scope
 template<int WAIT> class CMinWait {
```

What is this Controller library, and I need to see this Controller.ino file - because it's when trying to compile that where your error is occurring - and it is claiming that FastLED.h is being included from. line 2 of controller.ino - where FastLED.h. is. included. in line 1 here - https://github.com/hansjny/Natural-Nerd/blob/master/SoundReactive2/led_master.ino

You might have a random/stray curly brace somewhere ( { ) - which is confusing the compiler.  Also - it's strange that it's compiling a .ino file for a library.  Did you rename led_master.ino to Controller.ino?  (As an aside, projects generally shouldn't live underneath libraries - that folder is where you put library code that other projects are going to use) - if you didn't - then where did you get that Controller.ino file from?

> But with the change in the Button-Pin D1, I think that needs to remain the same because of the controller units wiring with the actual button.

Some platforms/targets will make. defines for pins that start with a D - but not all of them do - it sounds like some esp boards do, and some don't.

focalintent, 2019-01-21:
Ah -- here's the problem -- from the first set of compiler messages:

```
In file included from /Users/Petze/Documents/Arduino/libraries/Controller/Controller.ino:2:0:
/Users/Petze/Documents/Arduino/libraries/libraries/FastLED/FastLED.h: In function 'void setup()':
/Users/Petze/Documents/Arduino/libraries/libraries/FastLED/FastLED.h:14:21: note: #pragma message: FastLED version 3.002.001
 #    pragma message "FastLED version 3.002.001"
                     ^
In file included from /Users/Petze/Documents/Arduino/libraries/libraries/FastLED/FastLED.h:44:0,
                 from /Users/Petze/Documents/Arduino/libraries/Controller/Controller.ino:2:
/Users/Petze/Documents/Arduino/libraries/libraries/FastLED/fastled_delay.h:13:1: error: a template declaration cannot appear at block scope
 template<int WAIT> class CMinWait {
```

see that line "In function 'void setup()'?  It looks like Controller.ino might begin with something like this:

```
void setup() { 
#include <FastLED.h>
```

and if so - you don't want to do that - the ```#include``` statements need to be _outside_ of the body of any functions.

petzeaj, 2019-01-22:
I see what you're talking about, but I'm not sure what how to go about your solution (in terms of how to fix the actual code)

petzeaj, 2019-01-22:
I fixed the issue there, but now I am getting a new error (Link Bellow). Also before this error, I kept getting different errors with the { , saying they can't have a void statement before them (so I moved the brackets before the void functions). 

https://gist.github.com/petzeaj/2e92c45989635b6e7c5fe72875be13ab

focalintent, 2019-01-22:
Can you put the exact code that you are trying to compile into gist.github.com - it’ll be easier to see/explain what’s actually causing the syntax errors the compiler is giving you.

petzeaj, 2019-01-22:
Yes I can, Here is the link. https://gist.github.com/petzeaj/27ad899a2c1842a15eea9aee178c4504

mkriegsman, 2019-01-22:
There are a number of syntax problems with the code in that gist.
First, delete line 6 entirely, the first one that says "void setup() {"
Then, you need to fix several of the function definitions.  In C/C++, the correct way to define a function is (for example) 

`void functionName( args) {
`
not like this:

`{ void functionName( args) 
`

It appears the wrong way in several of the functions in this file; they all need to be corrected, eg lines 34, 56, 85, 99.

Finally, this code defines the function "loop" twice.  Delete the second, empty definition from lines 181-184, and see if that helps, too.

kriegsman, 2019-01-22:
There are a number of syntax problems with the code in that gist.

First, delete line 6 entirely, the first one that says "void setup() {"; that's just a mistake there, I think.

Second, you need to fix the syntax of several of the function definitions.  In C/C++, the correct way to define a function is like this:

`void functionName( args) {
`
not like this:

`{ void functionName( args) 
`

The syntax appears the wrong way in several of the functions in this file; they all need to be corrected, eg lines 34, 56, 85, 99.

Finally, this code defines the function "loop" twice.  Delete the second, empty definition from lines 181-184, and see if that helps, too.

(Sorry for the comment spam- I was logged in with the wrong account before; fixed now!)

kriegsman, 2019-01-22:
Also just as a side note, when there are bugs with the FastLED library itself, this is the right place to file the bug tickets.  For general help getting individual code and projects like this one working, please ask over on the general FastLED discussion group; there are _lots_ more people there, and lots more help available.  Here's a shortcut: http://fastled.io/+ . Thanks!

tobi01001, 2019-01-22:
I would encourage anyone being new to programming for arduino to take an online course for C and C++at e.g https://www.sololearn.com
It's actually free and provides you with the basics of programming in C and C++. 

If you then work your way through the arduino and fastled examples line by line it will be much easier to understand and follow what's going on there... 

Have fun with (creating) your sound reactive LEDs. For me the fun is in creating that stuff not actually using it...

petzeaj, 2019-01-22:
> There are a number of syntax problems with the code in that gist.
>
> First, delete line 6 entirely, the first one that says "void setup() {"; that's just a mistake there, I think.
>
> Second, you need to [...]

Thank you for your input, The reason the void lines have the bracket before is that I kept getting an error that said Void function cannot be before a {. So when I did this to the Void lines, then the errors went away once I moved the brackets.

kriegsman, 2019-01-22:
I suspect that the error you just described is because of the erroneous line 6, which just needs to be deleted first, to then give clear and meaningful errors for the rest of the file.

focalintent, 2019-01-22:
Generally speaking - whenever you have compiler errors - you should start from fixing the _first_ errors that come up, because depending on how bad those syntax errors are, they can create more and weirder compile errors later in the file.  If you try to fix compiler errors by working backwards, you're going to end up chasing your tail and not actually fixing real problems, just causing more for yourself.

closed 2019-08-12 by focalintent