FastLED/FastLED #725: Does not work with Wavgat Arduino Nano

remenikomer, 2019-01-21:
Does not work with Wavgat Arduino Nano v.3. There are a lot of errors when compiling. Other libraries are compile (for example, NeoPixel).
Arduino Nano: https://www.aliexpress.com/item/WAVGAT-Nano-Mini-USB-With-the-bootloader-compatible-Nano-3-0-controller-CH340-USB-driver-16Mhz/32868170637.html
Errors  for example :

> In file included from C:\Arduino\libraries\FastLED\platforms/avr/fastled_avr.h:6:0,
> 
>                  from C:\Arduino\libraries\FastLED\platforms.h:32,
> 
>                  from C:\Arduino\libraries\FastLED\FastLED.h:52,
> 
>                  from C:\Arduino\libraries\FastLED\FastLED.cpp:2:
> 
> C:\Arduino\libraries\FastLED\platforms/avr/clockless_trinket.h:76:0: warning: "D3" redefined
> 
>  #define D3(ADJ) DINT(T3,ADJ)
> 
>  ^
> 
> In file included from C:\Arduino\hardware\WAV\avr\variants\lgt8fx8e/pins_arduino.h:37:0,
> 
>                  from C:\Arduino\hardware\WAV\avr\cores\lgt8f/Arduino.h:225,
> 
>                  from C:\Arduino\libraries\FastLED\led_sysdefs.h:43,
> 
>                  from C:\Arduino\libraries\FastLED\FastLED.h:41,
> 
>                  from C:\Arduino\libraries\FastLED\FastLED.cpp:2:
> 
> c:\arduino\hardware\wav\avr\variants\standard\pins_arduino.h:84:0: note: this is the location of the previous definition
> 
>  #define D3 3 /* PD3 */
> 
>  ^
> 
> C:\Arduino\libraries\FastLED\FastLED.cpp: In member function 'void CFastLED::delay(long unsigned int)':
> 
> C:\Arduino\libraries\FastLED\FastLED.cpp:132:9: error: 'yield' was not declared in this scope
> 
>    yield();

focalintent, 2019-01-21:
What version of Arduino are you using and what platform are you selecting in the drop down in Arduino when you are compiling?  It looks like the WAVGET platform may not be properly defining the yield function, which most other Arduino avr platforms defined a while ago.  For now, try adding the following line into your code:

```
void yield(void) {}
```

which should force a definition of the yield function for FastLED to use, until I can figure out how they've broken their platform definition  (FastLED does a lot more lower level work than a lot of other libraries, which is likely why their platform definition is having problems with it).

focalintent, 2019-01-22:
Get the most recent version of the library, 3.2.4 - I've checked in a fix for building for these boards.

closed 2019-01-22 by focalintent

remenikomer, 2019-01-22:
> What version of Arduino are you using and what platform are you selecting in the drop down in Arduino when you are compiling? It looks like the WAVGET platform may not be properly defining the yield [...]

Library version updated.
Installed update IDE for wavgat (platform - wavgat arduino Nano) by the motherboard manufacturer: https://drive.google.com/file/d/10gwrG9uTDwaEO-7EudsmBkfgdcyrcABI/view?usp=drive_open

Platform: https://yadi.sk/i/bKIDWeivmCp7Vw

`void yield(void) {}`
Function definition did not help.

Here is the entire text of the collection with errors: https://yadi.sk/i/cZ4nCGLV4xyIhQ
This is a sketch: https://yadi.sk/d/-JZFi_y-g8mIhA
Another moment ... errors are issued on all examples (scatches) of fastlad

focalintent, 2019-01-22:
Take this line out of your code:

```
void yield(void){}
```

The updated library version now takes care of making sure that it is defined properly - which is why your error is now a complaint about yield being defined multiple times, vs. not being defined:

```
C:\Users\TOR\AppData\Local\Temp\arduino_build_891875\libraries\FastLED-master\FastLED.cpp.o: In function `yield':

C:\Arduino\libraries\FastLED-master/FastLED.cpp:236: multiple definition of `yield'

C:\Users\TOR\AppData\Local\Temp\arduino_build_891875\sketch\gyverLight_v1______________5__________2.ino.cpp.o:F:\Arduino Files Compiling\gyverLight_v1______________5__________2/gyverLight_v1______________5__________2.ino:63: first defined here

collect2.exe: error: ld returned 1 exit status
```

remenikomer, 2019-01-22:
> Sketch uses 14172 bytes (47%) of program storage space. Maximum is 29696 bytes.
> Global variables use 536 bytes of dynamic memory.

All built-in examples of the library also give such memory errors.

focalintent, 2019-01-22:
That's not an error - it's just letting you know how much memory the program is using, and if you are seeing those messages it means that it compiled correctly (otherwise I can't report the memory usage/sketch size)

remenikomer, 2019-01-22:
Thank you so much. Recycled, I do not understand anything already)

commit 1a235790 (2019-01-22): Fix #725 - make sure yield gets defined for WAV boards