FastLED/FastLED #726: STM32F4 Discovery

zytra, 2019-01-27:
hi guys

I'm getting an AVR/io.h error when compiling and it looks like my board may not be compatible.
Although I'm not sure as I found some old forum posts saying it was resolved but they didn't give a clear fix.

I have tried the latest release as well as a custom/modified 3.0.3 that according to forums worked with the STM32 but it didn't work for me..

any pointer would be appreciated, thanks.

> In file included from C:\Users\steph\Documents\Arduino\libraries\FastLED/led_sysdefs.h:32:0,

                 from C:\Users\steph\Documents\Arduino\libraries\FastLED/FastLED.h:41,

                 from C:\Users\steph\Documents\Arduino\IRIS-NEURO\IRIS-NEURO.ino:3:

C:\Users\steph\Documents\Arduino\libraries\FastLED/platforms/avr/led_sysdefs_avr.h:12:20: fatal error: avr/io.h: No such file or directory

 #include <avr/io.h>

                    ^

compilation terminated.

exit status 1
Error compiling for board STM32 Discovery F407.

focalintent, 2019-01-27:
The current STM32 code is for the STMF1 - it doesn’t look like I don’t have any STM32F4 definitions - I’ll have to add it as a platform to support in the future.  

(When it doesn’t find a platform definition, FastLED defaults to the avr platform)

zytra, 2019-01-27:
Thanks for the quick and helpful response! How likely is it that it could get supported in the coming 2-3 weeks? I'm early enough on this project that switching to the Due is still an option, but if the STM32F4 Discovery could get implemented quickly then I would rather wait and not start over.

Thanks again

focalintent, 2019-01-27:
Probably not that quickly

zytra, 2019-01-27:
Got it, thanks Daniel!

keukenrol, 2019-10-05:
I am looking for the F4 support as well. I am planning to make a light system with 2 WS2812B strips connected to a STM32F446 Nucleo kit.

zackees, 2025-08-25:
I'm leaving for burning man. I can maybe look at it next week

zackees, 2025-09-13:
[image]

Official support has been added. Will be released with 3.10.3