FastLED/FastLED #732: 4 WS1812b strips on nano D2-D5 will not compile

tip32a, 2019-02-04:


tip32a, 2019-02-04:
I am using the cyclone demo and have 4 ws2812b 144 led strips on D2-D5 on a genuine arduino nano. 
The code in question looks like
LEDS.addLeds<WS2812,5,RGB>(leds,NUM_LEDS);
LEDS.addLeds<WS2812,4,RGB>(leds,NUM_LEDS);
LEDS.addLeds<WS2812,3,RGB>(leds,NUM_LEDS);
LEDS.addLeds<WS2812,2,RGB>(leds,NUM_LEDS); 

This causes a compile error. 
FastLED version 3.002.001

lto-wrapper.exe: fatal error: C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-gcc returned 1 exit status

compilation terminated.

c:/program files (x86)/arduino/hardware/tools/avr/bin/../lib/gcc/avr/5.4.0/../../../../avr/bin/ld.exe: error: lto-wrapper failed

collect2.exe: error: ld returned 1 exit status

exit status 1
Error compiling for board Arduino Nano.

>>>>>> Yet when I use 5,4,3,6 it compiles.
The problem is the nano is already epoxied in place and I really need to use D2-D5 if possible.

Thanks in advance.

focalintent, 2019-02-04:
This looks like it is most likely a bug in Arduino/the compiler Arduino uses -  what version of Arduino are you using?

tip32a, 2019-02-06:
1.8.8

tip32a, 2019-02-06:
In my app if I have 5,4,3, and 2 OR 5,4,3 and 6 I get

lto-wrapper.exe: fatal error: C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-gcc returned 1 exit status

compilation terminated.

c:/program files (x86)/arduino/hardware/tools/avr/bin/../lib/gcc/avr/5.4.0/../../../../avr/bin/ld.exe: error: lto-wrapper failed

collect2.exe: error: ld returned 1 exit status

exit status 1
Error compiling for board Arduino Nano.