FastLED/FastLED #733: Cannot build on esp8266 Arduino

CRCinAU, 2019-02-08:
Recently, when building a new binary for an existing project, FastLED fails to compile with the following flood of errors:

```
Processing d1_mini (platform: https://github.com/platformio/platform-espressif8266.git#feature/stage; board: d1_mini; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/d1_mini.html
PLATFORM: Espressif 8266 (Stage) > WeMos D1 R2 & mini
HARDWARE: ESP8266 160MHz 80KB RAM (4MB Flash)
Converting ESP_MQTT_Digital_LEDs.ino
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 30 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <FastLED> 3.2.6
|   |-- <EspSoftwareSerial> 3.4.1
|-- <ESP8266WebServer> 1.0
|   |-- <ESP8266WiFi> 1.0
|-- <ESP8266httpUpdate> 1.3
|   |-- <ESP8266HTTPClient> 1.2
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <ESP8266WiFi> 1.0
|-- <ESP8266HTTPClient> 1.2
|   |-- <ESP8266WiFi> 1.0
|-- <PubSubClient> 2.7
|-- <ArduinoJson> 5.13.4
|-- <ESP8266WiFi> 1.0
|-- <SimpleTimer>
|-- <ESP8266HTTPUpdateServer> 1.0
|   |-- <ESP8266WiFi> 1.0
|   |-- <ESP8266WebServer> 1.0
|   |   |-- <ESP8266WiFi> 1.0
Compiling .pioenvs/d1_mini/libf5d/FastLED_ID126/FastLED.cpp.o
Compiling .pioenvs/d1_mini/libf5d/FastLED_ID126/bitswap.cpp.o
Compiling .pioenvs/d1_mini/libf5d/FastLED_ID126/colorpalettes.cpp.o
Compiling .pioenvs/d1_mini/libf5d/FastLED_ID126/colorutils.cpp.o
Compiling .pioenvs/d1_mini/libf5d/FastLED_ID126/hsv2rgb.cpp.o
In file included from libs/FastLED_ID126/led_sysdefs.h:43:0,
from libs/FastLED_ID126/FastLED.h:41,
from libs/FastLED_ID126/FastLED.cpp:2:
/home/netwiz/.platformio/packages/framework-arduinoespressif8266@src-31d658a59f41540201fc3726a1394910/cores/esp8266/Arduino.h:191:14: error: conflicting declaration 'typedef bool boolean'
typedef bool boolean;
^
In file included from libs/FastLED_ID126/led_sysdefs.h:27:0,
from libs/FastLED_ID126/FastLED.h:41,
from libs/FastLED_ID126/FastLED.cpp:2:
libs/FastLED_ID126/platforms/esp/8266/led_sysdefs_esp8266.h:15:17: error: 'boolean' has a previous declaration as 'typedef uint8_t boolean'
typedef uint8_t boolean;
^
Compiling .pioenvs/d1_mini/libf5d/FastLED_ID126/lib8tion.cpp.o
Compiling .pioenvs/d1_mini/libf5d/FastLED_ID126/noise.cpp.o
In file included from libs/FastLED_ID126/led_sysdefs.h:43:0,
from libs/FastLED_ID126/FastLED.h:41,
from libs/FastLED_ID126/bitswap.cpp:2:
/home/netwiz/.platformio/packages/framework-arduinoespressif8266@src-31d658a59f41540201fc3726a1394910/cores/esp8266/Arduino.h:191:14: error: conflicting declaration 'typedef bool boolean'
typedef bool boolean;
Compiling .pioenvs/d1_mini/libf5d/FastLED_ID126/power_mgt.cpp.o
^
In file included from libs/FastLED_ID126/led_sysdefs.h:27:0,
from libs/FastLED_ID126/FastLED.h:41,
from libs/FastLED_ID126/bitswap.cpp:2:
libs/FastLED_ID126/platforms/esp/8266/led_sysdefs_esp8266.h:15:17: error: 'boolean' has a previous declaration as 'typedef uint8_t boolean'
typedef uint8_t boolean;
^
In file included from libs/FastLED_ID126/led_sysdefs.h:43:0,
from libs/FastLED_ID126/FastLED.h:41,
from libs/FastLED_ID126/colorpalettes.cpp:4:
/home/netwiz/.platformio/packages/framework-arduinoespressif8266@src-31d658a59f41540201fc3726a1394910/cores/esp8266/Arduino.h:191:14: error: conflicting declaration 'typedef bool boolean'
typedef bool boolean;
[... 167 lines cut ...]
/home/netwiz/Documents/Arduino Code/Bed RGB/src/ESP_MQTT_Digital_LEDs.ino:719:17: warning: unused variable 'starthue' [-Wunused-variable]
static uint8_t starthue = 0;
^
/home/netwiz/Documents/Arduino Code/Bed RGB/src/ESP_MQTT_Digital_LEDs.ino:735:47: warning: NULL used in arithmetic [-Wpointer-arith]
if (transitionTime == 0 or transitionTime == NULL) {
^
/home/netwiz/Documents/Arduino Code/Bed RGB/src/ESP_MQTT_Digital_LEDs.ino:763:47: warning: NULL used in arithmetic [-Wpointer-arith]
if (transitionTime == 0 or transitionTime == NULL) {
^
/home/netwiz/Documents/Arduino Code/Bed RGB/src/ESP_MQTT_Digital_LEDs.ino:785:48: warning: NULL used in arithmetic [-Wpointer-arith]
if (transitionTime == 0 or transitionTime == NULL) {
^
/home/netwiz/Documents/Arduino Code/Bed RGB/src/ESP_MQTT_Digital_LEDs.ino:813:48: warning: NULL used in arithmetic [-Wpointer-arith]
if (transitionTime == 0 or transitionTime == NULL) {
^
/home/netwiz/Documents/Arduino Code/Bed RGB/src/ESP_MQTT_Digital_LEDs.ino:833:37: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
if ((millis() - flashStartTime) <= flashLength) {
^
/home/netwiz/Documents/Arduino Code/Bed RGB/src/ESP_MQTT_Digital_LEDs.ino:882:23: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
if (now - lastLoop > transitionTime) {
^
*** [.pioenvs/d1_mini/src/ESP_MQTT_Digital_LEDs.ino.cpp.o] Error 1
```

I'm non the wiser about how this has happened to fail. Has anyone else come across this?

focalintent, 2019-02-08:
Huh - looks like the esp8266 platform may have recently added a boolean typedef - I’ll have to poke around to fix this in a way that won’t break folks running against an older esp8266 platform.

CRCinAU, 2019-02-08:
I know I'm in over my head now - but if it helps, this is built with platformio...

The platform.ini file is as follows:
```
[env:d1_mini]
board_build.f_cpu = 160000000L
platform = https://github.com/platformio/platform-espressif8266.git#feature/stage
board = d1_mini
framework = arduino
build_flags =
  -Wl,-Teagle.flash.4m1m.ld
  -D PIO_FRAMEWORK_ARDUINO_LWIP2_HIGHER_BANDWIDTH
  -DNDEBUG
```

From another project that doesn't use FastLED, I get the following versioning strings:
```
SDK:3.0.0-dev(c0f7b44)/Core:unspecified=0/lwIP:STABLE-2_1_2_RELEASE/glue:1.1/BearSSL:6778687
```

CRCinAU, 2019-02-14:
So just as a note to others, I managed to get this working again by commenting out a single line:

File: /platforms/esp/8266/led_sysdefs_esp8266.h
Line: 15
Change to: //typedef uint8_t boolean;

Things now compile and run ok - but I'm not sure how you'd do this as a conditional for the SDK version?

I wonder if there's a single #ifdef that would apply?

mattwoolnough, 2019-05-11:
Commenting out the line worked for me as a workaround also. Thanks @CRCinAU.

ccoenen, 2019-05-12:
Commenting that line (as mentioned by others!) fixes the problem for me, too.

https://github.com/FastLED/FastLED/blob/403464a499a8feffa48f6f85d205550b9bc9c89b/platforms/esp/8266/led_sysdefs_esp8266.h#L15

ljmf00, 2019-05-18:
This issue should be closed due to 6b6c9c1c35f235e846334d9cd8984b4b4399f471 @focalintent .

Edris89, 2019-06-01:
Same problem!

closed 2019-08-12 by focalintent