FastLED/FastLED #735: FastLED doesn't support hardware SPI on SAMD21

DavidHaitch, 2019-02-13:
According to [issue 652](https://github.com/FastLED/FastLED/issues/652), FastLED doesn't currently use any hardware acceleration on the D21. I'm trying to do something very similar to what the issue poster wanted to do, so I'm considering taking this task on.

Before I sink too much time into this, I wanted to check if there was any reason why no one else has done this already, either because of some technical or architectural limitation, or because bitbanging on the D21 isn't significantly slower than SERCOM SPI. 

I also want to check if this task is on anyone else's radar, since this would be my first contribution to FastLED and I don't want to step on any toes.

focalintent, 2019-08-12:
Mainly I haven’t had the time to do it.

danimateos, 2024-05-14:
Hi! I'd be very interested in this! Is there any I could help?

I have a custom SAMD21E18A-based board that I'm using for a POV project and I'd love to squeeze a few us off the refresh time.