FastLED/FastLED #736: Adafruit M0 Express Built-in LED

McCallisterMicro, 2019-02-14:
I've been experimenting with an Adafruit Metro M0 Express board, and I'm having some trouble activating the built-in Neopixel on the board. The pixel is on pin 40. The code works fine when hooking up a strip externally - I tested on on Pin 7 and had no issue. I just can't seem to access the built-in Neopixel.

I tested the Adafruit Neopixel library and was able to get it to work using simliar code. I'm guessing it has something to do with how the board is defined, but that's a bit beyond my current coding understanding. 

Here is my example code with the Adafruit library code commented out for comparison:
```

#include <Metro.h>
#include <FastLED.h>
//#include <Adafruit_NeoPixel.h>

uint8_t isLEDON;
Metro Ticker = Metro(100);
CRGB dot[1];
//Adafruit_NeoPixel strip = Adafruit_NeoPixel(1, 40, NEO_GRB + NEO_KHZ800);

void setup()
{
	pinMode(13, OUTPUT);
	pinMode(40, OUTPUT);
	Serial.begin(115200);
	FastLED.addLeds<NEOPIXEL, 40>(dot, 1);
	//strip.begin();

}

void loop()
{
	if (Ticker.check())
	{
		digitalWrite(13, isLEDON ? HIGH : LOW);
		isLEDON = !isLEDON;
		FastLED.show();

	}
	dot[0] = CRGB::White;
	//strip.setPixelColor(0, 255, 255, 255);
	//strip.show();
}

```

focalintent, 2019-02-14:
the pin/port definition for pin40 might be wrong on that board (fastled uses its own mappings to go from pins to low level hardware registers) - i’ll have to see if i can dig up what it should be

hiduino, 2019-08-29:
There currently is no pin definitions for the ADAFRUIT_METRO_M0_EXPRESS as of FastLED v3.3.2.

McCallisterMicro, 2021-01-01:
By any chance, has this been addressed in release 3.4? If not; I think I may be able to assist, but I don't quite know where to start. If I could get some guidance on the files/directories where this is handled, I bet I could suss it out and add in the needed code.