FastLED/FastLED #737: GW6205 + ESP8266 - All pixels slightly lit

EmilJa, 2019-02-16:
Hello!
I have aquired a couple of LED-strands with the GW6205 chips.
They work just fine with an Arduino UNO, but with the ESP8266 (same code) all pixels have their last 4 bits set to 1, which results in the pixels assigned the color 0x000000 (black) get at least the value 15 in both R, G and B channels.

I have tried looked in to the source code to find out what's going on, but it's too complicated for me.
I suspect that it has something to do with the conversion from 8-bit values to 12-bit values.
Below is some confirmation on this error aquired with a logic analyzer.

`ESP8266 LED0: 0x00FF00  LED1: 0x000000
--------------
0000 0000 1111 
1111 1111 1111 
0000 0000 1111

0000 0000 1111
0000 0000 1111 
0000 0000 1111

Arduino UNO LED0: 0x00FF00  LED1: 0x000000
--------------
0000 0000 0000
1111 1111 1111
0000 0000 0000

0000 0000 0000
0000 0000 0000
0000 0000 0000

Arduino UNO LED0: 0xFF0000  LED1: 0x00FF00  LED2: 0x0000FF
--------------
1111 1111 1111
0000 0000 0000
0000 0000 0000

0000 0000 0000
1111 1111 1111
0000 0000 0000

0000 0000 0000
0000 0000 0000
1111 1111 1111

ESP8266 LED0: 0xFF0000  LED1: 0x00FF00  LED2: 0x0000FF
--------------
1111 1111 1111
0000 0000 1111
0000 0000 1111

0000 0000 1111
1111 1111 1111
0000 0000 1111

0000 0000 1111
0000 0000 1111
1111 1111 1111`

focalintent, 2019-02-16:
I see what’s wrong - I’ll try to get a fix in for it today

focalintent, 2019-02-16:
Pull the latest from git and let me know if that takes care of it - and I'll spin up a new release some time this week (need to see if there's anything else that I want in before doing that).

closed 2019-02-16 by focalintent

EmilJa, 2019-02-16:
Thank you for your quick response!
I have tried the fix and it works perfect now.

commit 3db75d4b (2019-02-16): Fix #737 - don't invert the bits in what's being written out until after shifted, otherwise we end up writing out 1's for the low bits on 12-bit output devices