FastLED/FastLED #739: GS1903 still Problems

Pfannex, 2019-02-19:
Hi,

for my drawer lights I want to run my GS1903 led strip with the fastLED library. A ESP8266 is used as controller.
Generally the wiring and the simple sketch runs.
The GS1903 is fired with 12VDC, the data pin is directly connected to the ESP8266 GPIO 5.

I use the work around with the WS2812B model. #684
But I´m not able to controll the GS1903 stable.
Sometimes it works sometimes don´t.

For me it looks like communication problems, may be sometimes some bits are missing / wrong?
Or timing problems?

Can I run some further tests?

greetings
Pf@nne

focalintent, 2019-02-19:
It’s possible that you need a 5v level shifter on the data line.

Pfannex, 2019-02-21:
OK, with the level shifter the communication runs quite better, but not error free. It looks like that there are still some bits missing / corrupted.
Does a video helps for solving the problem?

With a WS2812(b) (5VDC) I did not have any problems.

focalintent, 2019-02-21:
I found something claiming to be the data sheet for it - and according to the timings in there, SM16703 has the same timings 300/900us for a 0 and 900/300us for a 1) - so try SM16703

focalintent, 2019-02-21:
Also it has a really short reset time - only 40us - so try disabling interrupts if you are still seeing it be off.

Pfannex, 2019-02-21:
After digging a while, I also found a chinese datasheet and can confirm the timings.

I will test the SM16703 when I´m home.

What do you mean with disabling interrupts?
ESP8266 interruppt or is there a interrupt functionallity within the fastLED-library?

Thanks for your quick support!
And sorry for asking about that simple problems.
I did not spend enough time in changing the controller from WS2812b (works out of the box) to the GS1903.
I run all my software and breadboard test with the WS2812b.
Only to reduce the current I change to the 12VDC GS1903.

That is most provoking, because my PCBs are in shippment and no level shifter is present......

greatings from Hamburg
Pf@nne

Pfannex, 2019-02-21:
I got only a 16Mhz logic analyzer (resolution 62,5ns) but I think SM16703 HIGH/LOW-timing is OK.
But what about the reset time, the datasheet below shows 40µs.
The measured time is 1,5µs. 

But with this gap (40µs / 1,5µs) shouldn´t run at any time?
[image]
[image]

focalintent, 2019-02-21:
The reset time is how much time there needs to be between frames not between individual bits/pixels.

Pfannex, 2019-02-23:
After rewiring and minimalizing my breadboard setup everything works as expected! 
Also controlling the GS1903 with a ESP8266 GPIO level of 3V3!
So no level shifter is used.

What ist the actual timing of the GS1903 (WS2812B)?
T0H=400ns / T0L=850ns; T1H=800ns / T1L=450ns; res>50µs?

I think the timming is a little bit different to the searched GS1903 datasheet? 
T0H=300ns / T0L=900ns; T1H=900ns / T1L=300ns; res>40µs?
So maybe some GS1903 will not run errorfree.

So adding a real GS1903 makes sense.

thanks for you support an great library!

greatings
Pf@nne

[image]

WS2812B datasheet
[image]