FastLED/FastLED #740: ws2812b flicker jitter when using WiFi on esp8266

XBoter, 2019-02-19:
So where do i start. I'm dealing with the Problem for a while now and at this Point i don't know what to do next. The Problem is that my ws2818b leds start flickering or jitter when using the FastLed library and esp8266Wifi library. I use 3 LED Strips with 3 different Ports but the same Problem occurs when only using one Port.
I tried follow things:

- Tested the Cyclone example from the FastLED library without WiFi (no success)

- played with the defines FASTLED_INTERRUPT_RETRY_COUNT and FASTLED_ALLOW_INTERRUPTS (no success)

- tried resistors on the data line (no success)

- tried NeoPixel Library (no success)

- used different ws2812b leds (success but doesn't help with the original problem)

- used a level shifter(3.3V to 5V) on the data line (no success)

- used a esp32 with one Core for Wifi and the other Core for FastLed (no success)

- used 4 different esp8266 (no success)

- used 3 different power supplys (no success)

- used an arduinio Uno with the Cyclone Sketch (success)

So after the success with the Uno i tried sending commands over Serial form my esp8266 (gets the Original Commands over WiFi and sends those to the Uno) to the Uno to control them this way. But i didn't get the Serial Communication Stable so i stopped with that

I Analyzed the Data Signal with the Oscilloscope and it looks like it has some ringing. I reduced the ringing with some resistors but that dint helped either.

In my Opinion it looks like a timing problem but i don't know how to fix it. I'm close to defeat to the China Led Strip Gamble and this is my last fight before its over. So I'm grateful for any help greetings XBoter

Singal without resistor
[image]

Signal with 50 ohm resistor
[image]

Gif of the Problem (cant upload the Video :( )
[image]

XBoter, 2019-02-19:
Oops, a mistake happened the two images are reversed. The top is with a 50 ohm resistor and the bottom one without

tobi01001, 2019-02-19:
Some ideas to narrow it down:
- data line appears to be quite long (hard to see). Did you try shorter lines? What about crosstalk (I can hear the led colours in the radio when being close. 
- data line has a good connection to the port? I found some breadboards providing bad connections. 
- Your osci measurements were taken without the level shifter? It will usually not hurt using one... 
- did you look at the port or also at the data line? 

- Did you try with fewer leds each strip? Maybe only 10 each to see if it's better? 
How many leds do you have in each strip? 

- you said your problem is with fastled and WiFi? However the first bullet with the cyclone sketch was without WiFi as well and did not work? So it's not WiFi?

Don't know if that brings you forward....

focalintent, 2019-02-19:
What level shifter did you try?  Preferably a 74HCT245 - a number of other shifters can’t keep up with the signal and can make the timing worse.

XBoter, 2019-02-20:
> Some ideas to narrow it down:
>
> * data line appears to be quite long (hard to see). Did you try shorter lines? What about crosstalk (I can hear the led colours in the radio when being close.
> * data [...]

Thanks for you help I forgot one important thing, that if you leave the led strips + esp8266 with no power for like 2 min and restart it. Then it works just fine but after +30sec it start flickering and gets worse with the time.

Now to your questions 

I can't make the data lines between the strips segments shorter (if you mean that with the "data line appears to be quit long"), because where I install them I need the 50cm of space between them. And if i but them 1 meter away from each other its still the same.
I don't use a breadboard to connect the esp8266 to the data line (It's there from another project, the esp8266 is the thing wit the blue led on the bottom left). And I tried a direct cable from the pin on the esp8266 to the data line with no success. My measurements from the osci where with no level shifter (but i will try the ones from focalentents comment). 
They were taken form the first input to the led strip. One Strip got 90 leds and I tried it with 30 already but the same problem occurred. And to answer your last question Yeah after searching for a solution I got stuck on the solution that it has to do something with the WiFi. But yeah i don't think it has to do something with the WiFi

XBoter, 2019-02-20:
> What level shifter did you try? Preferably a 74HCT245 - a number of other shifters can’t keep up with the signal and can make the timing worse.

I can't find a name or a number on the level shifter (I think it was from a set or something). But I bought some off the 74HCT245 shifters and when they arrive I will try those out.

A Picture of the level shifter
[image]

focalintent, 2019-02-21:
Yeah - at best those shifters will do nothing for you, at worst they’ll make things even worse :)

muddin721, 2019-03-01:
https://github.com/FastLED/FastLED/wiki/Interrupt-problems

 FastLED.show() disables interruptions and it may cause your data to be lost. I assumed that the reason of your problem is the strict timing problem you mentioned. 

I solved this timing ploblem using two boards, which can be just a cheap nano or uno. Two of them are connected with each other using serial or Whatever and interruption pin. Concept of mine is this : One working as 'Output board', it requests data by interruption pin connected to 'Input Board' and receives data by serial. Then it finally sends data to LED strips. After the work is done, it requests again and loops forever. The other working as  'Input Board', it receives data which is from wifi or usb, and stacks on its memory. When it receives REQUEST data from the Output board, it sends data stacked on mem to the Output.

If you need my code, please give me your email. I hope this solution will help you.

XBoter, 2019-03-01:
> Oh...Am I too late to reply?
> 
> https://github.com/FastLED/FastLED/wiki/Interrupt-problems
> 
> FastLED.show() disables _interruptions_ and it may cause your data to be lost. I assumed that the reason of your problem is the strict timing problem you mentioned.
> 
> I solved this timing ploblem using two boards, which can be just a cheap nano or uno. Two of them are connected together using serial or something else, and _interruption_ pin connected from Output to Input (I will mention below).
> 
> Concept of mine is this : One working as 'Output board', it requests data by _interruption_ pin connected to 'Input Board' and receives data by serial. Then it finally sends data to LED strips. After the work is done, it requests again and loops the same cycle forever.
> 
> The other working as 'Input Board', it receives data which may be from wifi modules or usb serial communication, and stacks them on the board's memory. When it receives _REQUEST_ from the Output board, it sends data stacked on the memory to the Output.
> 
> It might disable your _interruption_ of the Output in the cycle of showing LED, but won't affect anything to your Input board, so your data will be sustained.
> 
> If you need my code, please give me your email. I hope this solution will help you.

Thanks for your help. I tried something similar myself but failed. It would be very nice if you could send me your code. I have sent you an e-mail to your e-mail which you deposited with github :D

XBoter, 2019-03-14:
After many tests with the ESP8266 and ESP32, trying with DMA and bus drivers, I gave it up and threw away the LEDs. I surrender to the China gamble and will close this issue with it.

closed 2019-03-14 by XBoter

marmilicious, 2019-03-15:
@muddin721 
I think others might be interested in this sort of solution.  If you have a basic working example I encourage you to put the code on github or pastebin and share the link on the Reddit group.
https://www.reddit.com/r/FastLED
Others can learn and be able to search it out there in the future.

mplover, 2020-07-18:
So I may be too late to this, but I wanted to put my 2 cents in. I was experiencing the same issue when power was provided to the NodeMCU solely from USB. All I did to remove the flickering was provide a dedicated ground between the NodeMCU and the power supply I used to provide 5V for the LED strips. When it's all powered with just the power supply alone (No USB), I didn't experience any issues.

XBoter, 2020-07-18:
> So I may be too late to this, but I wanted to put my 2 cents in. I was experiencing the same issue when power was provided to the NodeMCU solely from USB. All I did to remove the flickering was [...]

I tried different power supply with the same ground share but that didn't help either. After that I tried it with different LEDs and it worked fine so in the end my LEDs were just garbage.

msmaximuss, 2021-12-09:
Struggling with this issue for a few days. Then just connected ESP8266 GND  to external source GND and magic happens!

blacknull, 2023-02-12:
The same thing happen to me when I use wifi and fastled on an esp8266 board, but finally I solved it. You have to downgrade ur arduino core from 3.xx to 2.74, then everything goes right. I don't know why but it did work.