FastLED/FastLED #741: compilation fail for STM32F103C (Blue Pill)

AndKe, 2019-02-20:
```
home/andre/Arduino/projects/libraries/FastLED/platforms/arm/stm32/led_sysdefs_arm_stm32.h:4:25: fatal error: application.h: No such file or directory
 #include <application.h>
                         ^
compilation terminated.
```
tested FastLED 3.2.1  ... 3.2.6(current release) 

@kolyshkin  you surely have a working version?

kolyshkin, 2019-02-20:
I took a look yesterday and I can't figure out where the hell `application.h` is coming from. Note that I'm not the one who added this `#include` statement, I just modified the `#ifdef`. So it's still a mystery to me :-\

I took another look, and it seems that `application.h` is some file which is auto-generated by Arduino. Perhaps you can either try changing `<application.h>` to `"application.h"`, or just commenting this line altogether.

kolyshkin, 2019-02-20:
For the reference, this `#include "applcation.h"` statement was added by commit f149084f46 (by @focalintent) and later changed to `#include <application.h>"` (which I believe is wrong since it's supposed to be a local include) by commit 6fd236684. 

@AndKe which arduino version are you using? Please try the workarounds I provided above, let us know if it helps.

kolyshkin, 2019-02-20:
Well, I have removed `#include <application.h>` and everything seems working just fine. Not sure why it worked for me before...

kolyshkin, 2019-02-20:
@focalintent maybe you can shed some light? It seems this `#include <application.h>` is some kind of leftover that needs to be removed

focalintent, 2019-02-21:
The stm32 code was originally done for the particle photon boards and that platform needed the application.h - support for other stm32 variants has mostly come in from folks tweaking things to support other stm32 based boards, and they never took it out - if someone can figure out when it should be there vs when not - I’d prefer being able to gate when the include is used - otherwise I’m removing it I just open myself up to a bunch of issues complaining about things missing :)

kolyshkin, 2019-02-21:
Oh, in such case it should be very easy to fix -- since the original code used `STM32F10X_MD` define, this is what we should use :)

@focalintent PTAL at https://github.com/FastLED/FastLED/pull/742

closed 2019-02-21 by focalintent

AndKe, 2019-02-21:
I tried FastLED master, on Arduino 1.8.7 

FastLED.addLeds<NEOPIXEL, RGB_FL>(leds, 0, NUM_LEDS_PER_STRIP);  

RGB_FL is PAx or PBx throws this error:
```
/home/andre/Arduino/projects/libraries/FastLED/fastpin.h:207:2: error: static assertion failed: Invalid pin specified
```
certainly an improvement, but it does not seem to "understand" PA0 pr PB6 or any other ports/pins I tried.

focalintent, 2019-02-21:
What is the actual value for RGB_FL - if there isn’t a definition for that pin in platforms/arm/stm32/fastpin_stm32.h, then that will cause this error (for performance reasons, FastLED does its own definitions/mappings for pins to ports)

focalintent, 2019-02-21:
Looking at the commit for blue pill support - it looks like it doesn’t have pin definitions for all the pins on the blue pill board (eg pins 25-28, possibly others?)

AndKe, 2019-02-21:
I tried PA0...6 and PB5..9    (mentioned it before)     so RGB_FL is/was  PA0 , then PA1 and so on 
to me, it seems like all the pins are just fine here: https://github.com/FastLED/FastLED/blob/master/platforms/arm/stm32/fastpin_arm_stm32.h#L107

focalintent, 2019-02-21:
What’s the whole compiler output - there should be more context to that message to show what it thinks RGB_FL is - Also where is RGB_FL getting defined?  It wasn’t referenced in the Arduino STM32 core that I found.

focalintent, 2019-02-21:
Oh - also, what happens if you use integers instead of PA*?  I generally tell folks to use integer pins - because on some platforms D* or other ways of defining pins end up not being constant expressions.

focalintent, 2019-02-21:
Oh - I see what you mean, you’re defining RGB_FL to be something like PA0 - it depends on how the core/platform you’re using defines PA0 and friends.   But again, I generally discourage the use of named pins vs PIN numbers (there’s a reason you never see D10 in an example, but rather just 10)

AndKe, 2019-02-21:
yes, using integers build it successfully... except: 
"Sketch uses 96600 bytes (147%) of program storage space. Maximum is 65536 bytes"

this is kind of crazy, I had this program, with the same libraries, fit on a Atmega 328 just fine, but not on this 64k STM32
Something must be eating lots of space ?

AndKe, 2019-02-21:
..if I omit FastLED library and a few kb of code related to it, I am down to 52% of space.

focalintent, 2019-02-22:
That sounds odd - I wonder if the stm32 compiler is doing a poor job of pruning unreferenced code?

kolyshkin, 2019-02-22:
Haha, @AndKe, you're following my steps! The reason is, you need to add a flag. Please see https://github.com/rogerclarkmelbourne/Arduino_STM32/pull/444 
and
https://github.com/rogerclarkmelbourne/Arduino_STM32/issues/296

AndKe, 2019-02-22:
@kolyshkin 
"I generally tell folks to use integer pins"   I tried to use integers like 10 for PA0 ,   11 for PA1 , then it compiles but the application does not seem to run on microcontroller (crashes ( hangs at FastLED.addLeds)  (I see it on UART activity)
then I tried   14,15,16,17 for PA4,5,6,7  then it runs, but the led strips does not do anything

as for the compiler flags, I hope that becomes a permanent fix at some point, or easily selectable.