FastLED/FastLED #743: Gradient palettes get stored in RAM on ESP8266

Aircoookie, 2019-02-22:
Hello!
I've noticed that if I use the following syntax to define a palette:

```
DEFINE_GRADIENT_PALETTE( heatmap_gp ) {
  0,     0,  0,  0,   //black
128,   255,  0,  0,   //red
224,   255,255,  0,   //bright yellow
255,   255,255,255 }; //full white
```

the palette gets stored in RAM, not PROGMEM, even though the docs say so.

In `platforms/esp/8266/led_sysdefs_esp8266.h` I found this definition:

```
// Default to NOT using PROGMEM here
#ifndef FASTLED_USE_PROGMEM
# define FASTLED_USE_PROGMEM 0
#endif
```

I have some ~2kB worth of palette definitions in my application and would like to free up some memory.
Can someone explain the issue of enabling PROGMEM on ESP8266?

kriegsman, 2019-02-22:
Thank you for pointing this out; I think it’s probably just an oversight and something we can change easily. I’ll take a look later today if I can.

Aircoookie, 2019-02-22:
Thanks for your quick response! I just tried `# define FASTLED_USE_PROGMEM 1` in my sketch.
Unfortunately, it seems like the default of disabling it for esp8266 was in fact intentional.
It compiles just fine and reduces the memory usage as expected, but as soon as a palette stored in PROGMEN is selected, the application crashes (using latest 2.5.0 esp8266 core and FastLED 2.3.6):

```
Exception 9: LoadStoreAlignmentCause: Load or store to an unaligned address
Decoding 10 results
0x402080b6: CRGBPalette16::operator=(unsigned char const*) at C:\Users\cschwinne\Documents\Arduino\libraries\FastLED/colorutils.h line 836
0x40269a7d: lwip_init at ?? line ?
0x4020966a: WS2812FX::handle_palette() at C:\Users\CSCHWI~1\AppData\Local\Temp\arduino_build_5212\sketch/WS2812FX_fcn.cpp line 699
0x40100370: millis at C:\Users\cschwinne\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\cores\esp8266/core_esp8266_wiring.c line 183
0x40100370: millis at C:\Users\cschwinne\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\cores\esp8266/core_esp8266_wiring.c line 183
0x40100370: millis at C:\Users\cschwinne\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\cores\esp8266/core_esp8266_wiring.c line 183
0x402096e3: WS2812FX::service() at C:\Users\CSCHWI~1\AppData\Local\Temp\arduino_build_5212\sketch/WS2812FX_fcn.cpp line 66
0x40219c26: loop at C:\Users\cschwinne\git\Arduino\WLED02\wled00/wled00.ino line 560
0x4022f524: loop_wrapper() at C:\Users\cschwinne\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\cores\esp8266/core_esp8266_main.cpp line 125
0x40100ea5: cont_wrapper at C:\Users\cschwinne\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\cores\esp8266/cont.S line 81
```

Maybe I can try it again later with a simple palette example sketch to verify it's not an issue with my rather complex application.

Aircoookie, 2019-02-22:
Alright, didn't know why ColorPalette.ino worked just then. Here is a minimal example where the issue occurs, without using WiFi libs or anything.
[PaletteProgmemTest.zip](https://github.com/FastLED/FastLED/files/2894523/PaletteProgmemTest.zip)

By the way, I don't need a solution quickly, so take your time, just making you aware of it :)

kriegsman, 2019-02-22:
ESP microcontrollers requires 4-byte-aligned reads from PROGMEM; AVR do not.  I'm pretty sure that when we first ported to ESP, I said "oh well, I'll just stick it in RAM for now, and come back to deal with the specifics of ESP's needs about PROGMEM later."   Now, it appears, is 'later'.

kokospalme, 2019-11-19:
Hello,
I have a similar problem with my ESP8266 ( I think its the same bug as described above).
When I choose a gradient palette the whole controller crashes and throws this over Serial:

>Fatal exception 9(LoadStoreAlignmentCause):
>epc1=0x40207c9a, epc2=0x00000000, epc3=0x00000000, excvaddr=0x3ffe85cd, depc=0x00000000
>
>Exception (9):
>epc1=0x40207c9a epc2=0x00000000 epc3=0x00000000 excvaddr=0x3ffe85cd depc=0x00000000
>
>\>>>stack>>>
>
>ctx: cont
>sp: 3fff0d20 end: 3fff0fa0 offset: 01a0
>3fff0ec0:  3ffe8b0a 000000a0 000000a0 00000078  
>3fff0ed0:  3ffef511 3ffef510 3ffeff0c 40208d0c  
>3fff0ee0:  00000000 00000000 3ffeff0c 00000078  
>3fff0ef0:  3ffeff0c 3ffef510 00000004 402081ea  
>3fff0f00:  00000097 3ffef510 00000004 40206fce  
>3fff0f10:  00000004 00000000 00000000 000000e4  
>3fff0f20:  00000000 000000ff 3ffef510 ffffffff  
>3fff0f30:  3ffef986 3ffef986 3ffef510 3fff0fd8  
>3fff0f40:  00000000 00005000 3ffeeb88 402075d3  
>3fff0f50:  3fffdad0 00000000 3ffeeb88 40208056  
>3fff0f60:  3fffdad0 00000000 3ffeeb88 40207849  
>3fff0f70:  00000000 00000000 3fff0fa8 40202604  
>3fff0f80:  3fffdad0 00000000 3fff0fa8 40209ae4  
>3fff0f90:  feefeffe feefeffe 3ffe8510 40101159  
><<<stack<<<
>
> ets Jan  8 2013,rst cause:2, boot mode:(3,7)
>
>load 0x4010f000, len 1384, room 16 
>tail 8
>chksum 0x2d
>csum 0x2d
>v951aeffa
>~ld
>
>SDK:3.0.0-dev(c0f7b44)/Core:2.5.0=20500000/lwIP:IPv6+STABLE-2_1_2_RELEASE/glue:1.1/BearSSL:6778687

Other Palettes seem to work proper.

Will this be fixed any time soon or is there a workarround for now?
Thank you :)