FastLED/FastLED #744: Does not work on STM32

AndKe, 2019-02-23:
Arduino 1.8.8 , FastLED current master, the LED strip is working fine with this library on a Atmega328:

On a BluePill , STM32F103C8 , with STM32duino bootloader, I tried the example "ColorPalette" 
changed only "#define LED_PIN"

to 17  as "PA7" is not recognized
then tried PB5 , (recognized).
no LED activity.

focalintent, 2019-02-23:
FYI, the stm32 support is very loose - I don’t actually have any stm32 hardware to test with, outside of the photon, and so mostly have been relying on what other folks supply for the code - especially for things like pin/port mapping.

AndKe, 2019-02-23:
they cost next to nothing, https://www.ebay.com/itm/STM32F103C8T6-Minimum-System-Developmen-Board-Module-Core-Learning-For-Arduino/323633385054?hash=item4b5a0d8e5e:g:ilcAAOSw9m5cAggN:rk:10:pf:0
If you could look into it, you could order, and I'll paypal yoy some $ ?

AndKe, 2019-02-27:
I have tried both to address port as PB5  or  "41"  - no success.
I've also checked with oscilloscope, and there is no save output on any pins, 
This is a map of what works , and what not:
(existing port/pins)  OK(pins recognized)  "not"= pins not recognized
//PA 0...7,13,14    OK=2,3,4,13,14  (not 0,1,5...7)
//PB 0..15          OK=0...6  14,15 (not 7..12)
//PC 13,14,15      OK=13,14,15

Example: "PA2" works, "PA6" does not.
This tells me that it is likely, that the pin defines for "Blue Pill" does not apply as expected here: https://github.com/FastLED/FastLED/blob/master/platforms/arm/stm32/fastpin_arm_stm32.h#L107

stujamac, 2019-04-09:
So I have had the same issue, it seems to be a simple issue in that the pins are defined wrongly in "fastpin_arm_stm32.h".
_DEFPIN_ARM is being called with the physical pin number as option 1, whereas it should be the required Arduino pin that it will be aliased as.
Am testing on a Maple Mini and all seem to work now I have got the defines straightened out.  As mentioned above

Abhyudaya98, 2019-09-06:
> Arduino 1.8.8 , FastLED current master, the LED strip is working fine with this library on a Atmega328:
>
> On a BluePill , STM32F103C8 , with STM32duino bootloader, I tried the example "ColorPalette" [...]

I was dealing with the same issue of getting WS2812B leds work with my blue pill using the fastLED library. I found a quick fix to this issue - I set the variant to be STM32F103CB (20k RAM, 128k flash) instead of (20k RAM, 64k flash) in the tools menu of the Arduino IDE - which helped me bypass the error messages. You can venmo me if this works 🤑

berylleigh, 2019-09-10:
I have it working.

I have used fastpin_arm_stm32.h from here: https://github.com/13rac1/FastLED-STM32. I simply replaced the existing file in FastLED\platforms\arm\stm32 with the one from the stm32 fork git. The pin mapping that works looks like this (though get the file as the formatting didn't copy correctly):

`#if defined(STM32F1)
#define _RD32(T) struct _gen_struct ## T { static attribute((always_inline)) inline gpio_reg_map* r() { return T->regs; } };
#define _IO32(L) _RD32(GPIO ## L)

_IO32(A); _IO32(B); _IO32(C); _IO32(D);

#define MAX_PIN PB1

_DEFPIN_ARM(PB11, 11, B);
_DEFPIN_ARM(PB10, 10, B);
_DEFPIN_ARM(PB2, 2, B);
_DEFPIN_ARM(PB0, 0, B);
_DEFPIN_ARM(PA7, 7, A);
_DEFPIN_ARM(PA6, 6, A);
_DEFPIN_ARM(PA5, 5, A);
_DEFPIN_ARM(PA4, 4, A);
_DEFPIN_ARM(PA3, 3, A);
_DEFPIN_ARM(PA2, 2, A);
_DEFPIN_ARM(PA1, 1, A);
_DEFPIN_ARM(PA0, 0, A);
_DEFPIN_ARM(PC15, 15, C);
_DEFPIN_ARM(PC14, 14, C);
_DEFPIN_ARM(PC13, 13, C);
_DEFPIN_ARM(PB7, 7, B);
_DEFPIN_ARM(PB6, 6, B);
_DEFPIN_ARM(PB5, 5, B);
_DEFPIN_ARM(PB4, 4, B);
_DEFPIN_ARM(PB3, 3, B);
_DEFPIN_ARM(PA15, 15, A);
_DEFPIN_ARM(PA14, 14, A);
_DEFPIN_ARM(PA13, 13, A);
_DEFPIN_ARM(PA12, 12, A);
_DEFPIN_ARM(PA11, 11, A);
_DEFPIN_ARM(PA10, 10, A);
_DEFPIN_ARM(PA9, 9, A);
_DEFPIN_ARM(PA8, 8, A);
_DEFPIN_ARM(PB15, 15, B);
_DEFPIN_ARM(PB14, 14, B);
_DEFPIN_ARM(PB13, 13, B);
_DEFPIN_ARM(PB12, 12, B);
_DEFPIN_ARM(PB8, 8, B);
_DEFPIN_ARM(PB1, 1, B);

#define SPI_DATA BOARD_SPI1_MOSI_PIN
#define SPI_CLOCK BOARD_SPI1_SCK_PIN

#define HAS_HARDWARE_PIN_SUPPORT

#endif`
[fastpin_arm_stm32.zip](https://github.com/FastLED/FastLED/files/3595396/fastpin_arm_stm32.zip)

AndKe, 2020-03-07:
I have been trying to make it work on STM32 Core, not the stmduino, as STM32 Core is significantly better/more developed.

SWAT10101, 2020-06-24:
hello, guys, I have this [code ](https://github.com/Wifsimster/adalight_ws2812) and I can't verify to **STM32F103C** everything ok but when I set `DATA_PIN` to `A1`, I got this error.
Is this library support board, Is there any way to run this code on the board?

[image]

closed 2025-08-01 by zackees

AndKe, 2025-08-11:
Oh, hello - it would require time travel. I had a project that could use that  >6 years ago. :(