FastLED/FastLED #745: FastLed

Horvat2003, 2019-02-25:


senic35, 2019-02-26:
What exactly don't you understand? There is the setBrightness command to, as the name implies, set the brightness of your leds. Check out the documentation in the wiki. IIRC there is also a "map" function to map a range of values from e.g your rotary encoder to the brightness value range of your leds.

marmilicious, 2019-02-26:
@Horvat2003  As senic35 mentions you could feed your increment change to FastLED.setBrightness(BRIGHTNESS);  or if you are using HSV to assign data to your LEDs you can have it adjust the V component (value/brightness).  See this wiki page for HSV info:  https://github.com/FastLED/FastLED/wiki/Controlling-leds

That said, the github here is for reporting bugs.  For general help with code or getting things like this working with your project please ask over at the FastLED Reddit discussion group.  There are lots more people there that can help out.  https://www.reddit.com/r/FastLED/new

Horvat2003, 2019-02-27:
Thank you very much, I will ask further questions on reddit.

closed 2019-08-12 by focalintent