FastLED/FastLED #746: GW6205 + ESP32

erikbrodnick, 2019-03-01:
I am unable to successfully get my GW6205 pixels controlled properly using a Node ESP32 MCU. I have even tried with SN74HCT245 level shifter. I am not sure what the problem is and have tried several different GPIO. For a sample string of 20 pixels, simply trying to blink an individual pixel is unresponsive and sometimes will display completely different color on the wrong pixel. I'm happy to provide any further information to help figure out how to get this working. I will also say that these pixels run off 24V DC power. I have a separate power supply for the pixel string and then have the data coming in from the esp32, sharing common ground. 

Appreciate the help in advance.

focalintent, 2019-03-02:
For laughs, try GW6205_400 as the led type - also make sure you’re using the most recent version of FastLED (3.2.6) as there have been esp32 fixes to the clock less code.

erikbrodnick, 2019-03-02:
Tried the GW6205_400 to no avail. I have also updated the FastLED library to the latest 3.2.6. I am using the example sketch from FastLED "Blink" for my testing.

erikbrodnick, 2019-03-05:
I have the Adafruit ESP32 feather as well as the Node-MCU ESP32. Have tried both with and without the level shifter and various GPIO pins still to no avail. 

I found this video [https://www.youtube.com/watch?v=ykwkaPpUBZY](https://www.youtube.com/watch?v=ykwkaPpUBZY) on youtube where they are using the FastLED library with another type of mcu and seem to get it working with the GW6205.

focalintent, 2019-06-12:
Ah - I see the problem -- I forgot that the GW6205 is a 12-bit per color channel chipset, vs. almost everything else which is 8-bit per color.  The esp32 implementations of the clockless controller are locked at 8-bits at the moment.

erikbrodnick, 2019-06-12:
I will give it another try. You are a smart man. I’m not nearly as technical as you tho this was my suspicion as well because what I have gathered is the fact that the gw6205 is 12-bit. I’ve seen others post videos on youtube having success with the gw6205 and FastLED however believe that they are using some variant of Arduino to control.

I have ~800 of the gw6205 pixels that I have installed around the perimeter of my home. Currently they are controlled by some Chinese controller that only has about 30 preset patterns and sequences. My goal is to be able to dynamically change these and create patterns with a simple iOS app. If you do a search for trimlight or gemstone lighting you will see what I mean.

Cheers,
-Erik

focalintent, 2019-06-12:
There's a template parameter for the clockless controller - XTRA0 - which it looks like the esp32 implementations are ignoring.  On other platforms, it just writes out a handful of extra 0's -- I don't remember how widespread the support for 12-bit output is, though - since I never had any leds that supported it so it wasn't part of my regular testing.  (Side effect of what I'm going to do for rgb/rgbw/rgb16/rgbw16 - support for chipsets like this will become more trivial/complete)

erikbrodnick, 2019-06-13:
I see. So if I understand correctly, because the XTRA0 is being ingnored by the esp32 that is why it doesn’t work on it but others have had success on arduinos? If all was working with the esp32, how many pixels would it be able to drive from single output? I have about 800 pixels I’m looking to control from single output. What Arduino device would you recommend?

Appreciate your help.
-Erik

focalintent, 2019-06-13:
That’s correct - I’ll ha e to check through the code to see what platforms support the 12-bit output.  

As for how many pixels to drive - that’s a complicated question to answer.  What frame rate are you looking for?  The gw6205’s take 45us to write a single led’s worth of data, or 36ms to write a frame - which means you’re looking at a max of 20-30fps. Then there’s a question of memory - 800 leds will want 2400 bytes of ram - which knocks out most of the avr based Arduinos outside of the mega.

erikbrodnick, 2019-06-14:
I see. Makes sense. I have reached out to the developer of the controller and iOS app for https://www.trimlight.com/ <https://www.trimlight.com/>. This is basically the controllability I seek. Unfortunately the GW6205 is not on their list of supported ICs. The developer said it may be possible but would need to see a data sheet. If you could send what you have to me when you’re back and able that would be greatly appreciated. 

-Erik