FastLED/FastLED #747: Support for nRF52xxx (e.g., nRF52840)

henrygab, 2019-03-03:
Is it possible to port to the nRF52840, at least for SPI LED types?

Although the nRF52xxx is similar to the nRF51xxx (which is supported), if I understand correctly, portions of the nRF52 core run at high interrupt level (for BLE), and they cannot be disabled by user application code.  Obviously, this could impact any code depending on tight execution timing.  Based on this, would some LED control types need to be explicitly excluded (e.g., NeoPixel)?

focalintent, 2019-03-04:
I haven’t seen anyone supply even a partial port - and I won’t be able to get to it for a while, yet.

pcaddict, 2019-03-15:
Take a look at my fork [here](https://github.com/pcaddict/FastLED). I have tested with the nrf52832 and it should work exactly the same for the nrf52840. I have only tested APA102 but I'd guess other SPI chipsets should work.

henrygab, 2019-03-19:
Wow, looks great, @pcaddict 

Would you be able to help me understand a couple items?

Line https://github.com/pcaddict/FastLED/blob/master/platforms/arm/nrf52/fastspi_arm_nrf52.h#L59 appears to set MISO to port0.pin1.
**Q# 1a. Is this intentional?**
**Q# 1b. ```PSEL.CSN``` is not explicitly set, and thus appears to be presumed to be the reset-default  value of ```0xFFFFFFFF```.  I think this is OK, as the .  Do you agree?**
**Q# 1b. Is it legal to leave MISO disabled while using SPIM, if you never care about input values?  (and if so, should this be set to ```0xFFFFFFFF```?)**

Line https://github.com/pcaddict/FastLED/blob/master/platforms/arm/nrf52/fastspi_arm_nrf52.h#L87 appears to state that the function returns immediately after setting the register.  However, it appears at line https://github.com/pcaddict/FastLED/blob/master/platforms/arm/nrf52/fastspi_arm_nrf52.h#L95 to wait for the end-of-transmission event.
**Q#2.  Is the comment wrong, the code wrong, or my understanding wrong?**

Thanks!

henrygab, 2019-03-23:
After lots of tinkering, compiling, and adding compilation asserts / pragma messages, I've started to understand some aspects of this port.  I think it will be possible to port this, which is great.  Even better, we may be able rely upon the availability of Nordic's nrfx repo (part of at least Adafruit's BSP for Feather Express 52840.  Work now underway....

pcaddict, 2019-03-23:
Sorry for the late reply, I did not see a notification that anyone had responded.
**Q1a**. I don't think setting MISO to pin 1 was intentional. It probably should have been set to 0xFFFFFFFF. As far as I can tell from the documentation it is perfectly fine to not specify a pin for MISO if you aren't using it.

**Q1b**. Agreed. I think I might have explicitly told PSEL.CSN to be 0xFFFFFFFF at some point but probably removed it for some unknown reason.

**Q2**. The comment is wrong. I took code from the NRF51 and Teensy platforms and it is likely something that originated elsewhere.

What features of the nrfx library are you thinking will be useful here?

I have not had much luck running this on my nrf52840. I have not been able to get the Adafruit BSP to play nicely with the Particle Xenon board I have, which is mostly used with the Nordic SDK anyway, so all of my testing has been with my feather52832.

henrygab, 2019-03-24:
Thanks for confirming #2 comment is just outdated.

Why use nrfx?
1. a hardware-abstraction layer (HAL), useful for switching between SPIM instances
2. same HAL allows common code for nRF51 and nRF52 (and other) boards
3. built-in workarounds for chip errata, owned and maintained by Nordic (i.e., less really migraine-inducing, head-scratching hardware debugging)

I've already gotten the simplest of items showing on a dotstar strip (RGB ordering verification, simple chaser).

What specific problems have you experienced?

pcaddict, 2019-03-24:
@henrygab I'm interested to see what improvements you've made to my hatchet job of a port. I honestly got it to a point of "hey pretty lights" one afternoon and did not explore it much further than that. I did try to understand how to get neopixels working and ran out of patience.

In regards to nrfx, I wonder if that isn't going to add a bunch of complexity that many users won't want to deal with, especially in Arduino land. Might make the possibility of porting to the Nordic SDK a little simpler though.

As far as validating that support is complete, I'd imagine that'll take some community testing with various led chipsets. This is probably something @focalintent should comment on.

For some reason Particle Xenon isn't playing nicely with the nordic QSPI libraries in the Adafruit BSP. I've tried removing everything I could find having to do with the QSPI memory and still have problems with most sketches compiling. Since I got the Xenon primarily for working with the Nordic SDK anyway I have not put too much time towards figuring out why I am having the problems I am. Other than that the Xenon is a great dev board and they were a steal at $9 during pre-order.

henrygab, 2019-03-27:
@pcaddict Ok, will share "real soon now"... as soon as I have anything that I believe is any improvement.

Another question... am I misunderstanding something here?
https://github.com/pcaddict/FastLED/blob/master/platforms/arm/nrf52/fastpin_arm_nrf52.h#L83
This appears to allow the pin's register to be set to literally any value, without regard to the value of _MASK.  Doesn't this break the encapsulation / purpose of _MASK?
In other words, shouldn't this instead be something more like:
```NRF_P0->OUT = val & _MASK```?

focalintent, 2019-03-27:
No it shouldn’t  - that’s used for an optimization in some places where I precompute the full value for the port so that setting the pin levels just becomes a single write vs a read, or/and, then write. (1 memory access in one instruction vs 2  memory accesses and 3 instructions (or, in the case of combining clock/data pin setting, 4 memory accesses and 6 instructions))

henrygab, 2019-03-28:
@pcaddict -- found one bug that may affect use of certain boards / pins, and thus might fix your Xenon issue.   I just need to confirm it compiles, then I will push the first major fix.  ETA: a couple days, maybe sooner if the Arduino IDE plays nice.

henrygab, 2019-03-30:
Ok.  Frustrated with Arduino.  Getting closer, but extending ETA.  Don't want to push stuff that doesn't work.  Wish there were better documentation on the implied contracts that the fastpin / fastspi functions must follow to work properly with FastLED….

henrygab, 2019-04-01:
@pcaddict -- Found the following note in the nRF52840 product specification, at the end of the SPIM chapter:
32 High bit rates may require GPIOs to be set as High Drive, see GPIO chapter for more details.

Based on the above, could your problems with data corruption when trying to use >4MHz be solved by simply reconfiguring the pins to high drive mode for both 0 and 1 (H0H1)?

If you have something that works at 4MHz, try using the above instead of the following lines in Init() in fastspi_arm_nrf52.h:
For the nRF52840 at least, this would be something such as:
```C
        //// pins must be set to input/output before enabling the SPIM instance
        //FastPin<_DATA_PIN>::setOutput();
        //FastPin<_CLOCK_PIN>::setOutput();

        // set drive mode to H0H1
        PIN_CNF[_DATA_PIN ] = ( PIN_CNF[_DATA_PIN ] & (~0x00030000u) ) | 0x3;
        PIN_CNF[_CLOCK_PIN] = ( PIN_CNF[_CLOCK_PIN] & (~0x00030000u) ) | 0x3;
```

If that solves your high-speed clocking for SPI, I'll be glad.

henrygab, 2019-04-10:
Getting closer... The nRF52840 has more than 32 pins, and thus has two ports.  This requires the _ARMPIN template to be … slightly more complex.  Throw in GCC's / C++'s restrictions on template parameter types, and it's just a matter of finding an expression that doesn't cause compile-time errors....

pcaddict, 2019-04-10:
Would it not make sense to just pass the result of g_ADigitalPinMap[pin] as the parameter for each pin?

henrygab, 2019-04-10:
@pcaddict - g_ADigitalPinMap[pin] is not a constexpr, and is not resolved at compile-time.  Therefore, it creates some limitations for use as a template parameter.

focalintent, 2019-04-10:
@henrygab there are other arm platforms that have multiple ports that you can use as a reference - e.g. the KL26 - https://github.com/FastLED/FastLED/blob/master/platforms/arm/kl26/fastpin_arm_kl26.h - and yes, multiple ports does make things a little bit more complicated.  As for the optimization away - it's possible that part of what's happening is the compiler is seeing two adjacent writes to the same register and assuming that this isn't a problem - in theory the register pointers and types should all be peppered liberally w/volatile - to tell the compiler to not optimize out writes to them - in practice, you might need to add another operation between the two calls to kick things into working right.  

I had to do something along these lines in the spi bitbang code because of irritations with overly optimizing compilers:

https://github.com/FastLED/FastLED/blob/master/fastspi_bitbang.h#L126

henrygab, 2019-04-11:
@focalintent 
Ouch.  The method used by KL26 hurts my head, but it might work... but it appears that I'll need to re-define the pin map for each and every distinct board (not just chipset).  Can you help me confirm my attempt to summarize this method?  Also, can you confirm my current path forward seems reasonable (listed at end)?

#### Summary of KL26 power macros
If I understand it correctly, the KL26 fastpin header works around the template parameter limitation that prevents easy use of a named structure pointer / named register.   It does so by defining (using preprocessor string concatenation) a unique structure using the `_RD32(T)` macro.  The structure defined by the `_RD32(T)` macro's sole purpose is to ... return the named structure / named register via the static member function r().

The actual `_ARMPIN` template is then defined to take `typename` parameters, using those unique structures (using `_R(T)` macro), while internally the template class's functions simply get the underlying value by calling `::r()` on the typename parameter.

**_Is that about right?_**

#### Applying to nRF52....
The nRF52 series lacks at least two key features of the nRF51:
* there are at least four microcontrollers, with differing pins
* not all boards have "natural" Arduino pin to port/pin mapping (`g_ADigitalPinMap[i] == i`) 
* some boards have >32 pins (and thus multiple ports)

However, because all the boards define `NRF_P0`, and those with more than 32 pins define `NRF_P1`, it seems we can (ab)use the same KL26 preprocessor tricks to pass the correct port name to `_RD32()` to generate a structure that returns an `NRF_GPIO_Type *` corresponding to either `NRF_P0` or `NRF_P0` as appropriate.

The, because the boards differ, it appears a unique `_DEFPIN_ARM()` series would need to be defined for each board, which basically duplicates the `g_ADigitalPinMap` for the given board.  Other than that, it's likely possible to support all the nRF52 series boards with a single common FastPin code base.

**_Because of the tricky work involved (preprocessor macro bugs are less than friendly to debug), your help confirming the above seems about right would be greatly appreciated._**  

Thank you!

focalintent, 2019-04-11:
Actually - if there’s only two ports, and they have different names, you could also do something like what the esp32 does - but yes, you generally have it right - the way that I do pins for FastLED means a custom map for each board layout (the AVR boards have this in spades).   The macros for assembling the template instantiations were because of weird subtle errors in getting the hand rolled instantiations right - but templating around pointers to ensure compiling down to direct register access was tricky to get right to begin with.  But it really was worth it to have 1-2 cycle pin access :)

pcaddict, 2019-04-11:
> 
> 
> @pcaddict - g_ADigitalPinMap[pin] is not a constexpr, and is not resolved at compile-time. Therefore, it creates some limitations for use as a template parameter.
> 
> > a non-type, non-template  template-parameter must be a pointer to
> > a constexpr that designates the address of an object with
> > static storage duration AND external or internal linkage  --OR--
> > a function with external or internal linkage.
> 
> So, the template receives a PIN, which is the Arduino definition and does NOT match the underlying chip PORT/PIN. Thus, the template must have a static _init() function that converts the passed-in PIN into the PORT/PIN that actually matters. For nRF52840, also need to save pointer to the (NRF_GPIO_Type*) for the PORT, because there are two distinct ports (yes, 48 pins... so either NRF_P0 or NRF_P1 is used).
> 
> Or at least, this is my working theory at this point. Loading the .elf in Ozone, my calls to `FastPin<PIN_LED1>::toggle()` is being optimized away, even with -O0, so I'm working through figuring that out at the moment....

Would using the following macros from the Nordic SDK make things any easier? 

constexpr uint32_t NRF_GPIO_PIN_MAP(uint32_t port, uint32_t pin) { return (((port) << 5) | ((pin) & 0x1F)) }

#define PIN_PORT(_pin) (((_pin) >= P0_PIN_NUM) ? NRF_P1 : NRF_GPIO)

henrygab, 2019-04-12:
@pcaddict Unfortunately, no.   The Nordic SDK is only dealing with **_board_** pins (port/pin).

In the Arduino world, people use the Arduino pin numbers.  Those pin numbers are mapped to underlying board port/pin through `g_ADigitalPinMap[]`, which allows people to write code in Arduino that targets pin 1 (e.g., digital pin 1 or analog pin 1), and switch mcus / boards without rewriting their code, and being assured that the pin has the same capability (digital GPIO or analog input).

Why do the nrf51 and your initial nrf52 port work?  The supported target boards happen to use identity mapping from Arduino pin to board port/pin.  That is `g_ADigitalPinMap[i] == i` is true for all pins, so they could be used interchangeably.

Look at variant.cpp for your board's pin definitions.

The good news: I'm getting close to a compiling FastPIN for nRF52 that supports 48-pin boards with non-identity pin mapping, such as my initial test board the nRF52840.....

pcaddict, 2019-04-12:
I was also too lazy to dig further into the NRF_P1 issue and just used NRF_P0 in my limited “port”. I recognized there was a problem but was really just after making it function as a proof of concept. 

I am glad to see that you’re making progress and am excited to see what you have come up with.  One of the things that kept me from exploring a solution further was figuring out templates. I’m a little more we’ll read in them now but they still make me go cross eyed, especially the templates used in FastLed.  I have also pretty much abandoned using Arduino for anything but the most basic tests now and have been using the Nordic SDK almost exclusively when working with their micros.

henrygab, 2019-04-12:
@pcaddict I'm thrilled you got as far as you did.  Note also that the nRF52 series does *NOT* support high-speed GPIO on all pins... limited to <=10kHz.  For those supporting high-speed GPIO, you may need to set additional bits in the configuration register (NRF_GPIO_Type*) beyond direction, such as H0H1 (high-drive when setting  to zero or one).

I had difficulty with templates also, so I just hard-coded a couple classes for `NRF_P0` and `NRF_P1` … while still allowing _DEFPIN_ARM() style macros to define the per-board pin mappings.  I will only be able to test/confirm the Feather nRF52840 Express board.  However, once that's done, the other boards should "just work" by defining the appropriate board variants (separate file for that).

Overall, I'm feeling optimistic, although the going is slow due to real-life.

henrygab, 2019-04-13:
I have something that works (software bitbang'd SPI).
I now have very high confidence that I can get FastPin support for substantially all nRF52 boards.
I now have high confidence that I can get FastSPI support for substantially all nRF52 boards.
I have low confidence that I can get clockless support working.

@focalintent , some questions so I prep stuff appropriately....
1. Do you want a PR, even if I only have FastPin complete?
2. Do you want a PR, even if I only have FastPin + FastSPI (no clockless)?
3. Do you want a PR to update the porting wiki pages? (e.g., add information on questions I had, as a newcomer to the code base, roadblocks hit, solutions considered and taken, etc.,)

focalintent, 2019-04-14:
Thank you!

I'll take a PR for as much as there is - but probably to a branch, not yet master if there isn't yet clockless support.

As for clockless support, the NRF52 is still a cortex m0 or m0+, right?  

In which case you can do something similar to what's done in these:

d21/clockless_arm_d21.h / d51/clockless_arm_d51.h / kl26/clockless_arm_kl26.h / nrf51/clockless_arm_nrf51.h

The first two template parameters are the number of bytes from the base port register to the port register for setting high values (e.g. writing 1 to bits on that register sets individual pins high and ignores all the pins at the bits that are 0) and the number of bytes from the base port register to the port register for setting low values (e.g. writing 1 to bits on that register sets individual pins lo and ignores all the pins at the bits that are 0)

I would prefer that there be clockless support in for a platform before I merge it in to the mainline (because SPI can always fallback to bit-banging I feel less strongly about hardware SPI).  Parallel clockless output is always optional : )

henrygab, 2019-04-14:
@focalintent
I'm not confident that I could make clockless work in a reasonable timeframe.
1. The nRF52 has interrupts (for BLE) that the platform does not allow to be disabled.  Therefore, there is always the chance for interrupts that will prevent clockless from working correctly.
2. I don't have a (real) oscilloscope.  Therefore, I wouldn't be able to trace / verify the code paths that ensure clockless works in the hard cases.
3. The processor is a 32-bit ARM M4F @ 64MHz.  I am not sure if that's sufficiently equivalent to M0 or M0+.
In contrast, I have gotten FastPin to a working, stable, and essentially complete state.  I also am confident about FastSPI (including preventing attempts to use low-speed pins if using high-speed SPI, and setting pins to high-drive zero / high-drive one mode for higher SPI speeds … both quirks for the nRF52 series).

focalintent, 2019-04-14:
In that case, I'd take a PR to a branch - and then pull the nrf52 over once I get a chance to actually get the clockless support in.  Given that the clockless chipsets are by far more widely used, it'd be kind of important to get it in before pulling the platform in to master.

henrygab, 2019-04-14:
Ok.  A few related questions:

1. Do I generate the branch in your depot as part of the PR, or do you set it up?  (either way, may I suggest 'nrf52' as the branch?)

2. Although I've added pin definitions for a large set of nrf52-based boards, I only have one of those to test with.  Do you want the PR to have the following line for the untested boards?
`#error "Untested board -- remove this line, test, and submit results via https://github.com/FastLED/FastLED/issues"`

3. Is there a "how to test your port" documentation anywhere?  If not, do you have recommendations of various levels or types of testing for any given board port?

With that, I think I'm ready to provide at least the FastPIN implementation for initial code review.

#### List of board variants prepped:
| From [AdaFruit nRF52 BSP](https://github.com/adafruit/Adafruit_nRF52_Arduino) | Tested | `#define` |
| --- | --- | --- |
| Adafruit Bluefruit nRF52840 Feather Express  | **Yes** | `ARDUINO_NRF52840_FEATHER` |
| Adafruit Bluefruit nRF52832 Feather | No | `ARDUINO_NRF52832_FEATHER` |
| Adafruit Bluefruit nRF52840 Metro Express | No | `ARDUINO_NRF52840_METRO` |
| Adafruit Bluefruit on nRF52840DK PCA10056 | No | `ARDUINO_NRF52840_PCA10056` |

| From the [Open Source Arduino nRF5](https://github.com/sandeepmistry/arduino-nRF5) | Tested | define |
| --- | --- | --- |
| Electronut labs bluey | No | `ARDUINO_ELECTRONUT_BLUEY` |
| Electronut labs hackaBLE | No | `ARDUINO_ELECTRONUT_HACKABLE` |
| Electronut labs hackaBLE_v2 | No | `ARDUINO_ELECTRONUT_hackaBLE_v2` |
| RedBear Blend 2 | No | `ARDUINO_RB_BLEND_2` |
| RedBear BLE Nano 2 | No | `ARDUINO_RB_BLE_NANO_2` |
| Nordic Semiconductor nRF52 DK | No | `ARDUINO_NRF52_DK` |
| Taida Century nRF52 mini board  | No | `ARDUINO_STCT_NRF52_minidev` |
| Generic nRF52832 | No | See header |

I added variants for the three AdaFruit BSPs

focalintent, 2019-04-14:
I think you can target a new branch in the PR - if not I should be able to edit it 

Use #warning instead of #error (#error will break/stop compilation)

The testing of a port involves running it against a variety of led types to make sure all the variations on things work : ) (my workshop has a lot of test strips for when I'm actively working on testing a board/port - I'll also rig stuff up on breadboards so I can rapidly try testing across the full set of ports, to be sure - which is a habit I got into after the Arduino mega - which had different cycle times for writing different pins!)

henrygab, 2019-04-16:
@focalintent Wow!  that's quite a lot.

I have validated most of the individual ports work on the nRF52840, connected to individual pins.  I ran the code under the Ozone debugger with a J-Link EDU.  This helped me realize how terrible inclining all code is for debugging.

I have validated basic functionality for 16-count APA102, using software SPI.

However, I couldn't create a new target branch with my PR, so I created against master.

henrygab, 2019-04-25:
@focalintent OK, there's now a new [pull request](https://github.com/FastLED/FastLED/pull/779), with both FastPIN and FastSPI implemented.  If someone pre-builds the buffers they want to use (e.g., more likely for POV displays), the FastSPI implementation supports a single DMA-enabled SPI transaction.

pcaddict, 2019-04-25:
@henrygab I do not have time to dig much into clockless support but if you take a look in the examples of the latest Nordic Thread and Zigbee SDK they have added a driver for WS2812 LEDs that might get you pointed in the right direction.

henrygab, 2019-04-27:
The sample code is not directly relevant to clockless.

The sample sets a PWM output to the right clock rate, pre-generates the full LED string's output, and then just starts a NRFX-based playback of that pre-generated output.

Positives:
1. Very simple to comprehend
2. Low CPU overhead while updating the LED strings
3. Easy to adapt to re-use the same LED update buffer (e.g., static or limited set of static outputs, such as used for POV)
4. Easy to adapt to allow single PWM to control multiple LED strings with single PWM instance, so long as the protocol clock rate is the same

Negatives:
1. Very RAM-hungry... 48 bytes extra RAM per pixel(!) … Is this reasonable for FastLED?

In contrast, the FastLED clockless implementations I've looked at appear to translate the RGB data 'on the fly' / 'just in time' to adjust the GPIO pins, and thus avoid this extra RAM overhead.  In exchange, these methods re-compute the output for each LED string every time show() is called, even when no data changes for that string.

While I **could** code something like this up, I am concerned it doesn't fit the "flavor" of FastLED.
I'm also presuming async callbacks are supported in the Arduino environment (haven't used them before).  Worst, the main benefit (re-use of the buffer for refresh) is not usable with the FastLED architecture.

henrygab, 2019-05-15:
@focalintent, will you accept a PR with clockless implemented, even if the it uses more RAM than you're used to?

I believe I can design it to have an upper-bound memory overhead of about <2.5k RAM, even with BLE enabled.

Click on each item below to expand details....

<details>
  <summary>Why reliable clockless is hard on nRF52</summary>

  <p>
The `softdevice` can cause any in-progress clockless output to fail, if it relies on CPU.  The interrupt processing time is on the order of hundreds of microseconds, while the clockless protocols can fail with deviations as small as 50ns.
<ul>
<li> Radio (BLE, ANT, Zigbee, etc.) is software-defined (called `soft device`)
<li> Interrupts for the softdevice cannot be disabled
<li> Softdevice may hard-lock if app uses interrupt priorities of 0 or 1
<li> BLE specifically interferes with timing of clockless, due to both long latency from the BLE interrupt processing and potentially rapid timing between instances of those interrupts.
</ul>
</details>
<details>
  <summary>Example BLE Interrupt Timing for <a href="http://infocenter.nordicsemi.com/pdf/S140_SDS_v1.3.pdf">SoftDevice v1.40</a></summary>

  <p>

Usage / Pattern | Max | Typ | Freq
--|--|--|--
BLE Advertising | 348μs | 229μs | 40μs
BLE Connection | 406μs | 219μs | 183μs
BLE Scanner / Initiator | 273μs | 175μs | ???

Note: `Freq` is the minimum time between such interrupts occurring.

These timings only cover softdevice processing that occurs at interrupt priority levels 0 and 1 … the delays at priority level 4 (deferred BLE processing) can substantially increase these values.
</details>
<details>
  <summary>PWM peripheral based prototype</summary>

  <p>
I have a working prototype.  It's not templated, it uses the PWM controller, and it takes 48 bytes of RAM per pixel.  The output clocking is rock-solid, once it starts.  The PWM sequence buffer is compile-time allocated based on the length of the longest LED string, but only one such buffer exists, regardless of the number of output ports.
(e.g. 144 LEDs would take just under 7k RAM for PWM sequence data)
</details>
<details>
  <summary>Memory Usage vs. Timing</summary>

  <p>

Starting with the following:
* SoftDevice may pre-empt any application code for 400μs
* A single bit from a 800KHz protocol ~= 1.25μs
* 400μs / 1.25μs == up to 320 protocol bits sent during a SoftDevice interrupt
* PWM peripheral uses two bytes per protocol bit
* 14 pixels is 336 protocol bits (672 bytes for PWM peripheral)

The PWM peripheral can chain / loop up to two sequences, and it must be setup
so that it will have enough data if pre-empted at any time to be able to push
out another 14 pixels of data, AND be able to give the peripheral additional data
before it's fully drained.

Since pre-emption can happen with zero bits left in the current sequence,
each sequence must have at least 14 pixels of data.  A three-buffer approach,
each having ~16 pixels of data (768 bytes per sequence buffer, 2304 bytes total)
would likely be possible without loss of reliability in all but extreme cases, and
even there would be detectable.
</details>

Anyways, please let me know if this PR (high-RAM but reliable) would be of interest.

henrygab, 2019-05-24:
PR is now ready via #802, supporting compile-time determined fixed maximum length of clockless.,

closed 2019-07-16 by henrygab

jasoncoon, 2022-01-17:
I just received a XIAO BLE from Seeed, which has an nRF52840 M4F@64MHz: https://wiki.seeedstudio.com/XIAO_BLE

I get the following error when trying to compile the `DemoReel100` example with FastLED 3.5.0 & Arduino 1.8.19 on MacOS 11.6:

`This platform isn't recognized by FastLED... yet.  See comments in FastLED/led_sysdefs.h for options.`

I've never worked on adding support for a board.  It looks like the following are defined by the board definition: ARDUINO_ARCH_NRF52840, ARDUINO_SEEED_XIAO_NRF52840_SENSE, ARDUINO_ARCH_NRF52, ARDUINO_ARCH_MBED

If I manually add `#define NRF52_SERIES` to the sketch before `#include <FastLED.h>`, I instead get the following errors:

<details>
  <summary>Full error</summary>

```
/Users/jcoon/Documents/Arduino/libraries/FastLED/src/fastpin.h: In member function 'void Pin::_init()':
/Users/jcoon/Documents/Arduino/libraries/FastLED/src/fastpin.h:41:14: error: 'digitalPinToBitMask' was not declared in this scope
   mPinMask = digitalPinToBitMask(mPin);
              ^~~~~~~~~~~~~~~~~~~
/Users/jcoon/Documents/Arduino/libraries/FastLED/src/fastpin.h:41:14: note: suggested alternative: 'digitalPinToPinName'
   mPinMask = digitalPinToBitMask(mPin);
              ^~~~~~~~~~~~~~~~~~~
              digitalPinToPinName
/Users/jcoon/Documents/Arduino/libraries/FastLED/src/fastpin.h:42:28: error: 'portOutputRegister' was not declared in this scope
   mPort = (volatile RwReg*)portOutputRegister(digitalPinToPort(mPin));
                            ^~~~~~~~~~~~~~~~~~
/Users/jcoon/Documents/Arduino/libraries/FastLED/src/fastpin.h:43:30: error: 'portInputRegister' was not declared in this scope
   mInPort = (volatile RoReg*)portInputRegister(digitalPinToPort(mPin));
                              ^~~~~~~~~~~~~~~~~
In file included from /Users/jcoon/Documents/Arduino/libraries/FastLED/src/platforms/arm/nrf52/fastled_arm_nrf52.h:8:0,
                 from /Users/jcoon/Documents/Arduino/libraries/FastLED/src/platforms.h:11,
                 from /Users/jcoon/Documents/Arduino/libraries/FastLED/src/FastLED.h:54,
                 from /var/folders/5z/9mpthc4n4jl904ywb78k3w7c0000gn/T/arduino_modified_sketch_601815/DemoReel100.ino:3:
/Users/jcoon/Documents/Arduino/libraries/FastLED/src/platforms/arm/nrf52/clockless_arm_nrf52.h: In static member function 'static void ClocklessController<_DATA_PIN, _T1, _T2, _T3, _RGB_ORDER, _XTRA0, _FLIP, _WAIT_TIME_MICROSECONDS>::startPwmPlayback_EnableInterruptsAndShortcuts(NRF_PWM_Type*)':
/Users/jcoon/Documents/Arduino/libraries/FastLED/src/platforms/arm/nrf52/clockless_arm_nrf52.h:99:44: error: 'configMAX_SYSCALL_INTERRUPT_PRIORITY' was not declared in this scope
         result = sd_nvic_SetPriority(irqn, configMAX_SYSCALL_INTERRUPT_PRIORITY);
                                            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /Users/jcoon/Documents/Arduino/libraries/FastLED/src/FastLED.h:50:0,
                 from /var/folders/5z/9mpthc4n4jl904ywb78k3w7c0000gn/T/arduino_modified_sketch_601815/DemoReel100.ino:3:
/Users/jcoon/Documents/Arduino/libraries/FastLED/src/fastpin.h: In instantiation of 'class FastPin<3>':
/Users/jcoon/Documents/Arduino/libraries/FastLED/src/platforms/arm/nrf52/clockless_arm_nrf52.h:52:38:   required from 'static void ClocklessController<_DATA_PIN, _T1, _T2, _T3, _RGB_ORDER, _XTRA0, _FLIP, _WAIT_TIME_MICROSECONDS>::startPwmPlayback_InitializePinState() [with unsigned char _DATA_PIN = 3; int _T1 = 6; int _T2 = 8; int _T3 = 6; EOrder _RGB_ORDER = (EOrder)66; int _XTRA0 = 0; bool _FLIP = false; int _WAIT_TIME_MICROSECONDS = 10]'
/Users/jcoon/Documents/Arduino/libraries/FastLED/src/platforms/arm/nrf52/clockless_arm_nrf52.h:174:44:   required from 'void ClocklessController<_DATA_PIN, _T1, _T2, _T3, _RGB_ORDER, _XTRA0, _FLIP, _WAIT_TIME_MICROSECONDS>::init() [with unsigned char _DATA_PIN = 3; int _T1 = 6; int _T2 = 8; int _T3 = 6; EOrder _RGB_ORDER = (EOrder)66; int _XTRA0 = 0; bool _FLIP = false; int _WAIT_TIME_MICROSECONDS = 10]'
/var/folders/5z/9mpthc4n4jl904ywb78k3w7c0000gn/T/arduino_modified_sketch_601815/DemoReel100.ino:124:1:   required from here
/Users/jcoon/Documents/Arduino/libraries/FastLED/src/fastpin.h:210:2: error: static assertion failed: Invalid pin specified
  static_assert(validpin(), "Invalid pin specified");
  ^~~~~~~~~~~~~
In file included from /Users/jcoon/Documents/Arduino/libraries/FastLED/src/platforms/arm/nrf52/fastled_arm_nrf52.h:8:0,
                 from /Users/jcoon/Documents/Arduino/libraries/FastLED/src/platforms.h:11,
                 from /Users/jcoon/Documents/Arduino/libraries/FastLED/src/FastLED.h:54,
                 from /var/folders/5z/9mpthc4n4jl904ywb78k3w7c0000gn/T/arduino_modified_sketch_601815/DemoReel100.ino:3:
/Users/jcoon/Documents/Arduino/libraries/FastLED/src/platforms/arm/nrf52/clockless_arm_nrf52.h: In instantiation of 'static void ClocklessController<_DATA_PIN, _T1, _T2, _T3, _RGB_ORDER, _XTRA0, _FLIP, _WAIT_TIME_MICROSECONDS>::startPwmPlayback_InitializePwmInstance(NRF_PWM_Type*) [with unsigned char _DATA_PIN = 3; int _T1 = 6; int _T2 = 8; int _T3 = 6; EOrder _RGB_ORDER = (EOrder)66; int _XTRA0 = 0; bool _FLIP = false; int _WAIT_TIME_MICROSECONDS = 10]':
/Users/jcoon/Documents/Arduino/libraries/FastLED/src/platforms/arm/nrf52/clockless_arm_nrf52.h:268:47:   required from 'static void ClocklessController<_DATA_PIN, _T1, _T2, _T3, _RGB_ORDER, _XTRA0, _FLIP, _WAIT_TIME_MICROSECONDS>::startPwmPlayback(uint16_t) [with unsigned char _DATA_PIN = 3; int _T1 = 6; int  [... cut ...]
        #error "Cannot define more than one board at a time"
    #else
        #define __FASTPIN_ARM_NRF52_VARIANT_FOUND
    #endif

    _FL_DEFPIN( 10, 15, 1); // D10 is P1.15 -- PIN_SPI_MOSI
#endif // defined (ARDUINO_SEEED_XIAO_NRF52840_SENSE) 

```

</details>

But I'm not sure how to fix the other errors.

If I choose `Adafruit nRF52 Boards` > `Nordic nRF52840DK` it compiles fine, but errors trying to upload.

@henrygab or anyone else have any ideas?  Any help would be greatly appreciated.  Thanks in advance!

henrygab, 2022-01-18:
Hi Jason,

Funny how small a world it is.  😄 

Yes, I have ideas.  Here's a [relevant post](https://github.com/FastLED/FastLED/issues/1055#issuecomment-667931207) for a different board (ItsyBitsy).

#### Caveats

* There was no testing on MBED OS.  There may be significant work required to get this working on MBED OS.

* nRF52840 is currently [hard-coded](https://github.com/henrygab/FastLED/blob/1a273c8d3c99bbecccd9e2b1456516da8156415b/src/platforms/arm/nrf52/clockless_arm_nrf52.h#L10) to maximum 144 pixels.  Currently, you have to edit the header to change this.

* There are [design considerations](https://github.com/henrygab/FastLED/blob/1a273c8d3c99bbecccd9e2b1456516da8156415b/src/platforms/arm/nrf52/clockless_arm_nrf52.h#L330-L385) that make this board very tough to get reliable outputs on, at least when BLE is enabled.  The solution trades high RAM usage in order to provide strict reliability via 100% DMA output.

#### Generally, how to enable a new nRF52840 board

1. Start by reviewing PR #1027, which enabled the Circuit Playground Bluefruit board.

2. Add a new section to platforms/arm/nrf52/fastpin_arm_nrf52_variants.h, based on whether `ARDUINO_SEEED_XIAO_NRF52840_SENSE` is defined.  (See [example](https://github.com/henrygab/FastLED/blob/1a273c8d3c99bbecccd9e2b1456516da8156415b/src/platforms/arm/nrf52/fastpin_arm_nrf52_variants.h#L52-L59))

3. Determine the Arduino pin number <--> nRF52840 Port/Pin mapping for each of the exposed / usable pins on the new board. 

4. For each exposed / usable pins, add a new line in that new section you've added in step 2.  Use the following form:
```C++
    _FL_DEFPIN( ARDUINO_PIN_NUMBER, NRF_PIN_NUMBER, NRF_PORT_NUMBER );
```

5.  Critically, test each pin that was added.  I think something along the lines of `FastPin<ARDUINO_PIN_NUMBER>::toggle()` in loop() (with delay of course) would let you test each pin number matches.  I've found errors in board documentation, so this is a **_critical_** step.

6. Test each of those added pins as a clockless data pin driving 100 pixels.  **_If this step is not performed**_, then please add a [warning that the board is currently untested](https://github.com/henrygab/FastLED/blob/1a273c8d3c99bbecccd9e2b1456516da8156415b/src/platforms/arm/nrf52/fastpin_arm_nrf52_variants.h#L149-L151).

7. Optionally, test at least a couple pins driving some APA102 (SPI ... data + clock) pixels.

henrygab, 2022-01-18:
Using https://wiki.seeedstudio.com/XIAO_BLE/ as a starting reference point, here's a strawman of what might work:

```C++

#if defined (ARDUINO_SEEED_XIAO_NRF52840_SENSE)
    #if defined(__FASTPIN_ARM_NRF52_VARIANT_FOUND)
        #error "Cannot define more than one board at a time"
    #else
        #define __FASTPIN_ARM_NRF52_VARIANT_FOUND
    #endif
    #if !defined(FASTLED_NRF52_SUPPRESS_UNTESTED_BOARD_WARNING)
        #warning "Seeed XIAO BLE and XIAO BLE Sense are untested boards -- test and let use know your results via https://github.com/FastLED/FastLED/issues"
    #endif
    _FL_DEFPIN(  0,  2, 0 ); //  D0 == P0.02
    _FL_DEFPIN(  1,  3, 0 ); //  D1 == P0.03
    _FL_DEFPIN(  2, 28, 0 ); //  D2 == P0.28
    _FL_DEFPIN(  3, 29, 0 ); //  D3 == P0.29
    _FL_DEFPIN(  4,  4, 0 ); //  D4 == P0.04
    _FL_DEFPIN(  5,  5, 0 ); //  D5 == P0.05
    // _FL_DEFPIN(  6, 11, 1 ); //  D6 == P1.11 // Don't enable Serial TX by default
    // _FL_DEFPIN(  7, 12, 1 ); //  D7 == P1.12 // Don't enable Serial RX by default
    _FL_DEFPIN(  8, 13, 1 ); //  D8 == P1.13
    _FL_DEFPIN(  9, 14, 1 ); //  D9== P1.14
    _FL_DEFPIN( 10, 15, 1 ); // D10 == P1.15
#endif  // ARDUINO_SEEED_XIAO_NRF52840_SENSE

henrygab, 2022-01-18:
If you submit a PR, would you mind also changing the currently hard-coded pixel limit, so that it's instead a hard-coded default?   Changing this to be user-definable will also help minimize RAM usage.

Just change:
```C++
#define FASTLED_NRF52_MAXIMUM_PIXELS_PER_STRING 144 // TODO: Figure out how to safely let this be calller-defined....
```

Into:
```C++
#if !defined(FASTLED_NRF52_MAXIMUM_PIXELS_PER_STRING)
    #define FASTLED_NRF52_MAXIMUM_PIXELS_PER_STRING 144
#endif // FASTLED_NRF52_MAXIMUM_PIXELS_PER_STRING
```

henrygab, 2022-01-18:
And... I found the [results](https://github.com/FastLED/FastLED/issues/917#issuecomment-573295953) of my prior look at enabling MBED OS support.

Short answer: 

It's non-trivial to add support for nRF52 boards that use MBED OS, because the code presumes the use of FreeRTOS.  I don't have the time (or hardware) to look at that currently.  

If you can get the Adafruit nRF52 bootloader working, and adapt the Adafruit nRF52 BSP for this board (likely just adding some variant headers), you might be able to run FreeRTOS on the board, which will allow the FastLED code to work.  That's likely much less work than re-implementing and re-testing the FastLED code.

Sorry!