FastLED/FastLED #748: parallel output to WS2812B ESP8266 (wemos d1)

alatdneg, 2019-03-03:
Hi is this supported?

Or likely to be, I see it should work for WS2811, but WS2812B isn't listed.

cheers
al

alatdneg, 2019-03-04:
```
#include <FastLED.h>

#define NUM_LEDS_PER_STRIP 16
#define NUM_STRIPS 4

CRGB leds[NUM_STRIPS * NUM_LEDS_PER_STRIP];

void setup() {
  LEDS.addLeds<WS2811_PORTA,NUM_STRIPS>(leds, NUM_LEDS_PER_STRIP);
  LEDS.setBrightness(32);
}

void loop() {
  CRGB colours[] = {CRGB::Red,CRGB::Green,CRGB::Blue,CRGB::Cyan};
  
  for(int i = 0; i < NUM_STRIPS; i++) {
    for(int j = 0; j < NUM_LEDS_PER_STRIP; j++) {
      leds[(i*NUM_LEDS_PER_STRIP) + j] = colours[i];
    }
  }

  LEDS.show();
  LEDS.delay(10);
}
```

focalintent, 2019-03-04:
The only currently supported timings for block output are WS2811,WS2813,TM1803, and UVS1903.   Also on the d1 it’s only supported for pins 12,11,13,10.    Also I’m not sure how well tested the esp8266 parallel output is.

alatdneg, 2019-03-04:
Yep I assumed it wasn't supported, I only wanted to check as some user seemed to imply it worked, it's possible they were listing the led type incorrectly or I misread their posts. 

Is it likely to be added as the WS2812B is very common ? Pretty much all the 3-wire leds I see are using it.

focalintent, 2019-03-04:
Try WS2811 and WS2813 - the timings are really close (and more problems are caused by the lack of a decent 5v level shifter than the clock - but all three use a 1.25us long pulse cycle for a bit). 

I’m in the middle of completely rewriting how all this is managed (among other things - fitting in as life/work allows), so I’m not putting a lot of work into the current master outside of significant bug fixes or code folks submit that’s clean to add.

alatdneg, 2019-03-04:
Using one of these 74AHCT125, which worked fine when not using parallel, So I guess the other chipsets are just not quite close enough.

Yep fair enough, looking forward to the new version (*RGBW* :)).

 I'll close this ticket, and thank you for all the hardwork, this library has allowed me to create a whole bunch of fun projects.

closed 2019-03-04 by alatdneg

alatdneg, 2019-03-04:
Whut... It worked.. I just removed the GRB flag I'd added on the end..

alatdneg, 2019-03-04:
Dumb.. I put the RGB_ORDER FLAG in the function call, instead of in the template..
Really sorry for wasting your time.. I'd removed if from the example but forgot to run it.