FastLED/FastLED #749: Time Conzcept

pglu, 2019-03-04:
Hi,
For our project [PGLU.CH](https://pglu.ch) we want to make FastLed commands available in our block language.
Now we are looking for a time concept that works with millis() instead of delay().

In the code below we have implemented this, but it doesn't work as desired: the time intervals are irregular and too long. We can live with slowness, but the irregular intervals make the code unusable.

FastLed Version: 3.2.0

Hardware: 
Arduino Nano
LED strip WS2812B with 10 pixels

```
#include <FastLED.h>

int i = 0;

#define NUM_LEDS (10)
#define DATA_PIN 9
#define BRIGHTNESS constrain(map(100, 0, 100, 0, 255), 0, 255)
CRGB leds[NUM_LEDS];

void setup() {

      FastLED.addLeds<NEOPIXEL, DATA_PIN>(leds, NUM_LEDS);
      FastLED.setBrightness(BRIGHTNESS );

}

void loop() {
   

   if (int(millis()) % int(100) == 0) {
         fill_solid( &(leds[(random(0, 9))]), (1), CHSV( (random(0, 255)), 255, 255) );

   }
   for (i = 0; i <= 9; i++) {
        leds[i].fadeToBlackBy( 1 );
  }
         FastLED.show();

}
```

JensGrabner, 2019-03-04:
You can use this : https://www.avdweb.nl/arduino/timing/virtualdelay

focalintent, 2019-03-04:
Also writing ws2812 data required disabling interrupts - which is how the clock time (reported by millis()) advances - FastLED tries to adjust the clock to account for this but it is not exact.  

If you need exact timing you will need an external RTC clock.  

Virtualdelay is going to have the same issue if it is, at its core, based on millis().

focalintent, 2019-03-04:
Also, the way you are using millis means means it will only fire if you happen to check the value of millis() in exactly a multiple of 100ms - which means you run a high risk of missing it.  

FastLED provides macros like EVERY_N_MILLIS which you can use like this:

```
void loop() {
    EVERY_N_MILLIS(100) {
        // fill solid
    }
    // do your dim, FastLED.show($
}
```

pglu, 2019-03-04:
This explains the gaps I've observed. Thanks for the hint, I will now experiment with EVERY_N_MILLIS.

pglu, 2019-03-04:
Is this true? ..
..I can only use one EVERY_N_MILLIS(1000) command..
..I can use several EVERY_N_SECONDS(1) commands..

..in the same mainloop

focalintent, 2019-03-04:
You should be able to use as many as you want

closed 2019-03-08 by pglu