FastLED/FastLED #754: not compiling on platformio, when selecting nodemcuv2

gnkarn, 2019-03-13:
in this project im using a nodemcu , apa102 leds , and an apds9960 .

if i use D1mini as the set on patformio it compiles ok,
but when using nodemcuv2 ( ESP12F) , i get the following error
` required`` from 'CLEDController& CFastLED::addLeds(CRGB*, int, 
int) [with ESPIChipsets CHIPSET = (ESPIChipsets)5u; unsigned char DATA_PIN = 14u; unsigned char CLOCK_PIN = 13u; EOrder RGB_ORDE
R = (EOrder)136u; unsigned char SPI_DATA_RATE = 4u]'
src/main.cpp:464:79:   required from here
/Users/Gus-nuevo/.platformio/lib/FastLED_ID126/fastpin.h:207:2: error: static assertion failed: Invalid pin specified
static_assert(validpin(), "Invalid pin specified");`
 

i think is not accepting the pins selected , when using nodemcu , but it is when selecting D1mini , as im using a NodeMCU  ( and is working fine) , there should be not reason for claiming such error .
tks

focalintent, 2019-03-13:
FastLED does pin/port mapping on its own based on the definition of the platform, and if it doesn't have one it'll fall back to a default mapping -- you can see the mappings for different platforms that've been defined so far here - https://github.com/FastLED/FastLED/blob/master/platforms/esp/8266/fastpin_esp8266.h -- specifically, the nodemcu definition in FastLED only defines 11 pins, 0-10 

Since you're using pin 13 and 14, my guess is that you're already trying to use the raw PIN numbers (vs. how nodemcu was labeling them in older versions) - in which case adding a define for FASTLED_ESP8266_RAW_PIN_ORDER before you include FastLED will tell the library to use a direct mapping - e.g.:

```
#define FASTLED_ESP8266_RAW_PIN_ORDER 1
#include <FastLED.h>
```

focalintent, 2019-03-13:
(pin specification on esp8266 boards out there was/is kind of a nightmare, with a variety of board manufacturers using arbitrary numbers on the silkscreens for the boards (I guess to keep the external pin numbering nice and neat?) - I gave up trying to track all the variations, and it didn't help that for some of those variations, folks were ignoring the numbers on the b board and just going with the raw gpio pin numbers anyway)

LongLiveCHIEF, 2019-06-22:
@focalintent I took a look at the FastLED mappings for nodemcu, and I'm wondering why it only uses pins 0-10, (with a max pin value of 10), when pins 5 - 10 can't (shouldn't) be used for GPIO.

Even when defining nodemcu pin order before including FastLED, I still can't use pins higher than 10.

> pin specification on esp8266 boards out there was/is kind of a nightmare, with a variety of board manufacturers using arbitrary numbers on the silkscreens for the boards

I've used esp8266 boards and chips from half a dozen manufacturers, and they always match. You just need to know what esp8266 variant your're using. In fact, the numbers are so standardized, that the arduino framework [has them mapped](https://github.com/esp8266/Arduino/blob/master/variants/nodemcu/pins_arduino.h) so you can refer to them by the values on the silkscreens.

I can submit a PR to fix these issues, but want to make sure I'm not missing something first?

LongLiveCHIEF, 2019-06-22:
With the `Arduino.h`, included, you can use the `D#` to refer to matching silkscreen labels (ex... D0 = pin 16 = GPIO 16). However, you actually have to define `FASTLED_ESP_8266_RAW_PIN_ORDER` to get this behavior.

It is very counter-intuitive, because the wiki appears to have the assumptions backwards:

> There appear to be two major types of pin layouts for ESP8266 boards. There's a raw layout, where pin 0 is GPIO0, pin 1 is GPIO1, etc... up to pin 16 being GPIO16. Then there's the NodeMCU layout where pin 0 is GPIO 16, pin 1 is GPIO 5, etc...

If you're using arduino framework (which I assume most are), then `D0 = GPIO16 = pin 16`. The NodeMCU layout in this case `pin 0 = GPIO0 = D3`. `D1 = GPIO5 =  pin 5`. _This_ is the NodeMCU mapping.

I don't think there are _board_ vendors that do pin layouts differently, I think it's the _frameworks_ that map out the pins differently. The pin labels are standard for all esp8266 variants (each variant has same labels regardless of manufacturer), however the platform/framework maps the pins differently so as to support different boards without any code changes required by the user.

I'm not sure if this means the approach to the pin order definitions needs to change or not.

If i have the following code _on a NodeMCU board_ :

```
#include <Arduino.h>
#include <FastLED.h>

#define LED_DATA D1 
```
My data will be sent from the pin _labeled on the board_ as `D5` instead of `D1`.  It's a _NodeMCU_ board, but I have to tell FastLED it's a "RAW" board, which makes absolutely no sense at all.

LongLiveCHIEF, 2019-06-22:
P.S. It was an interesting use of a Saturday digging through FastLED and Arduino code, and vendor datasheets for esp8266 and nodemcu to figure this all out. :smile:

LongLiveCHIEF, 2019-06-22:
Update: neither pin order definition will send data out the desired pin when using `D#` pin references. 

Using `D1` in my code, has the same result for all 3 attempts (setting raw and nodemcu pin orders as well as letting it fallback).

:thinking:

focalintent, 2019-06-23:
Generally speaking - don’t use Dxx pins for FastLED (they aren’t always constants - depending on platform or library version) - it’s why none of the FastLED examples use Dxx for a pin.  

As far as the silkscreening and PIN numbers and such - this stuff changed around a couple of times with the early esp8266 arduino platforms and I haven’t had a chance to get everything fully in line with the current state of the world (but I absolutely have multiple esp8266 boards where the silk screen’d PIN numbers map to different sets of low level gpio entries)

LongLiveCHIEF, 2019-06-23:
Yeah, that's fine... my bigger concern though is that it opens up a loophole to use pins that FastLED "disables" because they're used for flash/boot. Probably should be a different issue.

closed 2019-08-12 by focalintent