FastLED/FastLED #756: 8-bit HSV color animations produce poor results & blinking

itay-grudev, 2019-03-15:
While very efficient both in terms of storage and computation speed `8-bit` HSV colour is a horrible idea. That equates to a HUE step of `1.4°`, which produces a difference of `4` bits per colour component (`8` in total in an animation) in RGB space. This is the same as doing `64` steps out of `256` per colour component. This only gets worse after setting colour correction factors which further decreases the available colour resolution.

The end result is HSV animation that is visibly blinking. The difference of `8` between steps is too large and the resulting animation is not smooth.

To solve this I opted for a custom HSV function that uses higher resolution, but that means that the HSV utilities provided by FASTLed are useless as they produce very poor results.

I believe that for the sake of efficiency you've decreased the resolution so much that it is no longer practically useful. At the moment an upgrade to a `16-bit` hue is required.

The colour conversion algorithm is simple and  a `16-bit` hue would cause minimap performance penalty, but would produce significantly smoother animations.

P.S. This issue is similar to, but not the same as #664.

kriegsman, 2019-03-15:
Glad you found a good work around. I use custom 16bit HSV sometimes myself too when I need more detail. 

FULL 16-bit support is coming in a future release. I’m going to close this ticket, as we already have the work planned, but you should know that we are definitely going to go 16-bit.

closed 2019-03-15 by kriegsman

itay-grudev, 2019-03-16:
@kriegsman Is there an ETA?

itay-grudev, 2019-03-16:
@kriegsman Do you mind keeping it open until it's resolved? It's easier to find this way.

kriegsman, 2019-03-16:
We don’t have an ETA date for full-16-bit support, no. And let me confer with Dan about how we want to organize and track requests for “roadmap work” that we already have laid out.  I’ll let you know here either way as soon as we discuss that and come to some plan for how we want to do it.

focalintent, 2019-03-16:
It’s already in here as #11