FastLED/FastLED #757: Controller stops responding

nicexe, 2019-03-19:
I have an RF remote control receiver with 4 data pins and a trigger pin. Whenever you press a button on the remote the trigger pin and the corresponding data pin are pulled high.

I have set the trigger pin as an interrupt on pin 2.

I am using 3 monochrome LED stripes on the same controller. One for each channel.

Here is my code:  
```c++
#include <FastLED.h>

#define NUM_LEDS 1
#define DATA_PIN 5
#define CLOCK_PIN 4

#define DELAY_SECTION 100

CRGB leds[NUM_LEDS];

const byte interruptPin = 2;

volatile byte pattern = 0;

const int CONTROL_A = 8;
const int CONTROL_B = 6;
const int CONTROL_C = 9;
const int CONTROL_D = 7;

void setPattern() {
  if (digitalRead(CONTROL_B)) {
    pattern = 1;
  } else if (digitalRead(CONTROL_C)) {
    pattern = 2;
  } else if (digitalRead(CONTROL_D)) {
    pattern = 3;
  } else {
    pattern = 0;
  }
}

void patternA() {
  leds[0] = 0xFF0040;
  FastLED.show();
  
  if (pattern != 0) return;
  delay(DELAY_SECTION);
  if (pattern != 0) return;
  
  leds[0] = 0xFF4000;
  FastLED.show();
  
  if (pattern != 0) return;
  delay(DELAY_SECTION);
  if (pattern != 0) return;
  
  leds[0] = 0x40FF00;
  FastLED.show();
  
  if (pattern != 0) return;
  delay(DELAY_SECTION);
  if (pattern != 0) return;
  
  leds[0] = 0x00FF40;
  FastLED.show();
  
  if (pattern != 0) return;
  delay(DELAY_SECTION);
  if (pattern != 0) return;
  
  leds[0] = 0x0040FF;
  FastLED.show();
  
  if (pattern != 0) return;
  delay(DELAY_SECTION);
  if (pattern != 0) return;
  
  leds[0] = 0x4000FF;
  FastLED.show();
  
  if (pattern != 0) return;
  delay(DELAY_SECTION);
  if (pattern != 0) return;
}

void patternB() {
//  leds[0] = 0xFF4000;
//  FastLED.show();
//
//  if (pattern != 0) return;
//  delay(DELAY_SECTION);
//  if (pattern != 0) return;
//  
//  leds[0] = 0x4000FF;
//  FastLED.show();
//
//  if (pattern != 0) return;
//  delay(DELAY_SECTION);
//  if (pattern != 0) return;
//  
//  leds[0] = 0x00FF40;
//  FastLED.show();
//
//  if (pattern != 0) return;
//  delay(DELAY_SECTION);
//  if (pattern != 0) return;

// at this point patternB() is functionally the same as patternA()
// I'm doing this to sanity-check my code (and my brain-logic)

  leds[0] = 0xFF0040;
  FastLED.show();
  
  if (pattern != 0) return;
  delay(DELAY_SECTION);
  if (pattern != 0) return;
  
  leds[0] = 0xFF4000;
  FastLED.show();
  
  if (pattern != 0) return;
  delay(DELAY_SECTION);
  if (pattern != 0) return;
  
  leds[0] = 0x40FF00;
  FastLED.show();
  
  if (pattern != 0) return;
  delay(DELAY_SECTION);
  if (pattern != 0) return;
  
  leds[0] = 0x00FF40;
  FastLED.show();
  
  if (pattern != 0) return;
  delay(DELAY_SECTION);
  if (pattern != 0) return;
  
  leds[0] = 0x0040FF;
  FastLED.show();
  
  if (pattern != 0) return;
  delay(DELAY_SECTION);
  if (pattern != 0) return;
  
  leds[0] = 0x4000FF;
  FastLED.show();
  
  if (pattern != 0) return;
  delay(DELAY_SECTION);
  if (pattern != 0) return;
}

void patternC() {
  
}

void patternD() {
  
}

void setup() {
  pinMode(interruptPin, INPUT);
  
  FastLED.addLeds<P9813, DATA_PIN, CLOCK_PIN, RGB>(leds, NUM_LEDS);

  leds[0] = CRGB::Black;
  FastLED.show();

  pinMode(CONTROL_A, INPUT);
  pinMode(CONTROL_B, INPUT);
  pinMode(CONTROL_C, INPUT);
  pinMode(CONTROL_D, INPUT);

  attachInterrupt(digitalPinToInterrupt(interruptPin), setPattern, RISING);

  // just for debugging
  Serial.begin(9600);
}

void loop() {
  // just for debugging
  Serial.print("pattern: ");
  Serial.println(pattern);

  FastLED.clear();
  
  switch(pattern) {
    case 1: patternB();
            break;
    case 2: patternC();
            break;
    case 3: patternD();
            break;
    default: patternA();
  }
}
```

Under loop() I'm basically checking which pattern should be running.

In this video I try changing patterns with the rf remote. Note that patternA() and patternB() are functionally the same just to debug this problem. Switching from pattern A to pattern B cause the animation to halt. Switching from any pattern to pattern A makes the animation come back again.

The serial output shows which pattern should be running

Now here is the video:
[LED controlling problem](https://youtu.be/kDjGJgFb89o)

Is there a problem with my code or is this an issue with FastLED?

focalintent, 2019-03-19:
Your function for patternB is littered with:

```
if (pattern != 0) return;
```

But pattern is going to be 1 for patternB, and 0 for patternA - this is why it stops doing anything when you select pattern B.

nicexe, 2019-03-19:
Oh! You are absolutely right!

closed 2019-03-19 by nicexe