FastLED/FastLED #758: Best Discussion will vanish when G+ goes down on April 2

mweinelt, 2019-03-19:
Google+ is about to go down on 2nd of April 2019. This means most discussions linked at https://github.com/FastLED/FastLED/wiki/Best-of-FastLED-Discussions will vanish.

Might be a good time to persist them elsewhere.

marmilicious, 2019-03-20:
@focalintent  I updated the the G+ links to the posts on the makerforums.   Here's a paste you can use to update the wiki.  Or if there's a way to allow me to edit the wiki I'll do so.
https://pastebin.com/Y02dWtHs

I put a temp version of this update here.
https://github.com/marmilicious/FastLED_examples/wiki/Best-of-FastLED-Discussions

mweinelt, 2019-03-21:
The video here is not working for me, can someone confirm that? https://forum.makerforums.info/t/anti-aliasing-example-this-example-shows-two-bars-of-light-moving-at-the-same-speed/60227

marmilicious, 2019-03-21:
@mweinelt   Yes I can confirm it's not viewable, probably because it was a google video and not a link to youtube.  
(Try running the code and you will be able to see it live!  ;P )

MartyMacGyver, 2019-04-04:
Whoever has the ability to edit the main project description as seen on https://github.com/FastLED/FastLED should remove the G+ links as they are completely broken now.

closed 2019-05-13 by focalintent