FastLED/FastLED #759: ESP32 + APA102 Question

Edris89, 2019-03-20:
Hi there,

I am doing a project with the ESP32 and the APA102 SMD RGB LEDs.
My project is about a text logo its called VUSAM, 5 letters.

I wanted to fill those letters with SMDs and have each letter it's own channel so that it could have different effects for each letter.

I understand that each letter would have it's own data pin. 
But the APA102 has also a clock pin. 
And this is where it's getting a bit confusing for me.

I have some couple of questions. 
1. Which gpio pins can I use for the data pins for each of the 5 channels?
2. Must I also have 5 different clock pins for each channel? Or can I connect all those clocks with one clock pin?
3. Question about question 2. What's the difference between using 5 different clock pins or 1 clock pin for all the 5 channels.
4. Which pin or pin's can I use for the clock pin.

Thanks in advance.

JanBosNL, 2019-03-22:
I was trying to figure similar issues out on an esp8266 and can confirm you only need 1 clock pin for all the data pins.

Edris89, 2019-03-30:
Link is broken?

JanBosNL, 2019-04-03:
Check my latest working code, Its runs on an ESP8266 but should also be fine on an esp32

Dont yet know how to apply different effects on different strips. This for me however solved the different ledcounts on strips when using 1 effect on all strips at once.

[](https://github.com/JanBosNL/NodeMCU-esp8266-fastled-apa102/blob/master/5_strips_multi_lenght_fastled_fire_demo.ino)

closed 2019-08-12 by focalintent