FastLED/FastLED #761: Fastled is not compiling for NodeMCU 1.0 (ESP-12E Module).

JanBosNL, 2019-03-22:
Seems Fastled is not compiling through the arduino IDE for the NodeMCU 1.0 (ESP-12E Module).

I wrote and tested a simple sketch that is working on the WEMOS D1 R1 (also esp8266 based)

When I try and compile te same sketch for the LOLIN NodeMCU V3 with the 1.0 (ESP-12E Module) I get this Error.

> exit status 1
> Error compiling for board NodeMCU 1.0 (ESP-12E Module).

Below the sketch I wrote to test 2 ledstrips on 2 data pins.
```

#include "FastLED.h"
#define DATA_PIN_A  12 //d6 on nodemcu esp8266
#define DATA_PIN_B  13 //d7 on nodemcu esp8266
#define CLK_PIN     14 //d5 on nodemcu esp8266
#define LED_TYPE  APA102
#define COLOR_ORDER  BGR
#define NUM_LEDS  30 // per strand
CRGB leds_A [NUM_LEDS];
CRGB leds_B [NUM_LEDS];

void setup() {
  delay(3000); // 3 second delay for recovery
  FastLED.addLeds<LED_TYPE,DATA_PIN_A,CLK_PIN,COLOR_ORDER>(leds_A, NUM_LEDS).setCorrection(TypicalLEDStrip);
  FastLED.addLeds<LED_TYPE,DATA_PIN_B,CLK_PIN,COLOR_ORDER>(leds_B, NUM_LEDS).setCorrection(TypicalLEDStrip);
  FastLED.setBrightness(96);
}

void loop() { 
  // Fill leds with rainbow
  fill_rainbow(leds_A, NUM_LEDS, millis());
  FastLED.show();
  delay(2000);

  fill_solid(leds_B, NUM_LEDS, CRGB::Yellow);
  FastLED.show();
  delay(2000);

  // Turn leds off
  fill_solid(leds_A, NUM_LEDS, CRGB::Black);  // Set all leds black
  FastLED.show();
  delay(1000);
  // Turn leds off
  fill_solid(leds_B, NUM_LEDS, CRGB::Black);  // Set all leds black
  FastLED.show();
  delay(1000);
  
}
```

focalintent, 2019-03-23:
I’d need to see the full build output - but I suspect this is a compiler bug

JanBosNL, 2019-03-23:
Arduino: 1.8.9 (Windows 10), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, Flash, Disabled, 4M (no SPIFFS), v2 Lower Memory, Disabled, None, Only Sketch, 115200"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\HP\AppData\Local\Arduino15\packages -hardware C:\Users\HP\Documents\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\HP\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\HP\Documents\Arduino\libraries -fqbn=esp8266:esp8266:nodemcuv2:xtal=80,vt=flash,exception=disabled,eesz=4M,ip=lm2f,dbg=Disabled,lvl=None____,wipe=none,baud=115200 -vid-pid=1A86_7523 -ide-version=10809 -build-path C:\Users\HP\AppData\Local\Temp\arduino_build_358501 -warnings=none -build-cache C:\Users\HP\AppData\Local\Temp\arduino_cache_368860 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.esptool.path=C:\Users\HP\AppData\Local\Arduino15\packages\esp8266\tools\esptool\2.5.0-3-20ed2b9 -prefs=runtime.tools.esptool-2.5.0-3-20ed2b9.path=C:\Users\HP\AppData\Local\Arduino15\packages\esp8266\tools\esptool\2.5.0-3-20ed2b9 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\Users\HP\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-3-20ed2b9 -prefs=runtime.tools.xtensa-lx106-elf-gcc-2.5.0-3-20ed2b9.path=C:\Users\HP\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-3-20ed2b9 -prefs=runtime.tools.mkspiffs.path=C:\Users\HP\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\2.5.0-3-20ed2b9 -prefs=runtime.tools.mkspiffs-2.5.0-3-20ed2b9.path=C:\Users\HP\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\2.5.0-3-20ed2b9 -verbose C:\Users\HP\Documents\Arduino\sketch_mar22a\sketch_mar22a.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\HP\AppData\Local\Arduino15\packages -hardware C:\Users\HP\Documents\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\HP\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\HP\Documents\Arduino\libraries -fqbn=esp8266:esp8266:nodemcuv2:xtal=80,vt=flash,exception=disabled,eesz=4M,ip=lm2f,dbg=Disabled,lvl=None____,wipe=none,baud=115200 -vid-pid=1A86_7523 -ide-version=10809 -build-path C:\Users\HP\AppData\Local\Temp\arduino_build_358501 -warnings=none -build-cache C:\Users\HP\AppData\Local\Temp\arduino_cache_368860 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.esptool.path=C:\Users\HP\AppData\Local\Arduino15\packages\esp8266\tools\esptool\2.5.0-3-20ed2b9 -prefs=runtime.tools.esptool-2.5.0-3-20ed2b9.path=C:\Users\HP\AppData\Local\Arduino15\packages\esp8266\tools\esptool\2.5.0-3-20ed2b9 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\Users\HP\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-3-20ed2b9 -prefs=runtime.tools.xtensa-lx106-elf-gcc-2.5.0-3-20ed2b9.path=C:\Users\HP\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-3-20ed2b9 -prefs=runtime.tools.mkspiffs.path=C:\Users\HP\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\2.5.0-3-20ed2b9 -prefs=runtime.tools.mkspiffs-2.5.0-3-20ed2b9.path=C:\Users\HP\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\2.5.0-3-20ed2b9 -verbose C:\Users\HP\Documents\Arduino\sketch_mar22a\sketch_mar22a.ino
Using board 'nodemcuv2' from platform in folder: C:\Users\HP\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0
Using core 'esp8266' from platform in folder: C:\Users\HP\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0
Detecting libraries used...
"C:\\Users\\HP\\AppData\\Local\\Arduino15\\packages\\esp8266\\tools\\xtensa-lx106-elf-gcc\\2.5.0-3-20ed2b9/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\\Users\\HP\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.5.0/tools/sdk/include" "-IC:\\Users\\HP\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.5.0/tools/sdk/lwip2/include" "-IC:\\Users\\HP\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.5.0/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:\\Users\\HP\\AppData\\Local\\Temp\\arduino_build_358501/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-rtti -falign-functions=4 -std=c++11 -ffunction-sections -fdata-sections -fno-exceptions -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536 -DLWIP_FEATURES=1 -DLWIP_IPV6=0 -DARDUINO=10809 -DARDUINO_ESP8266_NODEMCU -DARDUINO_ARCH_ESP8266 "-DARDUINO_BOARD=\"ESP8266_NODEMCU\"" -DFLASHMODE_DIO -DESP8266 "-IC:\\Users\\HP\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.5.0\\cores\\esp8266" "-IC:\\Users\\HP\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.5.0\\variants\\nodemcu" "C:\\Users\\HP\\AppData\\Local\\Temp\\arduino_build_358501\\sketch\\sketch_mar22a.ino.cpp" -o nul
"C:\\Users\\HP\\AppData\\Local\\Arduino15\\packages\\esp8266\\tools\\xtensa-lx106-elf-gcc\\2.5.0-3-20ed2b9/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\\Users\\HP\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.5.0/tools/sdk/include" "-IC:\\Users\\HP\\AppData\\Local\\Arduino15\\ [... cut ...]
C:\Users\HP\Documents\Arduino\libraries\FastLED/FastLED.h:253:77:   required from 'static CLEDController& CFastLED::addLeds(CRGB*, int, int) [with ESPIChipsets CHIPSET = (ESPIChipsets)6u; unsigned char DATA_PIN = 13u; unsigned char CLOCK_PIN = 14u; EOrder RGB_ORDER = (EOrder)136u]'

C:\Users\HP\Documents\Arduino\sketch_mar22a\sketch_mar22a.ino:14:76:   required from here

C:\Users\HP\Documents\Arduino\libraries\FastLED/fastpin.h:207:2: error: static assertion failed: Invalid pin specified

Using library FastLED at version 3.2.6 in folder: C:\Users\HP\Documents\Arduino\libraries\FastLED 
exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).

JanBosNL, 2019-03-23:
p.s. When I compile anything else for the LOLIN NodeMCU V3 with the 1.0 (ESP-12E Module) I dont get the compiling error.

I tried the adafruit Dotstar strandtest -> Working
I tried the APA102 Example by Polulu [https://github.com/pololu/apa102-arduino](url) -> Working

Only Fastled does not want to compile, and I realy want to use the best library there is = Fastled!

If you need anything else I will try my best to suply info. I am a bit of a noob though so you might have to guide me a bit to get more debug info etc...

focalintent, 2019-03-23:
This error:

```
C:\Users\HP\Documents\Arduino\libraries\FastLED/fastpin.h:207:2: error: static assertion failed: Invalid pin specified

static_assert(validpin(), "Invalid pin specified");

^

C:\Users\HP\Documents\Arduino\libraries\FastLED/fastpin.h: In instantiation of 'class FastPin<14u>':
```

Is what’s wrong - https://github.com/FastLED/FastLED/wiki/ESP8266-notes has some notes about PIN numbers and mappings and such

JanBosNL, 2019-03-24:
Thank you so much,

I instantly found the solution in the link you gave me.

Apparently I needed to difine `#define FASTLED_ESP8266_NODEMCU_PIN_ORDER`
To tell Fastled I am using the LOLIN NodeMCU V3 with the 1.0 (ESP-12E Module) 

I would never have figure that out myself, since I thought choosing this board in the arduino board manager would be sufficient.

Just now had a LOLIN NodeMCU V3 with the 1.0 (ESP-12E Module) laying around when I read your reply, plugged it in, tried compiling the same code I did before with the additional `#define FASTLED_ESP8266_NODEMCU_PIN_ORDER` code, And it works.

```
#include <FastLED.h>

#define FASTLED_ESP8266_NODEMCU_PIN_ORDER // The solution !!!!!!!!!!!!!!!!!!
#define DATA_PIN_A  12 //d6 on nodemcu esp8266
#define DATA_PIN_B  13 //d7 on nodemcu esp8266
#define CLK_PIN     14 //d5 on nodemcu esp8266
#define LED_TYPE  APA102
#define COLOR_ORDER  BGR
#define NUM_LEDS  30 // per strand
CRGB leds_A [NUM_LEDS];
CRGB leds_B [NUM_LEDS];

void setup() {
  delay(3000); // 3 second delay for recovery
  FastLED.addLeds<LED_TYPE,DATA_PIN_A,CLK_PIN,COLOR_ORDER>(leds_A, NUM_LEDS).setCorrection(TypicalLEDStrip);
  FastLED.addLeds<LED_TYPE,DATA_PIN_B,CLK_PIN,COLOR_ORDER>(leds_B, NUM_LEDS).setCorrection(TypicalLEDStrip);
  FastLED.setBrightness(96);
}

void loop() { 
  // Fill leds with rainbow
  fill_rainbow(leds_A, NUM_LEDS, millis());
  FastLED.show();
  delay(2000);

  fill_solid(leds_B, NUM_LEDS, CRGB::Yellow);
  FastLED.show();
  delay(2000);

  // Turn leds off
  fill_solid(leds_A, NUM_LEDS, CRGB::Black);  // Set all leds black
  FastLED.show();
  delay(1000);
  // Turn leds off
  fill_solid(leds_B, NUM_LEDS, CRGB::Black);  // Set all leds black
  FastLED.show();
  delay(1000);
  
}
```
For refference, the working code with the solution.

Thank you for your time,

closed 2019-03-24 by JanBosNL