FastLED/FastLED #764: Error compilation in Calibrate Example  LDP6803

ut01927, 2019-04-03:
I am trying to extend the life of my old LDP6803, taking advantage of the fact that it has returned in recent versions of the library.
I started with the most basic, the example of Calibrate.

The problem is that in the compilation gives the following error:

_________________________________
error: could not convert template argument 'LPD6803' to 'EBlockChipsets'
FastLED.addLeds<LPD6803, RGB>(leds, NUM_LEDS); 
exit status 1
no matching function for call to 'CFastLED::addLeds(CRGB [6], int)'
_________________________________

I have tried the same example with other LEDs and it works perfectly.

Does anyone know what is happening?
My old LPD6803 thanks you

focalintent, 2019-04-03:
You don’t mention what board you are trying to build for - but try explicitly setting the data and clock pins.  Also make sure you have the most recent version of the library, LPD6803 support was only recently re-added.

ut01927, 2019-04-04:
Sorry!!!!
My Board is D1 mini but my LDP6803 are very old 3 or 4 years old.  It sure does not work.
Thanks for your time!
Greeting

focalintent, 2019-04-04:
With the D1 you need to explicitly set the data and clock pins, for example:

```
#define DATA_PIN 5
#define CLOCK_PIN 6
...
  FastLED.addLeds<LPD6803, DATA_PIN, CLOCK_PIN, RGB>(leds, NUM_LEDS);
```

ut01927, 2019-04-05:
Focalintent Thanks. I'll try

closed 2019-08-12 by focalintent

DRSDavidSoft, 2023-08-26:
Sorry to bug everyone here, but I just found out that LPD6803 was added back in #701, why there wasn't any news about this 😅