FastLED/FastLED #766: FASTLED_NO_PINMAP -- What's it for?

henrygab, 2019-04-06:
**_Can you help me understand the purpose and reasoning behind FASTLED_NO_PINMAP?_**

I'm walking through the [steps](http://fastled.io/docs/3.1/md__p_o_r_t_i_n_g.html) to port FastLED.  I'm normally able to divine(*) the purpose of the `#define` options.  However, I'm stumped as to the purpose of `FASTLED_NO_PINMAP`.

The only platforms that `#define FASTLED_NO_PINMAP` are STM32 and NRF51.

There are only two conditional compilations, both in fastpin.h.

The first wraps a *NEARLY* identical definition of the `Pin` class.  When `FASTLED_NO_PINMAP` is defined, the class has the following differences:
1. private `_init()` function sets `mPinMask`, `mPort`, and `mInPort` to `NULL` instead of using `digitalPinToBitMask()`, `portOutputRegister(digitalPinToPort(mPin))`, and  `portInputRegister(digitalPinToPort(mPin))` (respectively)
2. public functions `setOutput()` and `setInput()` are given empty function bodies
3. additional public function added: `setPin(int pin)` (sets member and calls `_init()`)
4. removes public functions `strobe()` and `toggle()`
5. removes public functions `void hi(register port_ptr_t port)` and `void lo(register port_ptr_t port)`

Otherwise, the classes are identical.  Based on the above, one guess I came up with is that, when porting to a new board, one would sub-class the above Pin class, and override the functions `_init()`, `setOutput()`, and `setInput()`.  But, I can't find where the Pin class would actually be instantiated in FastLED.  Thus, it seems unsatisfactory.

My other guess is that this is to enable support for non-Arduino environments, by providing the `digitalPinToBitMask()`,  `digitalPinToPort()`, `portOutputRegister()`, and `portInputRegister()` capabilities that Arduino would normally provide.  But, this also seems somewhat unsatisfactory.

Your help understanding this will be appreciated... 

(*) the brute-force way, reviewing the code where they are checked.  A slow way to learn the code base, but it's a start.

focalintent, 2019-04-06:
FASTLED_NO_PINMAP is mostly a thing I use to keep things building when on platforms that don't provide the portOutputRegister and friend functions.

Realistically though, it's moot - nothing in FastLED uses the Pin classes, with the exception of some old debugging code in the power management code (in fact in the overhaul I'm doing for RGBW and friends, I won't even have those classes around anymore).

The STM32 and NRF51 ports were both started in environments with no Arduino around (they were contract jobs for private, other build environment ports of FastLED), which is why they have it defined.

focalintent, 2019-04-06:
(The FastPin classes/implementations are the important ones that FastLED uses)

closed 2019-04-06 by focalintent

henrygab, 2019-04-06:
Thanks for taking the time to help me understand.  It helps me retain some sanity, knowing that this really and truly isn't (really) being used.  Moving on...