FastLED/FastLED #768: Text on matrix

machmar, 2019-04-07:
Text on matrix with serpentine layout? Is that possible without using different library?

X-WL, 2019-04-07:
That's possible. To do this you will have to: 
To write text, program presets of letters for the entire alphabet. 
Write a function to invert even rows / columns and run it before calling FastLED.show()

machmar, 2019-04-07:
yea, I figured that out but its quite complex and I dont really know how to make the text and invert parts

X-WL, 2019-04-07:
> 
> 
> yea, I figured that out but its quite complex and I dont really know how to make the text and invert parts

Then you should think about using ready-made libraries. For example: https://github.com/marcmerlin/SmartMatrix_GFX

closed 2024-09-06 by machmar