FastLED/FastLED #771: 8 RMT Channels Borks with PlatformIO only on ESP32

davepl, 2019-04-13:
I have an app that's pretty demanding - it's doing LZ decompression on one core of a stream of WiFI data received on the other, and drawing it using FastLED on 8 channels at once using RMT.  So I'm really pushing things, but after some careful balancing of the load, all works well, at least if I build it using the Arduino IDE or the VSCode wrapper thereof.

If I compile it with PlatformIO (still Arduino Core, not ESPIDF) then it gets corruption, if anything at all, when talking to the LEDs.  Basically FastLED.show() doesn't work.  I recompile the same files with Arduino IDE, it works.  

I appear to be using the same version of FastLED and other libraries that I inspected, so it almost seems like a compiler issue.  Other than "I have a repro of this" I'm not sure what to do from here, but wondering if anyone else has had issues with PlatformIO and multiple channels on the ESP32?

Naturally, a simple case of one or two channels works fine...

davepl, 2019-04-13:
I found the difference - board config in PlatformIO had PSRAM enabled, so that made for at least one different compiler flag, which I changed.  That fixed it, even though the board did not have PSRAM and I was not calling PSRAMInit() or anything!

So, long story short, you can't even BUILD with PSRAM support enabled and have it work with >4 channels at once.  That should be an easy repro, but how urgent it makes it I have no idea...

outlandnish, 2019-09-12:
@davepl what was the flag you disabled in PlatformIO to get it to work? I can't get it to work with more than 2 channels before it borks.

davepl, 2021-10-12:
Two later:  the two flags are

    -DBOARD_HAS_PSRAM
    -mfix-esp32-psram-cache-issue

BeldrothTheGold, 2021-10-31:
FYI, I believe this is being caused something in the FastLED esp32 interrupt handler accessing the flash while the flash cache is disabled. 

If you remove `ESP_INTR_FLAG_IRAM |`  from the esp_intr_alloc call in clockless_rmt_esp32.cpp:132, I stop seeing the intermittent crashes that Dave's talking about. 

I found this discussed here: 

https://github.com/espressif/esp-idf/issues/6164#issuecomment-734262777

BeldrothTheGold, 2021-11-01:
Trouble is removing ESP_INTR_FLAG_IRAM causes issues with timing stability for driving LEDs, since the inturrupt doesn't run while the flash is locked. I am going to see if I can hunt down what the compiler is storing in flash and see if we can't fix the issue by adding DRAM_ATTR to the declaration.

BeldrothTheGold, 2021-11-01:
I am seeing the below stack trace when I have more than a couple of channels enabled. Is this the issue you're seeing too? Or is this another issue entirely? 

```
Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed)
Core 1 register dump:
PC      : 0x4016f9ac  PS      : 0x00060734  A0      : 0x80081434  A1      : 0x3ffc0250  
A2      : 0x3ffc4fa0  A3      : 0x00000002  A4      : 0x8008c658  A5      : 0x3ffba150  
A6      : 0x00000000  A7      : 0x00000002  A8      : 0x800813de  A9      : 0x3ffc0230  
A10     : 0x00000002  A11     : 0x00000000  A12     : 0x00000005  A13     : 0x3ffba130  
A14     : 0x3ff000dc  A15     : 0x00000001  SAR     : 0x0000001a  EXCCAUSE: 0x00000007  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  
Core 1 was running in ISR context:
EPC1    : 0x40081559  EPC2    : 0x00000000  EPC3    : 0x4008a1ac  EPC4    : 0x4016f9ac

ELF file SHA256: 0000000000000000

Backtrace: 0x4016f9ac:0x3ffc0250 0x40081431:0x3ffc0270 0x400814da:0x3ffc0290 0x4008154e:0x3ffc02c0 0x40089b71:0x3ffc02e0 0x4008a1a9:0x3ffba1f0 0x4008730f:0x3ffba210 0x4008c556:0x3ffba230
  #0  0x4016f9ac:0x3ffc0250 in rmt_set_pin at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/driver/rmt.c:394
  #1  0x40081431:0x3ffc0270 in ESP32RMTController::startNext(int) at lib/FastLED/src/platforms/esp/32/clockless_rmt_esp32.cpp:430 (discriminator 4)
  #2  0x400814da:0x3ffc0290 in ESP32RMTController::doneOnChannel(rmt_channel_t, void*) at lib/FastLED/src/platforms/esp/32/clockless_rmt_esp32.cpp:430 (discriminator 4)
  #3  0x4008154e:0x3ffc02c0 in ESP32RMTController::interruptHandler(void*) at lib/FastLED/src/platforms/esp/32/clockless_rmt_esp32.cpp:430 (discriminator 4)
  #4  0x40089b71:0x3ffc02e0 in _xt_medint3 at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/xtensa_vectors.S:1323
  #5  0x4008a1a9:0x3ffba1f0 in spi_flash_op_block_func at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/spi_flash/cache_utils.c:203
  #6  0x4008730f:0x3ffba210 in ipc_task at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/ipc.c:62
  #7  0x4008c556:0x3ffba230 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)
  ```

davepl, 2021-11-01:
FWIW, I have in the past run 8 channels with this, both on the bench and for the umbrella project.  There could have been a regression since then I suppose!

 

-Dave

 

espressif/esp-idf#6164 (comment) <https://github.com/espressif/esp-idf/issues/6164#issuecomment-734262777>