FastLED/FastLED #776: Not working with ATMEGA32U4

jbonnett92, 2019-04-21:
Hi, 
I have a custom board active as a LilyPad, I upload the program via USBASP.
My code looks like this:
```
#include <FastLED.h>

#define ROTARY_A_PIN 8
#define ROTARY_B_PIN 9
#define ROTARY_SWITCH_PIN 4
#define ROTARY_RED_PIN 5
#define ROTARY_GREEN_PIN 11
#define ROTARY_BLUE_PIN 10
#define RGB_DATA_PIN 13
#define RGB_NUM_LEDS 4

bool rotary_switch_state = false;
unsigned long button_time = 0;  
unsigned long last_button_time = 0; 

int16_t rotary_red_val = 255;
int16_t rotary_green_val = 255;
int16_t rotary_blue_val = 0;
int rotary_alpha_val = 255;

struct CRGB leds[RGB_NUM_LEDS];

void setup() {
  pinMode(ROTARY_A_PIN, INPUT_PULLUP);
  attachInterrupt(digitalPinToInterrupt(ROTARY_A_PIN), rotaryEncoderIncrease, CHANGE);
  
  pinMode(ROTARY_B_PIN, INPUT_PULLUP);
  attachInterrupt(digitalPinToInterrupt(ROTARY_B_PIN), rotaryEncoderDecrease, CHANGE);

  pinMode(ROTARY_SWITCH_PIN, INPUT_PULLUP);
  attachInterrupt(digitalPinToInterrupt(ROTARY_SWITCH_PIN), rotaryEncoderSwitch, CHANGE);
  
  pinMode(ROTARY_RED_PIN, OUTPUT);
  pinMode(ROTARY_GREEN_PIN, OUTPUT);
  pinMode(ROTARY_BLUE_PIN, OUTPUT);

  FastLED.addLeds<NEOPIXEL, RGB_DATA_PIN>(leds, 
  RGB_NUM_LEDS).setCorrection(TypicalLEDStrip);
  FastLED.setBrightness(100);
}

void loop() {
  for (int i=0; i<RGB_NUM_LEDS; i++) {
    leds[i] = CRGB(255, 255, 255);
    FastLED.show();
  }
 
  if (rotary_switch_state) {
    rotary_alpha_val = 0;
  } else {
    rotary_alpha_val = 255;
  }
  setRGBA(rotary_red_val, rotary_green_val, rotary_blue_val, rotary_alpha_val);
  
  //rotaryBreathe();
}

void rotaryEncoderIncrease() {
  rotary_alpha_val += 25;
}

void rotaryEncoderDecrease() {
  rotary_alpha_val -= 25;
}

void rotaryEncoderSwitch() {
  button_time = millis();
  if (button_time - last_button_time > 250)
  {
    rotary_switch_state = !rotary_switch_state;
    last_button_time = button_time;
  }
}

void rotaryBreathe() {
  for(int i=0; i<255; i++){
    setRGBA(rotary_red_val, rotary_green_val, rotary_blue_val, i);
    delay(6);
  }
  
  for(int i=255; i>0; i--){
    setRGBA(rotary_red_val, rotary_green_val, rotary_blue_val, i);
    delay(6);
  }
}

void setRGB(int r, int g, int b) {
  setRGBA(r, g, b, 255);
}

void setRGBA(int r, int g, int b, int a) {
  a = map(a,0,255,0,255);
  analogWrite(ROTARY_RED_PIN, map(((r * a)>>8),0,255,255,0));
  analogWrite(ROTARY_GREEN_PIN, map(((g*a)>>8),0,255,255,0));
  analogWrite(ROTARY_BLUE_PIN, map(((b*a)>>8),0,255,255,0));
}
```

I have a feeling the issue is interrupts, as the rotary encoder switch doesn't work either.

Any ideas?

NicoHood, 2019-04-21:
I had a similar problem, maybe it is the same?
https://github.com/FastLED/FastLED/issues/492

jbonnett92, 2019-04-22:
I fixed it by using the Adafruit ATMEGA32U4 Breakout board.

closed 2019-04-22 by jbonnett92