FastLED/FastLED #778: Question: WS2812B and interrupt issues

jesdiart, 2019-04-24:
Hello

I am very newbie with Arduino and FastLED, so sorry if this question is very basic. I am trying to implement a very simple feature for an Android app and an Arduino UNO to turn on / off a WS2812B led strip. I have implemented the communication sending a string with the RGB color and a timeout period in milliseconds (not used at the moment). 

Everything seem to work properly but I am experiencing some issues when a color is displayed for more than 20/30 seconds. It gets stuck and I cannot change color nor disable the LEDs. 
I have checked other issues and read the Interrupt problems article (https://github.com/FastLED/FastLED/wiki/Interrupt-problems). After that I have changed the command format to include a "L" as a header before the data is obtained, so the final format of the data is:
"L <R>, <G>, <B>, <TIMEOUT>;"
where R, G, B and TIMEOUT are integers.

I would appreciate any advice. Thanks in advance

This is my sketch:

`#include <FastLED.h>
#define LED_PIN     6
#define NUM_LEDS    50
CRGB leds[NUM_LEDS];

const uint8_t HEADER_L = 76;
const uint8_t SEMICOLON = 59;
unsigned long time_now = 0;
uint8_t period = 0;

void setup() {
  Serial.begin(9600);
  FastLED.addLeds<WS2812B, LED_PIN, GRB>(leds, NUM_LEDS);
}

void loop() {
  if (millis() > (time_now + period) && period > 0) {
    setColor(0, 0, 0);
  }  
  // if there's any serial available, read it until receiving the header:
  while (Serial.available() > 0) {
    uint8_t header = Serial.read();
    if (header == HEADER_L) {
      uint8_t red = Serial.parseInt();
      uint8_t green = Serial.parseInt();
      uint8_t blue = Serial.parseInt();
      period = Serial.parseInt(); 
      
      if (Serial.read() == SEMICOLON) {
        // send data to leds
        setColor(red, green, blue);
        time_now = millis();
      }       
    }
    
    // finally, flush out any data in the serial buffer, as it may have been interrupted oddly by writing out led data:
    while(Serial.available() > 0) { Serial.read(); }
  }
}

void setColor(int red, int green, int blue){
  for(int i = 0; i < NUM_LEDS; i++) { 
    leds[i].setRGB(red, green, blue);
  }
  FastLED.show();
}`

focalintent, 2019-04-24:
I’d suggest printing out what you’re reading from serial to make sure you aren’t getting stuck on one of the Serial reads

jesdiart, 2019-04-24:
I already did that. The variables are read properly and the colors are displayed also. The problem aries after 20 / 30 seconds in orange, the sketch gets stuck without sending any data. When I send a new command it's not received (the loop has stopped).

focalintent, 2019-04-24:
One possibility might be power draw and undercoating the controller, causing it to shut down - you can test for that by setting Num leds to something like 2 or 3.  

Also - with the print output - you can determine exactly where it isn’t coming out of - is it stuck in one of the serial parseInt calls, for example?  Or is serial just never becoming available?  Etc...

jesdiart, 2019-04-24:
The moment it freezes is random and not related to the serial reads (it stops even if I only send one command and keep the LEDs on).

I tested setting the num leds to 2 and it works fine. The power supply is USB so it seems to be a power draw problem.

Thank you very much for your help!

closed 2019-08-12 by focalintent