FastLED/FastLED #782: FastLED and arduino-esp8266

struthio, 2019-04-28:
hi,

I am using FastLED to drive around 422 WS2812B leds.
I work on esp8266 with arduino-esp8266 framework, building it in PlatformIO.

While using framework in version 1.8.0 (Arduino framework 2.4.3) everything works fine but if I change framework to 2.0.0 (Arduino framework 2.5.0), FastLED output is unusable.

in ver.2.0.0 everything complies fine terminal output is fine, just led are not working correctly.

Is there some incompatibility between FastLED and esp8266 arduino framework versions ?

focalintent, 2019-04-28:
None that I'm aware of off the top of my head - thought platformIO isn't yet an officially tested/supported build platform - out of curiosity, does it work if you build using the Arduino IDE and the most recent framework?

DaveCalaway, 2019-06-01:
On PlatformIO 2.10 ( core 3.6.7 ) i have this problem:
```
In file included from .piolibdeps/FastLED_ID126/led_sysdefs.h:43:0,
from .piolibdeps/FastLED_ID126/FastLED.h:41,
from .piolibdeps/FastLED_ID126/FastLED.cpp:2:
/Users/dave/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/Arduino.h:191:14: error: conflicting declaration 'typedef bool boolean'
typedef bool boolean;
```
every time the compiler find "typedef bool boolean" it give me that error.

closed 2019-08-12 by focalintent