FastLED/FastLED #783: ESP32, FastLED, and delay()

davepl, 2019-04-30:
I've debugged a serious problem with an ESP32 issue that comes down to the fact that FastLED replaces the delay() function with it's own non-multitasking-friendly version of a busy wait.

Is there any particular reason that FastLED needs to do this?  I have many tasks running and need to yield CPU time in my loop, but FastLED obscures the delay() function needed to do this.  Before I change it, I hope to understand why this FastLED behavior persists on FreeRTOS on the ESP32, and if its OK to change it?

focalintent, 2019-04-30:
FastLED.delay is designed to continuously run show to drive the dithering engine. It doesn’t replace the system delay function.  If you want delay without calling show in a loop, call delay.  If you want to drive the dithering engine, call FastLED.delay.

t413, 2019-05-02:
Two other ideas for you to try:
- Try designing your code to use the periodic functions `EVERY_N_MILLISECONDS` or `EVERY_N_SECONDS` macros provided by FastLED instead of using more delays. See [here](https://github.com/FastLED/FastLED/blob/master/examples/DemoReel100/DemoReel100.ino#L58) in one of the examples.
- Try adding another FreeRTOS task. Use `xTaskCreate()` in your setup() method.

closed 2019-08-12 by focalintent