FastLED/FastLED #784: coding error (ws82812) countdown timer

skennyboiii, 2019-05-01:
#include <FastLED.h>

#define DATA_PIN 11
#define LED_TYPE SK6812
#define COLOR_ORDER RGB
#define NUM_LEDS 56
#define BRIGHTNESS 250
#define FRAMES_PER_SECOND 5
#define CRGBSet
#define leds

const uint8_t pps = 4; // number of Pixels Per Segment
CHSV segON10(96, 255, 255); // color of 10s digit segments NOT TURNED RED
CHSV segON(96, 255, 255); // color of 1s digit segments

CRGBArray<NUM_LEDS> leds; /* CRGB leds[NUM_LEDS]; <--not using this. Using
CRGBArray instead. */
// Name segments (based on layout in link above) and define pixel ranges.
CRGBSet segA ( leds(pps * 0, pps - 1+(pps * 0) ));
CRGBSet segB ( leds(pps * 1, pps - 1+(pps * 1) ));
CRGBSet segC ( leds(pps * 2, pps - 1+(pps * 2) ));
CRGBSet segD ( leds(pps * 3, pps - 1+(pps * 3) ));
CRGBSet segE ( leds(pps * 4, pps - 1+(pps * 4) ));
CRGBSet segF ( leds(pps * 5, pps - 1+(pps * 5) ));
CRGBSet segG ( leds(pps * 6, pps - 1+(pps * 6) ));

void LEDStrip_setup() {
pinMode(DATA_PIN, OUTPUT);
FastLED.addLeds<LED_TYPE, DATA_PIN, COLOR_ORDER>(leds,
NUM_LEDS).setCorrection(TypicalLEDStrip);
FastLED.setBrightness(BRIGHTNESS);
FastLED.clear(); }

void LEDStrip_loop() {
FastLED.delay(1); }

void setSegments(uint8_t count) {
// Based on the current count set number segments on or off
uint8_t c1 = 0; // Variable to store 1s digit
uint8_t c10 = 0; // Variable to store 10s digit
uint8_t c;
CHSV segCOLOR(0, 0, 0);

if (count > 9) { // Split out 1s and 10s digits if count is greater then 9
c1 = count % 10;
c10 = count / 10;
} else {
c1 = count;
c10 = 0;}
// Operate on 1s digit segments first, shift them over, and then do the 10s digit segments

for (uint8_t i = 0; i < 2; i++) {
if (i == 0) {
c = c1;
segCOLOR = segON;
} else {
c = c10;
segCOLOR = segON10; }

segA = segB = segC = segD = segE = segF = segG = CRGB::Black; // Initially set segments
if (c == 0) {
segB = segC = segD = segE = segF = segG = segCOLOR; }
if (c == 1) {
segB = segG = segCOLOR; }
if (c == 2) {
segA = segB = segC = segE = segF = segCOLOR;}
if (c == 3) {
segA = segB = segC = segF = segG = segCOLOR;}
if (c == 4) {
segA = segB = segD = segG = segCOLOR;}
if (c == 5) {
segA = segC = segD = segF = segG = segCOLOR;}
if (c == 6) {
segA = segC = segD = segE = segF = segG = segCOLOR; }
if (c == 7) {
segB = segC = segG = segCOLOR;}
if (c == 8) {
segA = segB = segC = segD = segE = segF = segG = segCOLOR;}
if (c == 9) {
segA = segB = segC = segD = segF = segG = segCOLOR; }

if (i == 0) { // Shift segments over to 1s digit display area
for (uint8_t p = 0; p < (7 * pps); p++) { leds[p + (7 * pps)] = leds[p]; }}}}

ERROR:

In file included from /Users/swaggyt/Documents/Arduino/sketch_may01c/sketch_may01c.ino:2:0:
/Users/swaggyt/Documents/Arduino/libraries/FastLED/FastLED.h:14:21: note: #pragma message: FastLED version 3.002.005
 #    pragma message "FastLED version 3.002.005"
                     ^
/Users/swaggyt/Documents/Arduino/sketch_may01c/sketch_may01c.ino:19:1: warning: declaration does not declare anything [-fpermissive]
 CRGBArray<NUM_LEDS> leds; /* CRGB leds[NUM_LEDS]; <--not using this. Using
 ^
sketch_may01c:22:14: error: expected constructor, destructor, or type conversion before '(' token
 CRGBSet segA ( leds(pps * 0, pps - 1+(pps * 0) ));
              ^
sketch_may01c:23:14: error: expected constructor, destructor, or type conversion before '(' token
 CRGBSet segB ( leds(pps * 1, pps - 1+(pps * 1) ));
              ^
sketch_may01c:24:14: error: expected constructor, destructor, or type conversion before '(' token
 CRGBSet segC ( leds(pps * 2, pps - 1+(pps * 2) ));
              ^
sketch_may01c:25:14: error: expected constructor, destructor, or type conversion before '(' token
 CRGBSet segD ( leds(pps * 3, pps - 1+(pps * 3) ));
              ^
sketch_may01c:26:14: error: expected constructor, destructor, or type conversion before '(' token
 CRGBSet segE ( leds(pps * 4, pps - 1+(pps * 4) ));
              ^
sketch_may01c:27:14: error: expected constructor, destructor, or type conversion before '(' token
 CRGBSet segF ( leds(pps * 5, pps - 1+(pps * 5) ));
              ^
sketch_may01c:28:14: error: expected constructor, destructor, or type conversion before '(' token
 CRGBSet segG ( leds(pps * 6, pps - 1+(pps * 6) ));
              ^
/Users/swaggyt/Documents/Arduino/sketch_may01c/sketch_may01c.ino: In function 'void LEDStrip_setup()':
sketch_may01c:32:54: error: expected primary-expression before ',' token
 FastLED.addLeds<LED_TYPE, DATA_PIN, COLOR_ORDER>(leds,
                                                      ^
/Users/swaggyt/Documents/Arduino/sketch_may01c/sketch_may01c.ino: In function 'void setSegments(uint8_t)':
sketch_may01c:68:1: error: 'segA' was not declared in this scope
 segA = segB = segC = segD = segE = segF = segG = CRGB::Black; // Initially set segments
 ^
sketch_may01c:68:8: error: 'segB' was not declared in this scope
 segA = segB = segC = segD = segE = segF = segG = CRGB::Black; // Initially set segments
        ^
sketch_may01c:68:15: error: 'segC' was not declared in this scope
 segA = segB = segC = segD = segE = segF = segG = CRGB::Black; // Initially set segments
               ^
sketch_may01c:68:22: error: 'segD' was not declared in this scope
 segA = segB = segC = segD = segE = segF = segG = CRGB::Black; // Initially set segments
                      ^
sketch_may01c:68:29: error: 'segE' was not declared in this scope
 segA = segB = segC = segD = segE = segF = segG = CRGB::Black; // Initially set segments
                             ^
sketch_may01c:68:36: error: 'segF' was not declared in this scope
 segA = segB = segC = segD = segE = segF = segG = CRGB::Black; // Initially set segments
                                    ^
sketch_may01c:68:43: error: 'segG' was not declared in this scope
 segA = segB = segC = segD = segE = segF = segG = CRGB::Black; // Initially set segments
                                           ^
sketch_may01c:91:50: error: expected ',' before '+' token
 for (uint8_t p = 0; p < (7 * pps); p++) { leds[p + (7 * pps)] = leds[p]; }}}}
                                                  ^
sketch_may01c:91:50: error: expected identifier before '+' token
/Users/swaggyt/Documents/Arduino/sketch_may01c/sketch_may01c.ino: In lambda function:
sketch_may01c:91:63: error: expected '{' before '=' token
 for (uint8_t p = 0; p < (7 * pps); p++) { leds[p + (7 * pps)] = leds[p]; }}}}
                                                               ^
/Users/swaggyt/Documents/Arduino/sketch_may01c/sketch_may01c.ino: In lambda function:
sketch_may01c:91:72: error: expected '{' before ';' token
 for (uint8_t p = 0; p < (7 * pps); p++) { leds[p + (7 * pps)] = leds[p]; }}}}
                                                                        ^
/Users/swaggyt/Documents/Arduino/sketch_may01c/sketch_may01c.ino: In function 'void setSegments(uint8_t)':
sketch_may01c:91:63: error: no match for 'operator=' (operand types are 'setSegments(uint8_t)::<lambda()>' and 'setSegments(uint8_t)::<lambda()>')
 for (uint8_t p = 0; p < (7 * pps); p++) { leds[p + (7 * pps)] = leds[p]; }}}}
                                                               ^
/Users/swaggyt/Documents/Arduino/sketch_may01c/sketch_may01c.ino:91:61: note: candidate: setSegments(uint8_t)::<lambda()>& setSegments(uint8_t)::<lambda()>::operator=(const setSegments(uint8_t)::<lambda()>&) <deleted>
 for (uint8_t p = 0; p < (7 * pps); p++) { leds[p + (7 * pps)] = leds[p]; }}}}
                                                             ^
/Users/swaggyt/Documents/Arduino/sketch_may01c/sketch_may01c.ino:91:61: note:   no known conversion for argument 1 from 'setSegments(uint8_t)::<lambda()>' to 'const setSegments(uint8_t)::<lambda()>&'
exit status 1
expected constructor, destructor, or type conversion before '(' token

tobi01001, 2019-05-01:
'#define CRGBSet

' #define leds

Quick look at the code: These are causing your compile error. Could you remove them and try again?

Did not check any further though.

Regards, Toby

skennyboiii, 2019-05-01:
i removed both and i got this error

In file included from /Users/swaggyt/Downloads/LEDStrip/LEDStrip.ino:1:0:
/Users/swaggyt/Documents/Arduino/libraries/FastLED/FastLED.h:14:21: note: #pragma message: FastLED version 3.002.005
 #    pragma message "FastLED version 3.002.005"
                     ^
/var/folders/r6/w2nq2mkn6nzcmsk9y64qvt500000gn/T//cc3ruXnC.ltrans0.ltrans.o: In function `main':
/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/main.cpp:43: undefined reference to `setup'
/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/main.cpp:46: undefined reference to `loop'
collect2: error: ld returned 1 exit status
exit status 1
Error compiling for board Arduino/Genuino Mega or Mega 2560.

tobi01001, 2019-05-01:
Oh well. Yes, arduino needs a setup and loop function to work correctly.... It misses that in your sketch. 
So instead of LEDStrip_setup() it needs to be void setup () and likewise for the loop function.

Regards

skennyboiii, 2019-05-01:
it works! thanks

skennyboiii, 2019-05-01:
could you help me with another code? i would very much appreciate it

i am currently creating a project in which I am using an infrared sensor as an object counter and displaying the number of counts onto a 3-digit 7 segment display (with the use of an rgb ledstrip).

The ledstrip is only displaying the counts from 0-9 (singe digit) and when it reaches double digits it stops counting.

Here is my code:

include <Adafruit_NeoPixel.h>

#define PIN 2
#define LEDSPERSEGMENT 1
int ir_pin = 3;
int counter = 0;
int hitObject = false;

Adafruit_NeoPixel strip = Adafruit_NeoPixel(42, PIN, NEO_GRB + NEO_KHZ800);

unsigned long DigitBytes[] = {
0x0003FFFF, // 0
0x000001F8, // 1
0x001C7E3F, // 2
0x001C0FFF, // 3
0x001F81F8, // 4
0x001F8FC7, // 5
0x001FFFC7, // 6
0x000001FF, // 7
0x001FFFFF, // 8
0x001F81FF // 9
};

void setup() {
strip.begin();
Serial.begin(9600);
pinMode(ir_pin,INPUT);
}

void showDigit(int d, int offset) {
unsigned long digit = DigitBytes[d];

for (int i = offset; i < offset + 21; i++) {
if (digit & 1) {
strip.setPixelColor(i, 0xFF0000);
} else {
strip.setPixelColor(i, 0x000000);
}
digit = digit >> 1;
}
strip.show();
}

void showScore(int score) {
int ones = score % 10;
int tens = score / 10;
int hundreds = score / 100;

showDigit(ones, (LEDSPERSEGMENT * 7));
showDigit(tens, (LEDSPERSEGMENT * 7));
showDigit(hundreds, (LEDSPERSEGMENT * 7));

Serial.print(ones);
Serial.println(tens);
Serial.println(hundreds);

}

void loop() {
int val = digitalRead(ir_pin);

if( (val == 0) && (hitObject == false) ){
counter++;
hitObject = true;

}else if( (val ==1) && (hitObject == true) ) {
hitObject = false;
}

showDigit(counter,0);
showDigit(counter, 21);

Serial.println(counter);
}

marmilicious, 2019-05-01:
@skennyboiii  The FastLED Github here should mostly be used for FastLED library related bug reports.  The code you just posted is using the Adafruilt Neo pixel library.  It would be much better to ask in the Adafruit forums where there are lots of helpful folks that can help you out.  

Or if you are using the FastLED library you can ask questions or share your project at the reddit forums:  http://fastled.io/r

Also, I'm not sure where you got the code you posted in your very first post above, but the original can be found here if needed:
https://github.com/marmilicious/FastLED_examples/blob/master/CRGBSet_7segment_2digits.ino

closed 2019-08-12 by focalintent