FastLED/FastLED #786: Fastled compile error v3.2.0 (works on 3.1.3)

wrickout, 2019-05-09:
My code is bellow:

#include <c:\Program Files (x86)\Arduino\libraries\FastLED-3.2.0\FastLed.h>
#define NUM_LEDS 10
const byte LED_PIN = 53;
CRGB leds[NUM_LEDS];

void setup()
{
	FastLED.addLeds<WS2812B , LED_PIN>(leds, NUM_LEDS);
}

void loop()
{
	leds[0] = CRGB::White; FastLED.show(); delay(30);
	leds[0] = CRGB::Black; FastLED.show(); delay(30);
}

Error is:
Compiling 'LedStrip' for 'Arduino/Genuino Mega w/ ATmega2560 (Mega 2560)'
 
LedStrip.ino:7: In file included from
FastLed.h:14: note  #pragma message  FastLED version 3.002.006
   #    pragma message "FastLED version 3.002.006"
 
ccolX32X.ltrans0.ltrans.o*: In function __base_ctor
controller.h:67: undefined reference to CLEDController  m_pHead

Error linking for board Arduino/Genuino Mega w/ ATmega2560 (Mega 2560)
controller.h:67: undefined reference to CLEDController  m_pHead
Build failed for project 'LedStrip'
controller.h:67: undefined reference to CLEDController  m_pHead
controller.h:67: undefined reference to CLEDController  m_pHead
controller.h:68: undefined reference to CLEDController  m_pTail
controller.h:68: undefined reference to CLEDController  m_pTail
controller.h:69: undefined reference to CLEDController  m_pTail
controller.h:69: undefined reference to CLEDController  m_pTail
 
ccolX32X.ltrans0.ltrans.o*: In function addLeds
FastLed.h:309: undefined reference to CFastLED  addLeds(CLEDController*, CRGB*, int, int)
 
ccolX32X.ltrans0.ltrans.o*: In function show
FastLed.h:492: undefined reference to FastLED
FastLed.h:492: undefined reference to FastLED
FastLed.h:492: undefined reference to FastLED
FastLed.h:492: undefined reference to CFastLED  show(unsigned char)
FastLed.h:492: undefined reference to FastLED
FastLed.h:492: undefined reference to FastLED
FastLed.h:492: undefined reference to FastLED
FastLed.h:492: undefined reference to CFastLED  show(unsigned char)
 
collect2.exe*: error: ld returned 1 exit status

focalintent, 2019-05-09:
Also by including the library the way you did with the full path (instead of just <FastLED.h> ) you are probably breaking the Arduino IDE’s ability to find the other source files that it needs to compile to have everything available.

wrickout, 2019-05-12:
i tried with arduino 1.8.7 and 1.8.9.
<FastLed.h> says cannot found.

focalintent, 2019-05-12:
Arduino's logic for finding libraries is a bit simple and easily broken - the folder in libraries should just be FastLED not FastLED-3.2.0

wrickout, 2019-05-13:
The problem was that I had 3 version of FastLEd in my library folder, and somehow arduino ide link the sources wrong. Now i deleted 2 version and 3.2.1 only in library folder, and compiled ok!
Thanks!

closed 2019-05-13 by wrickout