FastLED/FastLED #787: TM1914 support?

NelloCris, 2019-05-10:
Hi all, I would like to know if this IC TM1914 is supported from FastLed library (platform Arduino).
If not, someone have the datasheet in English of this IC? Thanks in advance.

N.

BYohann, 2019-08-28:
Hello All, i would also like to know. and if rgbww or rgb+cct are supported, they are becoming standards for home lighting projects where RGB is not usefull.

focalintent, 2019-08-29:
Rgbw support will be coming as part of a nearly full rewrite of the library.  I will make an announcement on the project homepage and the Reddit when it’s available (I don’t have a time frame for it yet)

BYohann, 2019-09-09:
It sounds great, i hope it can released this year with roadmap. thank you for your work

grego87, 2019-09-25:
> It sounds great, i hope it can released this year with roadmap. thank you for your work

Unfortunately [there are some bad news](https://github.com/FastLED/FastLED/issues/579#issuecomment-534396376).

oraclesod, 2020-08-09:
Super sorry to read about the loss of one of the community, that super sad :( 

Is anyone still working on developing FastLed ? I also have the TM1914 chipset on my strip and was wondering if it would be possible to add support for this ?

Im more than happy to provide someone with a TM1914 strip etc if someone is able to help code the library for its support, i wish i could do it myself, i can code but not a this level...

I would swap the strip out for something else, but this is the only strip i could find that was addressable RGB with W/WW also on the same strip, i use a ESP8266 with ESPhome to control the W/WW perfectly, but have to use a crappy SP108E with its original firmware to control the colour side and the nasty app that doesn't allow much customization....

Or if someone can point me in the direction to whats needed to make this library work I'm happy to try to self teach myself.. i just dont know where to start...

NonaSuomy, 2022-09-19:
Did these get supported TM1914 RGB+CW+WW? 

https://www.alibaba.com/product-detail/TM1914-RGBww-5in1-addressable-LED-Strip_60702633541.html

https://www.alibaba.com/product-detail/RGBWCCT-addressable-24V-5in1-RGB-CCT_60718722774.html

Thank you.

marmilicious, 2022-09-20:
FastLED only supports RGB at this time.