FastLED/FastLED #790: FastLED and esp8266 v2.5.1

fabiuz7, 2019-05-14:
Hi, 
It seems there is an incompatibility with new esp8266-arduino release v2.5.1. The previous version 2.5.0 was fine. Even the simplest example (blink.ino) fails:

> C:\Users\myuser\arduino\libraries\FastLED/platforms/esp/8266/led_sysdefs_esp8266.h:15:17: error: conflicting declaration 'typedef uint8_t boolean'
> 
>  typedef uint8_t boolean;

zeitwesen, 2019-05-15:
can also confirm the problem. compiling on 2.5.0 is fine, build breaking with 2.5.1

kriegsman, 2019-05-15:
Thank you for the reports.  I'll try to take a look at this today.

kriegsman, 2019-05-15:
Looks like they (ESP) changed their definition from
`typedef uint8_t boolean;`
to
`typedef bool boolean;`
and we (FastLED) had a copy of the old definition in our header file, possibly from when we were working with the pre-2.0 versions of the ESP toolchain.  

Simply commenting out line 15 from led_sysdefs_esp8266.h allows the compilation to succeed.  

There could potentially be some extremely small and subtle semantic differences form this change, as has been discussed over in this related Arduino/esp issue https://github.com/esp8266/Arduino/issues/5440 but I really don't think that we're going to run into any of those problems in real life.

"Official" fix pending; in the meantime you're welcome to do your own local work-around by opening `arduino/libraries/FastLED/platforms/esp/8266/led_sysdefs_esp8266.h` and deleting line 15, the one that says `typedef uint8_t boolean;`.

Tested that compilation succeeds with this change under all published ESP8266 toolchains from 2.0.0 through 2.5.1.

kriegsman, 2019-05-15:
cc'ing @samguyer as a heads-up, just in case.

closed 2019-05-15 by kriegsman

samguyer, 2019-05-15:
Thanks @kriegsman. There have been a few of these weird collisions with ESP32 definitions.

zeitwesen, 2019-05-15:
Thanks @kriegsman. I can confirm it is working with your fix. Nice work!

commit 6b6c9c1c (2019-05-15): Removed conflicting (and unneeded) typedef of 'boolean', fixes #790