FastLED/FastLED #793: Very strange behaviour when manipulating 'crgb.red' value (Single pixel, CRGB type)

galgametentacle, 2019-05-17:
Cannot manipulate the value of a certain CRGB object's red channel, such as 'if (mycolor.red<255) { mycolor.red += 1; }'. Precisely, I can add by one, but it will immediately go back to zero. Here's my code, and in my serial port window shows 1,0,1,0.... Thank you

`#include <FastLED.h>

#define LED_PIN     5
#define NUM_LEDS    54
#define BRIGHTNESS  64
#define LED_TYPE    SM16703
#define COLOR_ORDER GRB
CRGB leds[NUM_LEDS];
#define UPDATES_PER_SECOND 100
CRGBPalette16 currentPalette;
TBlendType    currentBlending;
CRGB fuckingcolor;

void setup() {
  FastLED.addLeds<LED_TYPE, LED_PIN, COLOR_ORDER>(leds, NUM_LEDS).setCorrection( TypicalLEDStrip );
  FastLED.setBrightness(  BRIGHTNESS );
  Serial.begin(9600);
  fuckingcolor = CRGB(0,0,0);
}

void loop() {
  uint8_t secondHand = (millis() / 1000) % 20;
  if (secondHand >= 3, secondHand < 11) {Serial.println(fuckingcolor.red);if(fuckingcolor.red<255){fuckingcolor.red+=1;}}
  if (secondHand >= 11, secondHand < 14) {Serial.println(fuckingcolor.red);if(fuckingcolor.red>0){fuckingcolor.red-=1;}}
  fill_solid(leds,NUM_LEDS,fuckingcolor);
  FastLED.show();
  FastLED.delay(1000 / UPDATES_PER_SECOND);
}`

focalintent, 2019-05-18:
The problem is in these two lines of code, specifically in the if statements:

```
if (secondHand >= 3, secondHand < 11) {Serial.println(fuckingcolor.red);if(fuckingcolor.red<255){fuckingcolor.red+=1;}}
  if (secondHand >= 11, secondHand < 14) {Serial.println(fuckingcolor.red);if(fuckingcolor.red>0){fuckingcolor.red-=1;}}
```

See how you say `secondHand >= 3,  secondHand < 11` in the first line?  In C/C++ the value of a set of expressions separated by a comma is the last statement in the expression - so while secondHand is between 3 and 14, then the second expression in your first if statement, `secondHand < 11` and the second expression in your second if statement, `secondhand < 14` will both evaluate to true, which means both sets off statements will be executed.  Change your code to this:

```
if (secondHand >= 3 && secondHand < 11) {Serial.println(fuckingcolor.red);if(fuckingcolor.red<255){fuckingcolor.red+=1;}}
  if (secondHand >= 11 && secondHand < 14) {Serial.println(fuckingcolor.red);if(fuckingcolor.red>0){fuckingcolor.red-=1;}}
```

The && is "and" - and it means that both expressions have to be true for the if to get executed.

closed 2019-05-18 by focalintent

galgametentacle, 2019-05-18:
Thank you for your patient answer, and sorry for my stupid question...