FastLED/FastLED #797: A way to disable pragma messages

mlewand, 2019-05-19:
Would you like to add any option to disable pragma messages displayed during compilation time?

I started using it for ESP8266 programming a little while ago, but the displayed pragmas make getting any valuable output from the compilation info more difficult than it should be.

For example this is the output of a typical ESP8266 compilation:

```
Preparing boards...
Verifying...
In file included from sketch\./src/RgbLedComponent/RgbLedComponent.h:6:0,

                 from x:\projects\leds\app.ino:14:

X:\Arduino\libraries\FastLED/FastLED.h:14:21: note: #pragma message: FastLED version 3.002.001

 #    pragma message "FastLED version 3.002.001"

                     ^

In file included from X:\Arduino\libraries\FastLED/FastLED.h:65:0,

                 from sketch\./src/RgbLedComponent/RgbLedComponent.h:6,

                 from x:\projects\leds\app.ino:14:

X:\Arduino\libraries\FastLED/fastspi.h:110:23: note: #pragma message: No hardware SPI pins defined.  All SPI access will default to bitbanged output

 #      pragma message "No hardware SPI pins defined.  All SPI access will default to bitbanged output"

                       ^
```

I know that information about bitbanging is pretty important, but if it's working for me and it's a n-th compilation it does not bring anything new. Also the library version is not that important as it can be easily checked in Arduino library manager.

What I suggest is adding some sort of _opt-in_ way to disable these pragmas. Like a possibility to define some preprocessor constant or something. Also it would be nice to include this information (that it can be disabled) in the output of these pragmas.

For what it's worth I'm using vscode arduino extension to compile it but I believe same thing happens for Arudion IDE.

And thank you for a helpful library 👍

focalintent, 2019-05-19:
You mean like:

```
#define FASTLED_INTERNAL
```

Before including FastLED.h?

(I don’t have it openly documented because I figure that someone who knows how to read the code to find/disable it can be trusted to use it - given how often I discover people are building with the wrong library or expectations _still_ because of this output on builds, I lean towards leaving them)

mlewand, 2019-05-20:
Yes, this is what I've been looking for. Thanks!

closed 2019-05-20 by mlewand

teastainGit, 2023-01-24:
That is a STUPID solution that works perfectly!
I don't have any FastLED in my project but M5Stack includes it globally as a convenience...
I pasted in your STUPID fix and have a clean looking compile now!
Thanks Terry