FastLED/FastLED #799: BUG: Chipset.h has multiple invalid timing data

henrygab, 2019-05-21:


focalintent, 2019-05-21:
So the problem with data sheets is that sometimes there are multiple datasheets with different sets of timings (I have three different timings in three different WS2811 datasheets, for example :) 

Also while there is a flag for it FastLED doesn’t actually fully support inverted signs chipsets right now.  I’ll take a look at a PR - but there aren’t many issues that are ultimately caused by timing being off -
and so if he wary of taking a PR that is just based on the chipsets without also being in a position to test/verify that said chipsets are in fact working with the newly submitted timings.  

My solution for this is going to be to leave the existing timings alone for now, and making timing specification a lot easier and a lot more flexible in the future for folks to play with.

henrygab, 2019-05-21:
OK.  I totally get the different values from different datasheets, having tried to compare different vendors' stuff before … the numbers can have … flexible meanings.  I'll still make a PR with the more obvious errors.

Do you have a repo for the LED datasheets?

closed 2019-05-24 by henrygab