FastLED/FastLED #804: Gradient color palettes causes Crash

justarandomguyintheinternet, 2019-05-24:
I know there is already a Fix for this #696 but on my Esp whenever i call
`CRGBPalette16 myPal = heatmap_gp;` the still Esp Crashes. Maybe its due to a new Version of the Esp8266 Version 2.5.2. (I am using FastLed v.3.2.6)

kevank, 2019-06-17:
I'm seeing this on v3.2.9 running on an ESP8266 NodeMCU.
Here is the Gradient:

DEFINE_GRADIENT_PALETTE( bhw2_xmas_gp ) {
    0,   0, 12,  0,
   40,   0, 55,  0,
   66,   1,117,  2,
   77,   1, 84,  1,
   81,   0, 55,  0,
  119,   0, 12,  0,
  153,  42,  0,  0,
  181, 121,  0,  0,
  204, 255, 12,  8,
  224, 121,  0,  0,
  244,  42,  0,  0,
  255,  42,  0,  0}

Called with:  CRGBPalette16 palette = bhw2_xmas_gp;

Stack:

Fatal exception 9(LoadStoreAlignmentCause):
epc1=0x402014de, epc2=0x00000000, epc3=0x00000000, excvaddr=0x3ffe8cba, depc=0x00000000

Exception (9):
epc1=0x402014de epc2=0x00000000 epc3=0x00000000 excvaddr=0x3ffe8cba depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffd40 end: 3fffffc0 offset: 01a0
3ffffee0:  3fffdad0 3ffef944 3ffef2b0 4020eb4e  
3ffffef0:  3fffff30 00000000 3ffef2ec 4020aa29  
3fffff00:  3ffe8a71 4020633c 3ffeef8c 3ffef4c4  
3fffff10:  3fffdad0 00000000 3ffef294 402043c6  
3fffff20:  3ffef3a4 00000000 3ffef3a4 4020967d  
3fffff30:  63250069 005e2b00 38005931 4f3f0053  
3fffff40:  00604500 09017023 46021061 36410425  
3fffff50:  05374f06 7c033360 23a8012e 0016dd00  
3fffff60:  34fbff09 ffd3adc0 beffe7c2 40201c20  
3fffff70:  3fff027c feefeffe 3ffef286 4020111c  
3fffff80:  ffcd0000 00000000 feefef22 feefeffe  
3fffff90:  00000000 00000000 00000001 3ffef4c4  
3fffffa0:  3fffdad0 00000000 3ffef494 4020aec0  
3fffffb0:  feefeffe feefeffe 3ffe85bc 401008a5  
<<<stack<<<

Any suggestions?

focalintent, 2019-06-17:
Ah - #696 was for the esp32 - but that fix didn't include the esp8266 - I just checked in a similar fix for the esp8266 - try it out (you'll have to grab master by hand from GitHub - I haven't spun up a new release for Arduino yet) and let me know if that takes care of it for you!

kevank, 2019-06-17:
I just made the same edit locally and it fixed the issue.  Went to check in the change and saw that you already made it.   

Thanks!

justarandomguyintheinternet, 2019-06-18:
Thanks for the fix! ^^

closed 2019-06-18 by justarandomguyintheinternet