FastLED/FastLED #805: Does not compile for STM32F103 blue-pill

Kashouryo, 2019-05-25:
tested on the newest master branch, with STM32duino 1.5. In the 1.5 STM32duino release, some boards name got changed, that might be the case.

In file included from C:\Software\Arduino IDE\portable\sketchbook\libraries\FastLED/led_sysdefs.h:32:0,
                 from C:\Software\Arduino IDE\portable\sketchbook\libraries\FastLED/FastLED.h:41,

                 from C:\Users\think\AppData\Local\Temp\arduino_modified_sketch_17966\Blink.ino:1:

C:\Software\Arduino IDE\portable\sketchbook\libraries\FastLED/platforms/avr/led_sysdefs_avr.h:12:20: fatal error: avr/io.h: No such file or directory
#include <avr/io.h>

                    ^

compilation terminated.

markopl96, 2019-06-22:
I'm using platformIO and I'm getting the same error, did someone found solution for that?

It's looks for me like the library has some hardcoded #include from avr, coz when I compiled the same code from platformIO in ArduinoIDE where I have instaled avr and stm32 boards it compiled fine.

@Kashouryo what IDE are you using? if arduino try to install avr and stm32 cores.

Was sm able to run FastLED with STM32 in platformIO?

Kashouryo, 2019-07-01:
@markopl96 
I am using the original Arduino IDE 1.8.10

H4RL3K4SP3R, 2019-09-03:
I ran into the same or a similar issue, with platformIO 4.0.3 and FastLED 3.3.2.
When using `board = bluepill_f103_c8` i get the same errors with unsuccesful imports of avr/io.h.
If i use `board = genericSTM32F103C8` i run into a different issue. The avr/io.h import seems to only happen when `__STM32F1__` is not defined, then it will default to:
`#include "platforms/avr/led_sysdefs_avr.h"` in /FastLED_ID126/led_sysdefs.h which later causes the avr/io.h import. Using `board = genericSTM32F103C8` seems to correctly define `__STM32F1__` but reveals further issues in /FastLED_ID126/platforms/arm/stm32/fastpin_arm_stm32.h, there it errors in Line 52:85 with `"incomplete type '__gen_struct_GPIOB' used in nested name specifier"`

Fxlr8, 2019-10-15:
same for me,`incomplete type '__gen_struct_GPIOA' used in nested name specifier` error

stm32f100CB, platformIO, fastled 3.3.2

avinashsaraf, 2019-10-29:
Hi, I am getting same error can anyone please help me with the fix or is it still open?

markopl96, 2019-10-29:
@avinashsaraf what I did is kind of stupid 😄 but I was tired of looking for a solution so to walk around it I copied files from PIO into Arduino IDE and compile it there.

btw, it's really strange for me that solution was still not found, PIO is really popular just like STM MCU and FastLED so You would think that this is gonna be a big/common issue.