FastLED/FastLED #806: Documentation organisation is poor.

ah3243, 2019-05-25:
I'm trying to look for a definition of certain functions or methods and there's no apparent easy way to understand what parameters are needed. Why should i have to read a whole tutorial to find one bit of information. Such a great project is let down by poor documentation organisation/usability. Please fix this..

focalintent, 2019-05-25:
There's documentation here - http://fastled.io/docs/3.1/ - which is linked to/referenced from the project wiki - https://github.com/FastLED/FastLED/wiki/Frequently-Asked-Questions

I want to do an overhaul of the library documentation (but not until I finish the current overhaul of the library that I'm working on) - but time is not something that I have a whole lot of right now, so it's slow moving.

senorblasto, 2019-06-09:
ah3243, who exactly are you asking to fix this?  If not you then who?

closed 2019-08-12 by focalintent